grbl/grbl #636: About spindle control.

Dikabro, 2015-03-25:
Hello. I'm making laser engraver cnc (with only two axes X and Y) and wanted to ask - how I could disable Z axis and make on Digital Pin 12 Spindle enable / disable function and on Digital Pin 11 variable (PWM) spindle control?

chamnit, 2015-03-25:
@Dikabro : There isn't a way to do it now, but you could hack the spindle control.c source code to do what you want. I'm mulling a compile option to use the digital pin D13 as spindle enable disable and get rid of the spindle direction.

closed 2015-03-25 by chamnit

Dikabro, 2015-03-25:
Could you help me with this?

cri-s, 2015-03-25:
You could define z dir and step bit as 8 and use the unused z dir and step
output as any other output pin.

Dikabro, 2015-03-25:
@cri-s could you help me? I'm not a programmer, for me it's too difficult. Please I nees someones help.

cri-s, 2015-03-27:
contact me at otm254824@54.33mail.com

chamnit, 2015-03-27:
@Dikabro : I just finished writing up a compile option to have spindle speed PWM on pin D11 and spindle enable on pin D13 (instead of spindle direction). Spindle direction will be replaced if you enable this option and you'll lose the M4 spindle enable counter-clockwise command, since there is no direction pin. M3 spindle enable clockwise and M5 spindle disable will still work.

Dikabro, 2015-03-27:
Ok. Thank you @chamnit!
I think it would be very useful for people who makes laser cnc. As I use laser I dont need to control any spindle direction. For me best option would be enable/disable spindle (laser) with D12 pin and control its PWM with D11 (as it was on 0,8c GRBL version). :)
When I will download your new compile option I will add there some codes from 0,8c version to control laser PWM instantly.

Dikabro, 2015-03-28:
@chamnit how are things going? :)

chamnit, 2015-03-28:
@Dikabro : The updates were posted last night to the master branch. Download the source and install through the Arduino IDE method. You'll need to alter the config.h file in the library directory where the Arduino IDE has copied the source code.