grbl/grbl #637: Inverse time (G93) not working correctly? [0.9g]

natevw, 2015-03-26:
It is my understanding that if I issue `G1 G21 G91 G93 F60 X-5` as a block of words it means "controlled move in millimeters (relative) using inverse time such that the move takes 1/60 minutes, 5 to the left". That is, I would expect it to move 5 millimeters left in 1 second.

However, it instead takes about 12 seconds to do this move. Perhaps not coincidentally, if I change the final word to `X-50` it takes about 2 minutes! So it is as if `G93` were not enabling inverse time mode at all? This has been listed as supported for a while, AFAICT.

natevw, 2015-03-26:
Tried this with current master (0.9i) and same thing, does not complete the move in 1s.

natevw, 2015-03-26:
Oh well, normal feed mode should be more compatible with RepRap firmwares for [this particular project](https://vine.co/v/O3rPBhEU60L) but won't really know until tomorrow ;-)

chamnit, 2015-03-26:
@natevw : Thanks for the bug report. I'll look into it as soon as I can.

chamnit, 2015-03-27:
@natevw : Doh! Found the error. Total miscalculation. I'll push the fix up later tonight.

closed 2015-03-27 by chamnit