grbl/grbl #639: realtime_status_report resets system

gorazdr, 2015-03-27:
I have a problem with realtime_status_report
if I issue ? command controller starts to send report but then it resets and I get welcome msg.
I tried commenting out portions of realtime_status_report and I found that this part of code is the problem
```
// Report serial read buffer status
  if (bit_istrue(settings.status_report_mask,BITFLAG_RT_STATUS_SERIAL_RX)) {
    printPgmString(PSTR(",RX:"));
    print_uint8_base10(serial_get_rx_buffer_count());
  }
```
I am running 0.9i on atmega2560 
If i compile the same version for atmega328p it work's normally.

Thanks for your suggestions.

chamnit, 2015-03-27:
@gorazdr : Not sure what would be causing this. There isn't anything outstanding in the source code that looks suspicious. What exactly have you changed from the published source code? And how are you compiling it?

gorazdr, 2015-03-27:
All is default except in config.h I have defined CPU_MAP_ATMEGA2560. Right now I am running this on Arduino mega2560 without any peripherals. I compiled the source with make and uploaded to arduino with make flush (changed Makefile CPU to atmeag2560 and Programer  to stk500v2)

chamnit, 2015-03-27:
@gorazdr : Ok. I'll take a look at what's going later this weekend. Keep in mind that the Mega2560 is not officially-supported, but community supported, and things like this will likely crop up. If I can't track down the error quickly, I'll have to leave this problem open.

gorazdr, 2015-03-27:
Chamnit thank you. Here is my full compile upload log:
gorazd@gorazd-EP45-DS4:~/NetBeansProjects/grbl-master$ make flash
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/main.c -o build/main.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/motion_control.c -o build/motion_control.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/gcode.c -o build/gcode.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/spindle_control.c -o build/spindle_control.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/coolant_control.c -o build/coolant_control.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/serial.c -o build/serial.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/protocol.c -o build/protocol.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/stepper.c -o build/stepper.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/eeprom.c -o build/eeprom.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/settings.c -o build/settings.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/planner.c -o build/planner.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/nuts_bolts.c -o build/nuts_bolts.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/limits.c -o build/limits.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/print.c -o build/print.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/probe.c -o build/probe.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/report.c -o build/report.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -MMD -MP -c grbl/system.c -o build/system.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega2560 -I. -ffunction-sections -o build/main.elf build/main.o build/motion_control.o build/gcode.o build/spindle_control.o build/coolant_control.o build/serial.o build/protocol.o build/stepper.o build/eeprom.o build/settings.o build/planner.o build/nuts_bolts.o build/limits.o build/print.o build/probe.o build/report.o build/system.o -lm -Wl,--gc-sections
rm -f grbl.hex
avr-objcopy -j .text -j .data -O ihex build/main.elf grbl.hex
avr-size --format=berkeley build/main.elf
   text	   data	    bss	    dec	    hex	filename
  28964	      6	   1677	  30647	   77b7	build/main.elf
avrdude -c stk500v2 -p atmega2560 -P /dev/ttyACM0 -D -B 10 -F -U flash:w:grbl.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801
avrdude: reading input file "grbl.hex"
avrdude: writing flash (28970 bytes):

Writing | ################################################## | 100% 4.67s

avrdude: 28970 bytes of flash written
avrdude: verifying flash memory against grbl.hex:
avrdude: load data flash data from input file grbl.hex:
avrdude: input file grbl.hex contains 28970 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 3.74s

avrdude: verifying ...
avrdude: 28970 bytes of flash verified

avrdude: safemode: Fuses OK (H:00, E:00, L:00)

avrdude done.  Thank you.

gorazdr, 2015-03-27:
One side note, when I flash without -D default settings are all out of norm
All int values are 255 all bool values are 1 all masks are 1111111 and all mm/min are -217000.00 (something like that can't remember exactly)

and M30 command does this:
>>> M30
>>> ?
>>> G91 G0  Y1
>>> G91 G0  Y-1
>>> G91 G0  X-1
>>> G91 G0  X1
Not responding until I issue Soft-reset and I get this
ALARM: Abort during cycle
ok

Grbl 0.9i ['$' for help]
['$H'|'$X' to unlock]

my current GRBL settings:
Grbl 0.9i ['$' for help]
>>> $$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=100.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=40.000 (x, step/mm)
$101=40.000 (y, step/mm)
$102=40.000 (z, step/mm)
$110=1000.000 (x max rate, mm/min)
$111=1000.000 (y max rate, mm/min)
$112=1000.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=4000.000 (x max travel, mm)
$131=2400.000 (y max travel, mm)
$132=50.000 (z max travel, mm)
ok

cri-s, 2015-03-27:
I had similar issues when i was compiling one firmware for mega2560.
I increased defaults for buffer and other thing, and firmware was buggy.
buffers was inside limits, ie. below 256, but who knows.
Restored settings and firmware was ok.

chamnit, 2015-03-27:
@cri-s : Like I mentioned before, Mega2560 is not officially supported and will likely have random bugs. If you have noticed specific ones, please report them anyway or help out and submit fixes for them. I generally only have time to ensure the 328p build is as stable as possible.

cri-s, 2015-03-27:
As i don't own 2560, just compiled for friend i don't had possibility
to debug it and investigate.
Further the source was provided / downloaded from i-net and it was a
mix of 0.8c and 0.9g code with some modifications in order i'm
uncertain if the actual source at that time behave similar.
There was some hardware issue that complicated the overall thing
consumed considerable time and because of the nature of only remote
support, i don't have verified it if the same thing
happend with actual source at that time. sorry.

chamnit, 2015-03-27:
@gorazdr : Ok. I just uploaded Grbl to an Arduino Mega2560 R3 board. I followed the standard procedure to upload Grbl through the Arduino IDE and made sure to select the Mega2560 as the board I'm uploading to. I set the `$10=15` status report to show the RX buffer. And everything worked fine. Not sure what's going on your end, but all I can say is try uploading with the IDE.

gorazdr, 2015-03-28:
@chamnit: You are right I tried uploading via Arduino IDE and it works OK.

closed 2015-03-28 by chamnit