grbl/grbl #640: Homing Issue?

109JB, 2015-03-28:
I finally got my milling machine set up and am trying to get the homing sequence worked out correctly. First, I have only a single limit switch on each axis as follows:

X-axis moves negative to activate the switch
Y-axis moves positive to activate the switch
Z-axis moves positive to activate the switch
So, I have the homing direction invert mask set to $23=6

To get my axis motion correct, I have the Direction port invert mask set to $3=2

Curiously, the WIKI states that the normal homing directions are +Z, +X, and +Y which initially made me think that I would need $23=1. When $23=1 though, initiating the homing sequence starts Z moving in the negative direction. Additionally, I changed the config.h file so that the machine homes Z first, then X, then Y. Each axis moves at the homing seek rate, finds the switch, backs off and then approaches at the homing feed rate as expected, but then when it comes time to move to the pull-off distance, it moves closer to the switch rather than farther away.  I attempted to change the pull-off distance from 1.0mm to -1.0mm but received an error that the pull off cannot be negative.    At this point I have the soft limits turned off.

What am I doing wrong?

chamnit, 2015-03-28:
@109JB : This is the general procedure to set up your homing. I really need to write a Wiki on this, because this question comes up a couple times a month.

- First, ensure your coordinate frame and your axes are moving correctly within it. Otherwise, everything following this will get mucked up. Traditionally, if you are standing in front of your machine, Y+ is the tool is moving away from you relative to the work area, Z+ is up, and X+ is the tool moves to the right relative to the work area. There a lot of resources on CNC coordinate frames if you google it.

- Second, ensure your limit switches are wired correctly. By default, Grbl expects normally-open limit switches connected to GND. Meaning that when closed, the pin goes low. When open, the pin is high. Measure these at the pins. If it's the opposite, either invert the limit switch setting to account for the reverse logic, or re-wire them correctly.

- Third, Grbl homes in the positive direction for all axes. See coordinate frame references to see if Grbl is homing the correct direction and your limit switches are located in the right place. If not and you need to have Grbl search in the negative direction, thats when you use the homing direction invert mask.

- Lastly, Grbl will set the entire machine coordinate system in negative space. This is traditional. So the origin will be at the X+,Y+,Z+ most part of your machine, while your machine volume is limited to (-Xmax, -Ymax, -Zmax) travel.

109JB, 2015-03-28:
Thanks for the quick reply.  To address your comments.

1- the axes are all moving correctly which required setting $3=2
2- The limit switches are wired correctly. I have inverted the pins since the switches I have are operating through a breakout board and the breakout board sets the pin high when the switch is activated. I have tested this in GRBL and it operates correctly signalling an alarm within GRBL if a switch is activated.
3- As I said, it is curious that since my z limit is at maximum z positive that I had to invert the homing direction to get the axis to move the correct direction when homing. The homing direction is set to $23=6 which is exactly opposite of what I expected. However, all of the axes initially move correctly when homing is initiated.
4-I can't even get this far because after finding the switch GRBL is moving to a pull-off in the wrong direction.

chamnit, 2015-03-28:
@109JB : It sounds like your limit switches aren't wired the way you thought. Grbl's hard limit alarms upon any pin change (for numerous reason that I won't go into), so your limit swtich test is inconclusive. If your z-axis is set correctly for normal operation, then Grbl should move in the positive direction, not negative. That is unless your limit switch is the opposite. This isn't the first time someone has made this mistake.

109JB, 2015-03-29:
I assure you that the axes are moving in the correct direction.  I am not a newbie to CNC machines, either production machines or DIY, nor is this a new machine. It has been running on LinuxCNC for years. Homing has also been working on LinuxCNC for years in this configuration and was tested with LinuxCNC before and after the attempts with GRBL.

When I give a G91X0Z1 command the head moves upward confirming that the z direction movement is correct. The limit switch is activated when the head moves all the way up.

The switches are working correctly in GRBL as confirmed by the homing motion which is moving toward the switch, switch activates, the axis reverses a little, advance to switch again, repeats, and then when the pull-off is supposed to happen it moves closer to the switch.

I can provide a video if needed, but I am quite sure that the axes are moving the correct direction, and the switches are working and being detected by GRBL.

chamnit, 2015-03-29:
@109JB : Understood. One way to verify my thoughts is to change the homing seek rate. You should see no change in homing search speed. If you change homing feed rate, you should then see a change in speed. When things are working right, you should see the opposite.

Grbl's homing routine is rather dumb. It only moves the seek/search speed once. When it detects the switch, it then does all of the following moves in the homing feed/locate speed. If the switch is not working as you thought, Grbl would have immediately detected the trigger switch and go into locate mode. 

Let me know what you find. If it's indeed a problem on Grbl's end, I'll make sure to take a look at the code tomorrow. Either way, I'm putting in my to-do list that limits the locate cycle travel so that this type of confusion doesn't occur again. And like I said, this has happened more than once, and one user had everything wired right, but the switch itself was faulty. He had to crack it open to see that it was labeled the opposite of the switch operation.

chamnit, 2015-03-29:
@109JB : Also, do me a favor and post you `$$` settings and the `$I` build info string. This will help diagnose the problem if there is something truly wrong with the code. I was messing with the homing cycle a little while ago, but there shouldn't have been any changes that would make things awry.

109JB, 2015-03-29:
I just came in from the shop. Unfortunately no internet out there yet, so I can only post from the house.

As FYI, I have my GRBL Arduino wired up to a DB25 connector such that GRBL talks to the breakout board that is part of the machine system. Set up this way I can switch back and forth between GRBL and LinuxCNC by simply unplugging the GRBL's DB25 and plugging in the paraport on the LinuxCNC machine.

When out in the shop, I re-connected LinuxCNC and confirmed that everything is still working on that setup and it is. LinuxCNC homing worked as expected. I jogged around checking all of the axes for a few minutes. Finally, I moved the machine axes off of the limits by a few inches in each axis before shutting down LinuxCNC.

I wanted to make sure that the GUI I was using (Universal G-code sender) wasn't somehow the culprit, so I did the following using a the Arduino IDE Serial monitor:

I then switched back to GRBL and powered up the machine. I reset GRBL just to be sure that GRBL had freshly loaded all of the settings. I had earlier returned the $23 setting to $23=1 to reflect that the Z limit is at Z+, X linit at X-, and Y limit at Y+ as the machine is set up and as the WIKI describes the $23 setting.  After reset, I typed $X to unlock the machine, and then G20 to set inch units.  I then typed G91G0Z1, and the head moved upward 1 inch as expected. I did the same for the X and Y axes confirming their movement directions. I then typed $H to initiate a homing sequence and the head started moving downward. Since the head was supposed to move upward and there is no switch in the downward direction, I switched off power to my stepper drivers to stop the machine movement before it crashed.

Below are the requested $$ and $I info which results in the Z axis moving Z- when homing. All of the speed values below are converted from the speed values for the linuxCNC setup. In LinuxCNC my maximum axis speeds are 3.0 in/sec (180 IPM), and the homing feedrate is 1.5 in/sec (90 IPM).  I did attempt the homing in GRBL using the default homing feedrate and seek speeds of 50 and 635 mm/min respectively. The results were the same with the Z axis moving in the wrong direction.during homing.

Also, although I am not worried about it at the moment I did find that I had to lower my acceleration values for the axes quite a bit from the linuxCNC settings. In LinuxCNC my acceleration on all axes is 20 in/sec^2 which should equal 508 mm/sec^2, but I get motor stalling unless I lower the acceleration in GRBL to 175 mm/sec^2.

**** Connected to COM3 @ 115200 baud ****

Grbl 0.9i ['$' for help]
['$H'|'$X' to unlock]
>>> $$
$0=3 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=2 (dir port invert mask:00000010)
$4=1 (step enable invert, bool)
$5=1 (limit pins invert, bool)
$6=1 (probe pin invert, bool)
$10=15 (status report mask:00001111)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=1 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=1 (homing dir invert mask:00000001)
$24=1000.000 (homing feed, mm/min)
$25=2300.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=314.961 (x, step/mm)
$101=314.961 (y, step/mm)
$102=314.961 (z, step/mm)
$110=4572.000 (x max rate, mm/min)
$111=4572.000 (y max rate, mm/min)
$112=4572.000 (z max rate, mm/min)
$120=175.000 (x accel, mm/sec^2)
$121=175.000 (y accel, mm/sec^2)
$122=175.000 (z accel, mm/sec^2)
$130=470.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=254.000 (z max travel, mm)
ok
>>> $I
[0.9i.20150315:]
ok

chamnit, 2015-03-29:
@109JB : Thanks for the settings info. I'll check this on my test setup later today.

Meanwhile, were you able to check whether or not changing the `$25` setting alters the homing speed you first see? Like I stated before, this should help verify my suspicions on the causation. 

In addition, you can enable LIMIT_STATE_REPORTING in the config.h (recompile and upload). This will cause Grbl to send back the limit state in its '?' status reports in the form `Lim:010`, where 010 is X(low), Y(high), Z(low). GUIs won't support this and may freak out, so it's best to use a terminal program when using this debug feature.

chamnit, 2015-03-29:
@109JB : Also, as for your acceleration concern, I find it great that you can do a direct comparison of Grbl to LinuxCNC! I've never had the time to check performance differences in depth. Although, I've heard various reports. For one, I've heard that Grbl runs through high-speed machining tool paths much better, where the same tool paths on LinuxCNC would cause the tools to overheat and break. 

This is the first time I heard anything about accelerations. I'm pretty certain that Grbl and LinuxCNC handle acceleration and step generation differently, so it's likely related to that. There are some configure parameters you can tweak for your faster machine. In config.h, try changing the ACCELERATION_TICKS_PER_SECOND to something like 150. You'll also need to uncomment SEGMENT_BUFFER_SIZE and increase it to something like 8. Doing this will help smooth out the acceleration ramps for machines with high accelerations. Let me know if makes any difference.

chamnit, 2015-03-29:
@109JB : Ok, I took the time to upload a fresh build of Grbl and uploaded your settings. This should be an apples to apples comparison since the Arduino is standardized. Nothing is out of the ordinary. Everything works as it should. 

Again, if I were to guess what exactly is wrong, it would be the limit switches or the wiring/electronics between the limit switches and Grbl. Keep in mind that Arduino is bare-metal electronics and maybe more sensitive to external influences than a well-buffered PC port. You have to wire in things that you normally wouldn't and weird things do happen.

109JB, 2015-03-30:
I was away from home most of today, but did have a chance to do what you had asked.

Changing $25 does change the initial movement speed when homing. I tried many different speeds, but I don't think this is it.

I looked in config.h and there was no "LIMIT_STATE_REPORTING" in there, but I think you were referring to "#define REPORT_LIMIT_PIN_STATE" which I un-commented. I ran a ? query several times and I was never able to get a report of the limits. All that is printed on the status line is Lim: as in the cut and paste below. I even tried it by changing $5=0 and unplugging the DB25 cable which should have the pins for limits high, but still nothing showed on the status query. This was tried in alarm as below, and with a single axis on a limit, all axes off their limits, in unlocked mode, etc. Nothing I tried would give any results to a status query. Here is an example of the status report.

Grbl 0.9i ['$' for help]
['$H'|'$X' to unlock]
<Alarm,MPos:0.0000,0.0000,0.0000,WPos:-8.0074,10.8974,15.0111,Buf:0,RX:1,Lim:>
ok

I also re-downloaded GRBL and replaced all of the GRBL files residing on my computer just in case it was a corrupt download or something similar, but hte results were the same.

Having said that, I think the real problem is that the axis when homing is making its initial move in the wrong direction when $23=1.  As stated earlier, my limit swithces are at Z+, X-, and Y+, so $23 should =1. I for the life of me cannot figure out why this is. I have confirmed many times the axes are moving in the correct direction. I put together a video that shows what is happening. The video concentrates on the Z-axis, since it homes first, but I have found that to get the movements in the right direction I needed $23=6 which is the exact opposite of $23=1.  I think this is the crux of the problem.

Please look at the video in the link below and I think you will have a better understanding of what I am talking about.

http://youtu.be/iaKg6DeArz8

chamnit, 2015-03-30:
@109JB : Weird indeed. Thanks for posting the video. I located the issue with the limit pin reporting not working. I'll post it in a few minutes. But, this shouldn't affect how the homing cycle works. What I don't understand is why it works here on a bare Arduino Uno and not your machine. I can't recall if you told me, but how do you upload Grbl to your Arduino?

109JB, 2015-03-30:
I upload using the Arduino IDE V1.6.1. The method I use is as described in the WIKI, to select Examples | grbl | grbl Upload.  I don't think it matters, but I am using an Arduino Nano for this rather than an Uno.

chamnit, 2015-03-30:
@109JB : The Nano shouldn't matter, but I'll order one so that I have one on hand to test. Question though, your video shows LinuxCNC homing. When it does, I see that Z homing to the positive direction first, then X positive, and then Y negative. But, your first post says the the X switch is negative and Y switch is positive. Why the discrepancy? Typo?

109JB, 2015-03-30:
No problem. I confuse a lot of people with that, but it is just my preference. I worked for several years on vertical machining centers in a production environment. On those particular production machines, the table moved right for +X, and away from the operator for +Y. Still a right hand coordinate system, just with the X-Y plane rotated about the Z-axis 180 degrees.  It is what I was used to from that previous experience, so I set my machine up that way.

chamnit, 2015-03-30:
@109JB : So, I spent good amount of time last night reviewing the code and testing it. There isn't anything that I can see with the code that would cause this problem. I think we'll have to eliminate if the Arduino Nano or your electrical connections are the problem. I suggest that you unplug everything from the Arduino except your USB cable. Monitor the Z-direction(D7) and Z-limit(D12) pins for these scenarios. You will need to wire your Z-limit to GND or a NC-switch to ground to simulate your limit switches, as you have inverted them in your settings. 

The main thing we need to look for is whether or not the Z-direction pin is set correctly for a normal `G91G0Z1` move and the beginning of the Z-axis homing cycle, where you find that it's moving the wrong way.

109JB, 2015-03-30:
Good news. I was able to run some tests as you suggested and they pointed me to what I believe is a solution which was quite simple. Simply changing the $5=1 to $5=0 got it working. This leads me to believe that the breakout board is not working how I believed it did. I had though that the breakout board sent a 5V signal to the limit pin of the DB25, but it must simply ground it, essentially making it work like a NO switch.

What I saw happening during testing with a meter hooked to the Arduino pin for Z-direction and ground was that the pin was 0V for a +Z move. With nothing hooked to any other pins, the pin stayed at 0V when homing if $5=0 regardless of the pinstate prior to sending the $H command. With $5=1, I preformed a -Z move which set the Z-dir pin to +5V. Initiating the $h resulted in the pin going to 0V and then very quickly back to 5V. So, I believe that during homing, the routine set up for a +Z, but immediately detected a tripped limit and reversed to -Z.

However if the above is true, when the machine was away from all limits, the pin should be open. With $5=1, GRBL should have seen this as a tripped limit and given an alarm anytime that I used MDI input. As you saw in the video, I was able to do MDI without problem though even with $5=1..  

I am not sure about any of this and I want to do more testing. Unfortunately I have to go out of town for a few days for work and won't be able to do any more testing until I return. 

Please keep this issue open until I have a chance to do this checking which I should be able to complete quickly after my return.

Thanks for your help.

chamnit, 2015-03-31:
@109JB : Thanks for the update, and I'm glad you were able to diagnose the problem.

As for the lack of hard limit checks, you do bring up a good point. There isn't one when Grbl starts up, because it'll just put Grbl into a continuous freeze-reset cycle. But, I think if there is a hard limit check only when you `$X` kill the alarm, that should cover all scenarios. I'll add this to my to-do list to help immediately diagnose problems like yours.

109JB, 2015-03-31:
er re-reading some stuff it sounds like the limits work by looking for a change of state on the pins rather than a specific pin state. Is this correct?  Just trying to understand how the logic works.

chamnit, 2015-03-31:
@109JB : During homing, Grbl checks the actual state of the pins. During operation, hard limits trigger upon any pin change detected, because there is a small chance that a bouncing switch can cause a state check to misread the pin. Users can disable this behavior in config.h, if they want to, but it isn't recommended until they can guarantee well-filtered switch behavior.

doppelhub, 2015-04-01:
@chamnit: "I think if there is a hard limit check only when you $X kill the alarm, that should cover all scenarios. I'll add this to my to-do list to help immediately diagnose problems like yours."

I disagree.  Many people (myself included) use $X to automatically move the machine off its limits when it has crashed (assuming 1 limit switch per axis and soft limits).  I vote NOT adding a hard limit check when $X occurs, as this prevents a machine from automatically uncrashing itself.    

Uncrash code we've adding to our Grbl machine:
-Reset connection
-Unlock machine
-Move XYZ -1,-1,-1 to move off limits
-If Grbl throws error on limit switch state change, reset (it doesn't always do this)
-Home

chamnit, 2015-04-01:
@doppelhub : Yep. You're right. Grbl's state machine is getting pretty complicated so over-sights like this will likely be more common-place. We'll have to come up with an alternative solution.

ashelly, 2015-04-02:
My alternate solution to limit detection was to replace the on-change ISR with a quick check in the stepper ISR.  My thought is that if you check the limits before every step, you won't go more than a few steps past the limit even with a very bouncy switch.  In practice this has worked fine, although I haven't tested on a wide variety of sensors.

I replaced the line in stepper.c that clears the axislock with:
~~~
  uint8_t must_stop = ((LIMIT_PIN^limits.expected)&limits.active);
  if (must_stop) {
    st.step_outbits &= ~limits_stop_axes(must_stop);
    }
~~~
where `limits_stop_axes`  does the alarm checking that was in the ISR and returns a stepbits mask of the axes which are limited.    The limit control logic has become easier because I can just set the `expected` values and the `active` pins as needed. It also removes the need for the tight checking loop in `limits_go_home`.  I can make a pull request if this is useful to you.

chamnit, 2015-04-02:
@ashelly : Nice idea, especially the simplifying of the limit check routines. But If I remember right, I did the tight checking loop in the homing routine because it keeps the updating of that homing lock mask on one side of the ISR. This should in theory allow it to stay as a non-volatile variable and written to only by the main program. I never checked if this was actually true or not though.

109JB, 2015-04-05:
Thought that I would update and also comment.

I checked out my breakout board and I was indeed wrong in how it worked. That was the issue with homing. Nothing with GRBL.

As for the whole unlocking and staying unlocked even if not homed, I do not agree with this. The switch state should be read and show an alarm. As for being able to move off of a tripped switch, there are ways to do it without leaving the machine unlocked indefinitely.  Perhaps unlocking the machine for some period of time (10 seconds to 1 minute maximum) would allow enough time to move off the tripped switch and then the machine could be homed. LinuxCNC allows one move after unlocking.  Keeping it unlocked indefinitely has the potential  for problems. For example, one could unlock and run the machine in an un-homed, unlocked state and then the machine could crash into a hard stop if switches are only installed on one end of an axis. This could happen because if it isn't homed properly, the machine doesn't know where it is and the soft limits mean nothing. on some machines this might not be a big deal if the crash doesn't have enough energy or motor power to do damage, but on many machines, it could break something.

chamnit, 2015-04-05:
@109JB : Thanks for the update. The single move after unlocking is a good idea and would easy to implement. I'll add it to the to-do list as something to investigate. I'll have to make sure that there aren't any other implications with the communication protocol and breaking something. From the looks of it, it shouldn't.

closed 2015-04-05 by chamnit