grbl/grbl #641: D12 always high when issuing M05 or M03 command - 0.9i with PWM

Probedude, 2015-03-29:
Have an old PC board router (IBC Boardmaker), re-purposing to be a laser cutter/engraver but for now want to switch the solenoid Z stage using the spindle enable output on D12.

Commented out the below in 0.9i's (build [0.9i.20150315:]) config.h file
//#define VARIABLE_SPINDLE // Default enabled. Comment to disable.

Recompiled and uploaded in Arduino IDE.  

Checking D12 with DVM when issuing M03 or M05 command and it's always high.

chamnit, 2015-03-29:
@Probedude : Make sure its the config.h that's in the Arduino library folder, not the downloaded zip. The IDE makes a copy.

closed 2015-03-29 by chamnit

Probedude, 2015-03-29:
Ended up I had 2 folders in there due to having problems importing the library.  The config.h I modified was not the folder that was 'active'.  All fixed now and it works - already did a demo cut and worked fine. Sorry for the false alarm.