grbl/grbl #642: command line issue

kd8wdave, 2015-03-29:
I have built a 'home-brew' cnc machine. I am currently testing its movements using the command line function. When I execute ann 'X move' using G1 command by itself with X80 or X-80 at F100 speed it seems to work just fine. I have done this with a pencil as the marker on paper and with a home-made mounted spindle to hold my old dremel (until my purchased spindle comes). My Y axis and Z axis also seem to work fine using the 'grbl controller' GUI. However when I execute the X-move with F3 'or' less, I see the Y axis move also when I have only given the X command.?? This is screwing up my attempt to shave copper on a PCB board down the edge of the copper plate. This problem has been consistent, 
My background is a retired 'process engineer ' at a major steel mill. I have written code on more machines than I can count, and currently still write for PIC controllers for stuff I volunteer for. I built my own Arduino step/dir commands to the 8-step table for the bi-polar motors I am using. My card feeds the dual H-bridge drivers I found on ebay. I have no issues when the F speed command is about 30 mm/min or more. Any thoughts would be appreciated?

cheers

chamnit, 2015-03-29:
@kd8wdave : Thanks for the report. I'm finding it fascinating that more and more LinuxCNC users are starting to move over to Grbl recently. (especially the retired engineers! :) ) 

So, the problem you have is very unusual. The internal axes management is fairly simple and robust. There isn't anything that I can think of immediately that would cause this particular axis mix up problem. I do know that some GUIs do some 'pre-processing' to the g-code before its sent to Grbl. To rule out the GUI, try connecting directly to Grbl through a serial terminal program like CoolTerm. See if it does the same thing there. 

As for the minimum feed rate, Grbl has a conservative lower limit of 1.0mm/min (configured in config.h). This is primarily there to prevent issues related to floating precision for very slow motions below 0.1mm/min. In the past, this would cause problems within the step calculations, but not the axis moved like yours.

To help diagnose this issue, please post your `$$` settings and `$I` build info. Also, if possible, please try verifying that it's not something in your electronics by removing all connections to the arduino and performing the same test. Monitor the output on Pin 2 (X-step) and Pin 3 (Y-step).

chamnit, 2015-03-29:
@kd8wdave : Did a quick check and everything is working right. The direction pins are set correctly and the step pulse pins output correctly. Tried various feed rates from 1.0mm/min, 3.0mm/min, to 300mm/min. No change. Not sure what's going on.

But, once I have your settings, I should be able to do much closer simulation of your setup to see if it's settings specific.

kd8wdave, 2015-03-29:
I will do the things you have suggested. My primary reason for using 
'grbl' is to get away from the parallel port. I really believe the
USB PC to Arduino solution is much easier to deal with. My original 
issue was a bit of a surprise, but will do some more extensive
testing to see what gives. Thanks for response.
BTW the 'grbl contnroller' gui on v9 doesn't do the setup constants 
properly unless you use the command line. Is there any idea
when that feature will be fixed or is it just something to ignore. It 
displays the data just fine, but the command codes are wrong.
I just do the setup through the command line and it works fine that way.
I will report what I find on my issue.

kd8wdave, 2015-03-29:
Thank's Sonny, I'll get the settings and send them to you sometime today.

dave

kd8wdave, 2015-03-29:
Hi Sonny,
Here's a copy of my setup data taken directly from a hyperterm port setup.
Also is a couple of pics of my machine.
I will need some time to try out the different steps you suggested. I 
found I can connect with a hyperterm
port and command the machine, but I need to set up my scope so I can 
capture the step commands on
both the X and Y axis. I will try to do that tonight or tomorrow.

I'm going to get a better program to communicate with,(I can then echo 
my own char commands back to
myself, so I don't lose sequence of events and know if I make a typo.

cheers

I don't have switches installed yet.
I left the travel limits at default for now
I was trying to use the command "G1 X75 F3" or "G1 X-75 F3"
When I did this, I could see the 'Y-axis' stepping (always in the same 
direction).
 > $i
[0.9i.20150315:]
$G
<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
$$
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F120. S0.]
$0=100 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=1 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=314.961 (x, step/mm)
$101=66.929 (y, step/mm)
$102=314.961 (z, step/mm)
$110=150.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=150.000 (z max rate, mm/min)
$120=3.000 (x accel, mm/sec^2)
$121=3.000 (y accel, mm/sec^2)
$122=3.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

Norwegian-Gadgetman, 2015-03-29:
My thoughts are the 'Dual H-bridges you found on eBay'...
(A link to them would be nice)

you're running with 100uS step pulses?
(I think the default is 10uS)
Is there a reason for this long pulse?

What happens if you reduce it to 50uS, 25 or evne 10uS?

kd8wdave, 2015-03-29:
Hi Sir;
I measured the interval between step pulses from the Arduino-Uno device 
over long moves along the X-axis and found there is about a
two milisec interval whether I set the step pulse to 10 us or 100 us. I 
probably can get by with 50 us, but I developed a PIC16F628A program
that takes the step/dir/enable signals and converts that to the 
necessary 8-step wave pulse sequence needed by my stepper motors. The
PIC cards(home brew) send the wave(4-wire) stepping pulses directly to 
the dual H-bridge devices I purchased from ebay. The 100 us is
simply a 'safety margin' that allows me to operate without an 
interrupt-driver routine, so I am simply 'polling' the necessary 
signals. I use three
separate cards( one for each axis)
I think I would miss pulses at 25 us or less. I'm not really sure how 
folks are getting by with a 10 us setting. The other issue I did run 
into was
the unit would just stop after a command from the GUI controller, 
sometimes right in mid stream of a move command. Then I couldn't get it
to restart without a reset. Got to thinking about that, and decided to 
slow down the interface from115k to 38.4k. Haven't had it drop dead
since I did that. Might just be an issue with my Optiplex desktop, don't 
really know for sure. I am quite pleased that this runs using a USB
port and doesn't need a parallel port! Also with all due respect at 72, 
I'm still finding it difficult to just use 'someone else's' program and not
dicker with it myself. Being retired, I have several hobbies and this is 
just one of them.

This is the first time I have fooled with a CNC application, and its 
been fun screwing with it. I have found several things I will improve on
the next one, but I'm going to get this one working correctly first.
There are 'many' suppliers for this motor driver from several sources. 
Here's pics of mine.

Thank you very much for interest.

dave

kd8wdave, 2015-03-30:
Hi Sonny, I found the problem. When I ran the tests with the dremel 
turned off, everything seems to work albeit
with another issue. When I run very slow about 3-5 mm/min, I get some 
jerkiness, and some lost steps. I will have
to check my mechanical setup, to make sure I'm not binding somewhere.
As the the 'Y axis movement', when I turn on the dremel with the power 
cord plugged into the same power strip as the
switching power supply I use for my logic board, I get all kinds of 
'false' steps on the step input points(that are NOT
coming from the arduino!)They are induced by the a-c induction noise 
backfeed from the dremel motor. I plugged the
dremel into a 50 ft power cord with the source on a totally differennt 
circuit and the problem pretty much went away.
I'm going to get an  a-c filter set up to prevent this problem.

Thanks for taking the time to look, but I think I may have caused you 
undue stress.

I did notice, depending on the Fn speed setting, there are different 
timing intervals betweenn steps. After I finish the
filter setup I will test these timing issues, so I can set my parameters 
up accordingly.

My new spindle is a dc spindle, but I will have to look at 'noise' 
issues with it also.

Thanks again
Dave

chamnit, 2015-03-30:
@kd8wdave : No worries. Glad you located the source of your issue. Electrical noise accounts for about 90% of the problems I've seen. It's hard to debug and people often assume it's a problem with Grbl. So, you're not alone! :)

closed 2015-03-30 by chamnit