grbl/grbl #645: grbl for MaxNC 10 -- return of phase driver

1bigpig, 2015-04-03:
My last parallel port PC died, so I have to take drastic action...and made the changes to the latest .9i 03292025 build to support the MaxNC10 (pre 2004) phase signal drivers.  This came at the expense of the spindle, limits and homing.  On the flip side, coolant and probing should still work--not verified--

The phase code was left in "long hand" format so it would be easy to change if you wanted to control some older machine that also uses phase drive signals or control an L298 chip.

https://github.com/1bigpig/grbl-MaxNC

There is also information on settings and wiring up a DB-25 female port, so that no modifications to the machine have to be made.

Hope this is helpful to someone else with a MaxNC10 machine or other phase signal stepper driver.

Bruce

khimaros, 2022-10-07:
@1bigpig thank you for putting this together, it has been extremely helpful.

the diff view https://github.com/grbl/grbl/compare/master...1bigpig:grbl-MaxNC:master?expand=1#diff-4b876b2ab241bf12fbc7bbe63b09a93c9a7013ecc0f707b315565e9e70fe6dd5R232 especially so.

FluidNC's "unipolar" driver uses the same phases when run in half-step mode: https://github.com/bdring/FluidNC/blob/936305d19a8eb280264abdd880e1fdf57883b071/FluidNC/src/Motors/UnipolarMotor.cpp so i am trying to bring up a MaxNC 10 with this and an ESP32 controller.

in your documentation you mentioned that DB25 pin `#13` mapped to `A5` on your controller. do you happen to recall what the function of that pin is? do any of the DB25 pins map to probes or coolant?

1bigpig, 2022-10-08:
khimaros,

I am gonna be completely honest with you, I have not looked at this code in
7+ years.  I took a quick look at the code to see what you were referring
to.  If I remember correctly, I was trying to map the parallel pins on the
MaxNC to the pins of the Arduino.

For a MaxNC10 machine, my Arduino to DB25 was wired:

Arduino pin - DB25 pin (female connector)
D2 - 2
D3 - 3
D4 - 4
D5 - 5
D6 - 6
D7 - 7
D8 - 8
D9 - 9
D10 - 1
D11 - 14
D12 - 16
D13 - 17
A5 - 13
GND - 18

I believe that A5 (pin 13 of the parallel port) was the probe input for the
MaxNC software.  None of the other pins were mapped to the MaxNC controller
on my machine as there were no other "features" that the MaxNC offered on
my machine.  If I remember correctly, my machine was the first "batch" of
MaxNC10-2 model that MaxNC sold.  I believe that they later changed to a
more traditional step/direction controller and then a closed loop system
before they went out of business.

While I have not used my MaxNC in years, I believe that I still have it and
would be happy to try/test/validate the FluidNC implementation of a MaxNC
phase driver.

I hope this helps.

Bruce