grbl/grbl #646: GRBL support for G53?

109JB, 2015-04-04:
I noticed both in the WIKI and in experimentation that GRBL doesn't support G53 for moving in absolute (machine) coordinates. This is something that I have used quite frequently and is something that can be very useful. For example, G90G53G0Z0 would move the spindle all the way up for clearance for doing tool changes and such. G53 isn't modal, so you have to invoke it each time you want to use it. if you are in work coordinates G54-59, after the G53 command it reverts back to the same work coordinate system as before. Having this enabled within GRBL would be ideal, but if not, I am working on my own version of a GUI that will incorporate some macros. I would be able to add it as a macro to my GUI, but as said, I think it would be best to have it inside GRBL.

Thanks,

John B

chamnit, 2015-04-04:
Grbl supports G53. I'll take a look at the wiki to make sure it's listed.

chamnit, 2015-04-04:
Yup it's there on the main page and the Wiki home page, clear as day.

closed 2015-04-04 by chamnit

mtbrown4ayl, 2021-12-28:
Try just G53G0Z0. This seems to work fine for me. I'm kind of late to the party but maybe it will help someone out.