grbl/grbl #647: G53 support

109JB, 2015-04-04:
After reading the WIKI and performing some tests, it looks like G53 is not supported. Is this the case?  

G53 is movement in machine coordinates but it is non-modal and a G53 move requires invoking G53 every time it is used. It is very useful for things like moving to a specific machine position for tool change, part load, etc. I can implement G53 through the GUI I am writing, but I think this would be a nice one to have through GRBL.

As I said,I am currently writing my own GUI that is set up the way I would like to see one. Partly because I wanted a not too complicated programming project to occupy my time when on the road for work, and I am also making it so that it will work well on a small screen like a netbook. Most of the GUI's I have seen are either too big for a 1024x600 netbook screen, or are lacking in features. Mine is set up for a minimum screen size of 640x480. It is coming along nicely I could use it right now, but there are a few things I want to implement first.  Here is a peek at what I'm working on.

[image]

Some other features it has or will have are, a built in serial monitor, a milling calculator, support for canned cycles, support for user macros, and several other things.

chamnit, 2015-04-04:
@109JB : Like a stated last night, G53 is supported. It states so on the Grbl front page and the Wiki home page. That g-code is listed in the non-modal commands. I even tested it this morning on the off-chance that there was a bug, but it works fine.

If you tested it and found a problem, please describe the process to reproduce this bug. If you have found a place in the Wiki that states that G53 isn't supported, please point out where it is, so I can update it.

Lastly, your GUI looks nice. It looks like a fork of @gerritv's GrblPanel project. I would suggest working with @gerritv, who has done a great job in bringing more professional workflows into the Grbl GUI space. He's laid a great foundation to get to where you want to go. I agree, most Grbl GUIs are hobbyist-grade in terms of CNC features and understanding. I'd also like to see that approach more LinuxCNC or Mach3-level interfaces as well. For me, I'd love to see an open-source GUI that supports tool changes. Grbl provides the tools for GUIs to implement it, but doesn't do it on its own mainly not to have to support all types of tool change setups and manage them explicitly.

109JB, 2015-04-04:
Simply missed it in the wiki. Went back and checked again and found an typo in my GUI's code that was causing it to choke on G53. Made some changes and now it is letting it through. I had thought I saw the same thing on serial monitor, but could not duplicate this morning. Probably a case of not thinking straight at 3:00 am. 

My GUI is certainly very similar to GRBLPanel. I started looking at his code with the notion of just modifying it but in the end  decided to write my own from scratch for a couple reasons, one being that I want to target a couple of REALLY OLD Win98 & WinXP laptops I have. They were great machines back in the day and still run flawlessly but can't run a newer OS. The current version of Visual Studio doesn't support these old operating systems and judging by the number of errors thrown when I tried to open the GRBLPanel project in an older version of VS, I think it would be problematic to re-write it to work. Because of this, as I am writing mine from scratch in VS for newer computers I am also porting it to an old version of visual basic that I am running inside a  virtual machine to insure Win98 compatibility. Kind of a pain but I am able to write a piece of code at a time and then duplicate it in the older Visual Basic. Doing it this way I have just a few errors/changes at a time to make to get it working in both. Mine probably won't be as robust as GRBLPanel, but as I have it right now everything on the main page works and I can open and send a g-code file to the machine. Right now it is stupid and just sends the G-code file line by line without checking it or anything, The only thing it does now is ignore comments and strip the whitespace to save buffer, but it works. Next is to add some intelligence to the parser to check the code which is a big part of implementation of canned cycles and user macros. I would not call myself a good programmer, but I get by and a real programmer would likely laugh at my code.

gerritv, 2015-04-04:
The issue on WinXP and Win95 will be the .Net version, not the compiler itself.
I targetted .Net 4.5 on GrblPanel. Did you try retargetting for 3.5 which should be ok for XP? VS2013 community edition lets you go back as far as 2.0 but there might be some functionality that Grbl Panel needs at higher level. Going back might require a bit of manual editing in a project config file. I can help if you want to try.
(I will be adding some configurable buttons and macros soon :-)

Gerrit

109JB, 2015-04-04:
@gerritv , From what I understand, you cannot build an app in VS2013 and have it work in Win98 regardless of any project settings. VS2005 was apparently the most recent version that allowed building apps for Win98.

gerritv, 2015-04-04:
Ah, pity. Can you move that machine up to XP? :-)

chamnit, 2015-04-04:
Just a thought. I think today's ARM processors on the Beaglebone and the new RaspberryPi are probably on the order of speed of your Win98 machines, and maybe even faster. People have had success running other Grbl GUIs on the Linux distributions that they run. GrblWeb has a ready-made installation that creates a web server that you can access and control Grbl directly from a phone or laptop. Unfortunately GrblPanel is Windows only. :(

109JB, 2015-04-04:
Yeah. The beaglebones and that are certainly a possibility. I just thought I would give it a try. Those machines of mine have been sitting around for a long time. One of them, the Win98 machine, is a small form factor laptop and if I could repurpose it for my CNC mill I would accomplish a few things. 1-I would actually have a use for it.  2-It is a nice size for a machine control laptop and would work nicely with a little arm attached to the side of the machine. 3-I have been travelling a lot for work and it gives me something to do in the hotel and on the airplanes. 4-Finally, it costs nothing except time in the hotel staring at the boob tube.

I remember way back when I was actually using that machine that it did not like XP too much. It ran 98 really well, but XP made it into a dog. I may give a lean XP install another try on it.  I recently tried putting Linux on it, but could not get it to work. It doesn't have USB boot ability and has a compact flash CD drive. I can boot and start the linux install CD but it fails during install because it gets to a point where it loses the CD access.

closed 2015-05-17 by chamnit