grbl/grbl #648: Spindle on causes hard/soft limit alarm

tkur, 2015-04-04:
Hello,
I'm having an issue where my spindle causes a hard/soft limit alarm.  I have spindle control enabled, hard limits enabled, and soft limits disabled.  Seems that when my spindle starts rotating fast enough (via program control or manually turned up spindle speed) I get this alarm.  Has anyone else experienced something like this?

chamnit, 2015-04-04:
@tkur : This a very common problem caused by electrical noise from your spindle. Solutions really depend on your setup and even home wiring. Many install low-pass RC circuits on the limit pins and some even have reported problems from the spindle and computer sharing the same surge protector. The Arduino is bare metal electronics and problems like this are pretty standard.

jeferson-simoes, 2015-04-05:
I solved this problem in a machine with a spindle + VFD this way.

Ferrite everywhere. Endstop cable, Spindle cable, USB cable and all AC cables.
A low-pass RC circuit like Chamnit said.
Shielded twisted pair cables to endstop. It's hard to get this cable here, so I used shielded Lan cable.
I used a resistor about 300 ohm. Anyone know what is the ideal value? I hadn't the 1komh that was told me to use:
[image]

tkur, 2015-04-12:
Perfect! Low pass did the trick.  Thank you.

closed 2015-05-17 by chamnit

himijendrix24, 2018-01-20:
That didn't solve the problem for me. Eventually I used an optocoupler (LTV847) on the limit switches 
and (very important!) a diode + capacitor across the motor. That really stopped an alarm being triggered, when the spindle was turned off.

Here is the wiring. The switch cables don't need to be shielded now. 
Just Make sure that the cables from the optocoupler to the GRBL CNC shield don't run with the servo/motor cables or use a shielded cable there.

[image]

jahnj0584, 2018-01-20:
Is your machine frame grounded, and to the right location?

himijendrix24, 2018-01-20:
No, didn't do that because of the low voltage. The body of my spindle is also not grounded.
Maybe i will do it when using the z-probe.

jahnj0584, 2018-01-20:
A spindle is 110+V. Not low. You must ground the frame directly to the spindle powers ground (wall) to properly eliminate all the noise. You're machine is just a giant EMI antennae now.

himijendrix24, 2018-01-20:
Yes, some use +110V, mine is using +12V (mini cnc from china). Yes, that makes sense the spindle case should be grounded.