grbl/grbl #649: A few questions about operation and a few updates.

109JB, 2015-04-05:
Working on my GUI and have a few questions.

1 - When homing, the machine coordinates are set so that the usable area is all in negative space. I know this is the way that GRBL is set up, but is there a way to force what coordinates are set.

2 - Is there a way for the user to reset the machine coordinates other than a hard reset?

3 - I noticed that when I send a G-code to GRBL the status report shows run, the RX buffer filled, but when it goes to idle after the G-code run is finished the buffer still is reported as full. I haven't seen this as a problem, but noticed it and wondered if this was how it is supposed to be. Look at the video at the following link to see what I'm talking about. You will see 2 windows, one is the GUI and the other is my serial monitor showing the GRBL responses. When the status flips from run to idle, the RX buffer is at 120, and then it increases in steps of 1 to 127 which I believe is due to each "?" call. The buffer stays at 127. If I hit run again, the buffer immediately empties and then starts filling again. Is this behavior normal?

http://youtu.be/utek4NV-TgU

chamnit, 2015-04-05:
1. Yes. There is a config.h option called something like HOMING_FORCE_ORIGIN to set it as zero wherever you have your homing switches.

2. No. I don't see much use for this. Machine coordinates reset either upon homing or a hard reset since Grbl has no idea where it is upon a power-cycle.

3. Not sure what's going on, but it looks like you are sending characters to Grbl. Because the buffer fills after the program ends. The serial buffer computation is very simple and there isn't much that can go wrong with it. As this problem has never come up, I would think that it's something to do with your stream.

109JB, 2015-04-05:
Yes, the buffer filling after the end I believe is due to sending the the "?" status update command at 0.2 sec intervals. If you look, at program end, the buffer is at 120, and goes 121, 122, 123, 124.125, 126, 127. The increment of 1 would be the ? command. What I don't understand is why the previous 120 don't clear.

chamnit, 2015-04-05:
@109JB : '?' status updates are directly picked off from the serial stream. They do not get placed into the serial buffer. So it's not that.

109JB, 2015-04-05:
OK. I found what it is. The program I ran has an M02 code for program end. When I issue a M02 in a serial monitor, and then type ? the RX buffer goes up by 1 each time I pull a status update. If I then issue a "~" resume command the buffer goes back down, but I get a response from GRBL like a soft reset.

gerritv, 2015-04-05:
Are you sending just the ? or ?crlf. On GrblPanel I only send a ?

I have nothing available to test with at present, perhaps try same program with another GUI? GrblPanel shows the 2 buffers as well.

109JB, 2015-04-05:
@gerritv - That was it. writeline vs write.

109JB, 2015-04-05:
@chamnit In regard to forcing the machine coordinates during homing, I see that it can be forced to set zero where the switches are, but what about forcing it to something else? For example,what if you wanted something like 0, 7.5, 0 as the coordinates where the switches are?

chamnit, 2015-04-05:
@109JB : Nope. There isn't a way to set machine coordinates to some custom value. Do you have an example where this would be important enough to install?

I added a to-do to make it clear what Grbl is doing during an M2 or M0. The status report state should indicate that it's in a "HOLD" state or something similar.

109JB, 2015-04-05:
@chamnit  

Having the User able to set the machine coordinates could enable the use of soft limits without having limit switches or homing turned on as long as the user has a close reference to set the machine by eye. For example, considering only the x-axis, if it has a maximum travel of 200 mm, the user could set the $130 maximum travel setting to 190 mm and make himself a reference mark, move the axis to the reference by eye and set the machine coordinate.such that the soft limits could work The The example $130=190 mm vs the actual machine travel of 200 mm would provide a 5mm overrun area at each end to combat the inaccuracies of setting this by eye LinuxCNC provides this ability and is how I had my machine set up for quite a while. It provided protection for programmed moves outside of the machine envelope to prevent crashes into the hard stops and possible machine damage.

This is how LinuxCNC handles soft limits without switches:
1 - Upon Machine power up it is jogged to a machine reference position by eye
2 - The User zeros or sets the machine coordinates to reflect that reference position
3 - The machine now uses that along with soft limits to prevent programmed moves that would exceed the soft limit boundary

In regard to the second step above, if limit/home switches aren't installed, LinuxCNC lets the user touch the "Home" button which brings up a dialog box allowing the user to enter the coordinates that the machine is currently at. For example, if the user has a position that he knows is at X=+5 inches in machine coordinates, he can enter 5 into the dialog box and then the X machine coordinate is set to 5, much  like a G10 L2 P1 X5 command would do for setting the  G54 coordinate space.

For me personally, I had my machine set up in LinuxCNC to have the machine space in the +XY quadrant. Some of my programs have G53 moves in them for tool change, part change, etc. I would just have to change those G53 lines in those programs, but it isn't a big deal. 

I think the biggest thing is what I described above in reference to allowing soft limits without switches and would suggest this as something that should be implemented in the future.

gerritv, 2015-04-05:
What you are describing sounds a bit like G28 and G30? I use those for getting back to 'near' home at rapid speed, instead of doing a homing cycle (of course have to Home once after power up). G28/G30 info is persistent. Labelled as Spcl Posn 1 and 2 on GrblPanel

Gerrit

chamnit, 2015-04-05:
@109JB : Perhaps. I would categorize this as a nice-to-have rather than need-to-have. In my view, the machine coordinate system should be transparent to the user and where the origin is not really important.  I'd like to keep things simple, if possible.

Adding the flexibility to use soft limits without homing is intriguing, but this also opens up the issue of inexperienced users misusing it. I think it'll likely cause more problems than solve.

109JB, 2015-04-09:
@chamnit : I have run across something else. When pulling a $# request, the last response for the probe coordinates contains a :0 at the end. See the attached screenshot of pullung the $# on the Arduino IDE serial monitor. Is this supposed to be there? The WiKI indicates that the PRB parameter should be X,Y,Z coords of the last probing.

[image]

chamnit, 2015-04-09:
@109JB : Yes it's supposed to be there. I forgot to document that with the v0.9i push. In v0.9i, new probing commands G38.3,38.4,38.5 are supported. There needed to be a way to track if the probe was successful or not. That `:0` suffix is a boolean that indicates if the last probe was successful.

109JB, 2015-04-11:
As I was working on my GUI I realized another reason that it would be nice to be able to set the machine coordinates.  Work offsets without homing. If a user didn't have homing he would be able to make note of machine coordinates on one of the work offsets and if he lost position, he could use the known position on say G54 to reset the machine coordinates. Once machine position is set relative to G54 G55-G59 will follow. Not absolutely necessary but would be helpful in some situations.