grbl/grbl #651: STEP and DIR at different ports

sebbra, 2015-04-10:
Is it possible to have the STEP and DIR pins for x, y and z at different ports?

The advantage is that you have more flexibility for different arduino shields, 
e.g. [X-NUCLEO-IHM01A1](http://www.st.com/web/en/catalog/tools/PF260715) with L6474.
The disadvantage is that the pins are not set at the exact same time,
they have some cycles between the axes.

Maybe we can have a compiletime option for alternate pin-mappings.

**Example for X_STEP:**
*CPU_map.h*
```C
[..]
  #define X_STEP_DDR        DDRD
  #define X_STEP_PORT      PORTD
  #define X_STEP_BIT           2  // Uno Digital Pin 2
[..]
```
*Stepper.c*
```C
[..]
  if (st.counter_x > st.exec_block->step_event_count) {
    st.step_outbits |= (1<<X_AXIS);
    st.counter_x -= st.exec_block->step_event_count;
    if (st.exec_block->direction_bits & (1<<X_AXIS)) { sys.position[X_AXIS]--; }
    else { sys.position[X_AXIS]++; }
  }

[..]

   X_STEP_PORT = (X_STEP_PORT & ~X_STEP_BIT ) |
                                 ((st.step_outbits >> X_AXIS) << X_STEP_BIT );
[..]

```

chamnit, 2015-04-10:
@sebbra : Yes it's possible to separate them, but I don't plan on installing this or have to support it. It's fairly easy to do yourself.

closed 2015-04-10 by chamnit

dandumit, 2015-04-11:
@sebbra,   as far as I know all "intelligent" stepper motor drivers do 
NOT have a "Dir" pin. they have only Step pin.
Direction has to be changed over SPI.
I am afraid that you cannot use that board and any other kind of ST 
intelligent drivers with GRBL.
Br,
Daniel

On 10/04/2015 3:26 PM, sebbra wrote:
>
> re flexibility for different arduino shields,
> e.g. X-NUCLEO-IHM01A1 
> <http://www.st.com/web/en/catalog/tools/PF260715> with L6474.
> The disadvantage is that the pins are not set at the exact same time,
> they have some cycles between the axes.
>
> Maybe we can have a compiletime option for alt
>

csdexter, 2015-04-11:
@dandumit you're confusing the dSPIN and easySPIN families from ST. easySPIN (what L6474 is) has DIR and STEP inputs (see the [datasheet](http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00043117.pdf), page 16) and SPI is used for configuring electrical properties and reading faults.
dSPIN, on the other hand, has a full kinematic engine and has neither DIR, nor STEP as you tell the chip via SPI how many steps to move, in which direction and with which velocity profile.

sebbra, 2015-04-11:
@dandumit  the L6474 from ST ha dedicated pins for DIR and STEP, the SPI is for changing current and stepmode. You can configure the driver at startup and dont have to use the spi in operating mode, yust use step and dir.

At the moment i changed one Axes (Z) to the ST driver and it is working fine.
The problem with the X-NUCLEO-IHM01A1 is that they dont have the pins at the same Arduino Port.

sebbra, 2015-04-11:
@chamnit : everything is fine, i just wants to know if there is a problem with seperateing the pins.