grbl/grbl #652: Few issues

Seedler1981, 2015-04-10:
I cant get homing to work.  I'm using 0.9i with grbl controller.  When I start homing the axis move in the homing direction but then some will stop as if they have hit the switch and others will continue until they hit the switch.  sometimes they go fast, sometimes they go slow. If they make it to the end the switch will always stop them.  It is very unpredictable.  I have switched the z limit switch to spindle enable as suggested.

I also get an 'error: Undefined feed rate' message if I try to move any axis other than z, but after I move z, x and y then work.

I'm very confused.

Damien.

109JB, 2015-04-11:
If you are trying a G1 move with no federate defined, F10 for example, you will get an error. A G0 move doesn't require a federate since it is at whatever the defined maximum for the machine is.

doppelhub, 2015-04-13:
It sounds like your motors are skipping, possibly because they're stepping too fast.  Can you reliably move the axes with hard limits disabled?  If not, set the maximum velocity parameters much slower and see if the problem persists.

Seedler1981, 2015-04-14:
Hi,  The motors definitely aren't skipping.  Its as if the axes randomly think they have triggered the limit switches.  The machine works great without homing, but I would like to have this feature

109JB, 2015-04-15:
RF noise making GRBL think the switches are tripping. What is your homing debounce set at? It is $26 parameter

Seedler1981, 2015-04-15:
Hi,  homing debounce was originally set at 100, but I changed it to 25.  Made no difference.  So do you think I should shield the wires to switches?