grbl/grbl #653: M3 causes reset

tkur, 2015-04-12:
I setup grbl with spindle control.  When I try to start the motor, I get very strange behaviour.  Check this out.  I hit the hard reset button, and here is what happens after:

Grbl 0.9g ['$' for help]
>>> s1000
ok
>>> m5
ok
>>> m3
ok

**right after "ok", the following gets spit out all at once**

ALARM: Abort during cycle
['$H'|'$X' to                                        ['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
error: Expected command letter
ok
ok
ok
ok
['$H'|'$X' to unlock]
error: Expected command letter
ok
ok
ok
ok
['$H'|'$X' to unlock]

Grbl 0.9g ['$' for help]

Grbl 0.9g ['$' for help]

Any thoughts on that!?

doppelhub, 2015-04-13:
Probably electrical noise in the system.  What motor are you using?  What motor controller?  Have you scoped the digital lines for noise?  Tell us more and we'll brainstorm.

tkur, 2015-04-13:
I thought about noise too.  In fact, the output shown is that under the condition of my spindle not even being powered.  To make things worse, I also removed power to my stepper controllers.  Same gibberish output.  Its different a lot of the time too:

Take 1:

Grbl 0.9g ['$' for help]
>>> s12000
ok
>>> m3
ok
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' 

Take 2:

Grbl 0.9g ['$' for help]
ok
>>> m3

*here it just hung up and required a hard reset*

Take 3:
ok
>>> s12000
ok
>>> m3
ok
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' to unlock]

tkur, 2015-04-13:
I guess I didn't answer your questions though:

Stepper controller:
http://www.aliexpress.com/store/product/Free-Shipping-New-CNC-Single-Axis-TB6600-0-2-5A-Two-Phase-Hybrid-Stepper-Motor-Driver/343255_2034559754.html?spm=0.0.0.0.IUM4Wy

Spindle motor with controller:
http://www.aliexpress.com/store/product/DIY-engraving-machine-300W-Air-Cooling-spindle-motor-speed-controller-kit-MACH3-control-spindle-G-code/433005_1228706923.html?spm=0.0.0.0.ieBiIT

tkur, 2015-04-13:
in each input (including reset pin 1) I have 10 kOhm and a 104 cap as a low pass filter.  That eliminated problems I was having before with noise.

Check out my latest output!
Grbl 0.9g ['$' for help]
>>> s12000
ok
>>> m3
ok
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' to unlock]
ALARM: Abort during cycle
ALARM: Abort during cycle
ALARM: Abort during cycle
ALARM: Abort during cycle
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
ALARM: Abort during cycle
<Alarm,MPos:0.000,0.000,0.000,WPos:1
1.888,44.363,-1.913>
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]

Grbl 0.9g ['$' for help]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
error: Expected command letter
ok
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]
['$H'|'$X' to unlock]

Grbl 0.9g ['$' for help]

Grbl 0.9g ['$' for help]

chamnit, 2015-05-17:
This is typically caused by electrical noise. Power cables too close your your USB cable, a bad ground, and/or your spindle is causing a power spike on the same circuit. If you disconnect the spindle pin and read it with a multimeter, you'll find this isn't a problem with Grbl's software.

closed 2015-05-17 by chamnit

tkur, 2015-05-17:
I thought the same, but I unplugged the spindle, and the stepper drivers. I
also physically detached them from the atmega. The problem persists.

chamnit, 2015-05-17:
Are you using an Arduino or some clone? The type of behavior indicates that the spindle PWM output is interfering with the soft-reset pin. Have you tested on another Arduino, just bare? Have you modified the source code at all? Because this has never happened on any of my test Arduinos (Duemilanove, Uno SMD, Uno DIP).

tkur, 2015-05-17:
Actually I just bought the chip and am using my own breadboard. This is my
only chip so I haven't tested it on another. I tried also putting an rc
filter on the reset line without success.
But when I tested it I really made it bare bones. Completely disconnected
everything. If it makes any difference, I'm using serial over Bluetooth
rather than a USB cable.

The only thing I modified in code was enabling the pwm spindle. I tried
re-flashing from the latest files on github in case I had screwed something
up too.

chamnit, 2015-05-17:
Ok. Then this is still likely an electrical problem with you bare chip and breadboard, like a floating ground acting weird with the PWM and triggering other input pins. It's also possible the Bluetooth RF is causing some problems when in close proximity to the wires.

tkur, 2015-05-17:
I've never seen such behavior so I'm a bit surprised.  I guess I'll have to
struggle through it though!