grbl/grbl #655: motors spin same direction always

naja452, 2015-04-15:
I have my CNC almost completely set up but have run into a strange issue, the homing process complete successfully, all limit switches are set to normally open, and all of the axis move successfully in all directions during the homing process. 

As soon as the homing process is complete however, jogging the machine from Universal gcode sender causes the axis to always move in the positive direction. 

Even if I punch you in the G code for moving in the positive or negative direction manually the X & Y axis always only move in the positive direction. I don't think this is a hardware issue since the homing sequence completes successfully.

Also the z axis works perfectly when jogging from the UI.

timryder, 2015-04-16:
@naja452What velocity setting do you have for $110, $111, $112 and also $120, $121, $122 assuming you're using build > 0.9?

naja452, 2015-04-17:
I am running grbl on an arduino uno and the driver is a generic chinese TB6560 4-axis board (soon to be grbl shield), controlled from UGC sender 1.0.8

Settings:

$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.000 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=232 (homing dir invert mask:11101000)
$24=1000.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=50.000 (x max travel, mm)
$131=50.000 (y max travel, mm)
$132=50.000 (z max travel, mm)

video of behavior

https://www.youtube.com/watch?v=qiUUX4fqwg0

chamnit, 2015-04-17:
@naja452 : That is indeed weird. Some of your settings are not quite right. 

- First, the junction deviation. It should be 0.02mm.
- Your homing direction mask is a weird value. Any particular reason for this?
- Homing seek rate should be greater than the homing feed rate. The seek is used for quickly traversing for rough position and the feed is used for fine locating.

Also try sending G91G0X1 and G91G0X-1 commands to see if that makes a difference.

naja452, 2015-04-17:
I have changed the junction deviation to 0.02 and inverted the speeds for the homing seek and feed to no effect. The recommended codes still got nothing but motor movement in the same direction.

Also I don't quite understand the homing mask and have not changed it from the default.

naja452, 2015-04-17:
I just changed the homing invert mask to 1, this reversed the homing direction of the Y axis, no other effect so far.

chamnit, 2015-04-17:
@naja452 : Very strange. Perhaps it's something with the Chinese stepper drivers and the enable pin being disabled? Set $1=255 and this should keep the steppers enabled between motions.

chamnit, 2015-04-17:
@naja452 : What is the position that Grbl reports after a positive or negative motion? Does it increase and decrease appropriately, while the motion executed is incorrect?

naja452, 2015-04-17:
Yes the GRBL UI does report the motion correctly, also having the drivers constantly enabled had no effect.

chamnit, 2015-04-17:
Ok that means grbl is setting the direction pins correctly, at least in the firmware. Try measuring the direction pin voltages to see if they change. They should. Try it with your stepper drivers connected and disconnected.

naja452, 2015-04-17:
Rookie mistake :). The direction pins for X and Y were swapped. consequently, Whenever one was triggered, its direction pin was always low. when both were triggered (such as while homing) it only appeared to be capable of going backwards because each axis was triggering its own direction pin. ;)

Thanks for the quick responses and setting me looking closer anyways.

timryder, 2015-04-17:
Is it resolved now?

chamnit, 2015-04-17:
Yes

closed 2015-04-17 by chamnit

leohmcx, 2016-07-11:
Hi guys, i have the same problem but i have no idea how to solve it
NathanHartkopf, can you help me saying exactaly what you did?

jahnj0584, 2016-07-11:
Swap your X and Y direction pins.

NathanHartkopf, 2016-07-11:
Yes, that's correct.

leohmcx, 2016-07-11:
Sorry i am not an expert.
Is it a GRBL configuration or just wire invert?

bellow is my actual GRBL configuration

$0=30 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=180.000 (x, step/mm)
$101=180.000 (y, step/mm)
$102=180.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

srivijayj, 2016-11-29:
Hello Friend 
          PLEASE HELP ME! I am using GRBL  0.9 j And universal G_Code Sender, i was try to run jog movement and G code movement  But ""Moter only run -X,-Y,-Z  Negative direction only"" . And i Was swap All Connection but motor could not change direction

jaketeater, 2016-11-29:
Have you double checked your wiring?  In the case above, the person had the X and Y direction pin's switched.  You can find the pins here:
https://github.com/grbl/grbl/wiki/Connecting-Grbl

jahnj0584, 2016-11-30:
Double check your wiring.

srivijayj, 2016-12-01:
wiring also i did check but its not run

vMeph, 2016-12-01:
Why is $3=6 any special reason, set to 0 and make proper conections of motors and test

109JB, 2016-12-02:
There is nothing wrong with $3=6 and it doesn't mean anything is wired wrong.  This will only affect what direction a stepper turns with a positive or negative movement command. Specifically, it makes the Y and Z turn other directions.

coolrohit123, 2018-08-21:
This are my settings same issue, tried swapping X and Y axis on CNC Shield V0.9j
What to do please help me out!!!!!!

$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=1 (homing dir invert mask:00000001)
$24=50.000 (homing feed, mm/min)
$25=635.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=100.000 (x, step/mm)
$101=100.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

jahnj0584, 2018-08-27:
> help i need someone

Check your DIR cables

coolrohit123, 2018-08-28:
jahnj0584 thanks for reply.. which cables motor 4 wire pairs are correct and on board i have tried swapping X Step/DIR and Y Step/DIR pin also but still motors are running in same direction only the difference is now direction is changed.
Is my GRBL settings correct?
At z axis i am using servo and it is working properly. 
waiting for reply.

jahnj0584, 2018-08-28:
You can't run a servo (and steppers) wth regular grbl

coolrohit123, 2018-08-28:
jahnj0584 thanks for reply.. 
I am trying to buid a CNC plotter
X & Y axis is controlled by Stepper motor NEMA 17
z axis is controlled by servo DXW90 
Please suggest me a proper GRBL Shield ???
I am using CNC V3 Shield
Please checkout below attachment!!!!!!
[image]

agjupiter, 2020-04-25:
does anyone solve the problem.. i have also same problem

agjupiter, 2020-04-25:
> Swap your X and Y direction pins.

hey .. will u plz explain me ... how to swapthe dir pin . i m quite noob ... plz

jkportu, 2020-08-02:
> @naja452 What velocity setting do you have for $110, $111, $112 and also $120, $121, $122 and $24, $25. Assuming you're using build > 0.9? Need more information like board you're running grbl from and what stepper drivers you're using at also. Could possibly be the step rate is too high and the motor can't keep up?

I got the same problem. I change arduino boards, different stepper drivers, different stepper motors, etc. the outcome is always the same, which leads me to believe, that is something with the software. very frustrating indeed.
please, if anyone knows the answer, I really appreciate, because this is driving me crazy .

jkportu, 2020-08-02:
> Swap your X and Y direction pins.

if you swap the X and Y, the outcome it will be the same, just the Y will be the one that goes always same direction. also with the Z.
it is the GRBL program itself.
I have tried everything. different motors, different drivers, universal Gcode Sender, GrblControl, etc.
the problem must be the GRBL code itself.
unfortunately, I can't access it and I don't know how to change it.
building a CNC machine with an Arduino, is becoming a chattered dream.
also, controlling 2 stepper motors with the adafruit motor shield, doesn't work. maybe just DC motors.

jkportu, 2020-08-02:
> @naja452 What velocity setting do you have for $110, $111, $112 and also $120, $121, $122 and $24, $25. Assuming you're using build > 0.9? Need more information like board you're running grbl from [...]

I got the same problem, but just with the X axis. all the stepper motors are the same, Nema 23, and I got really powerful drivers, that can go up to Nema 32.
really frustrating. and no answers. all the answers are just nonsense meaningless stuff.

jahnj0584, 2020-08-02:
Take the motor that is spinning the undesired way and swap the pairs on its
driver. If you have AABB make it BBAA and it will now rotate the opposite
direction.

jkportu, 2020-08-02:
yeah, sure. than instead of turning always to the right, it will turn always to the left..(:

Bladezero99, 2024-06-22:
hello i am facing the same issue. I am using universal G code sender (GRBL 0.9), with CNC shield and A4988 Stepper motor drivers. The motors I am using is NEMA 17 with 9V 1A external power source. for both axis I am getting the same direction of rotation regardless of the x+ or x- presses. on universal g code sender the visual display shows proper functioning but motors are always in the same direction.
[0.9j.20160726:]
ok
>>> $$
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 1    (dir port invert mask:00000001)
*** Connected to GRBL 0.9j
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 0    (hard limits, bool)
$22 = 0    (homing cycle, bool)
$23 = 0    (homing dir invert mask:00000000)
$24 = 25.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 250.000    (x, step/mm)
$101 = 250.000    (y, step/mm)
$102 = 250.000    (z, step/mm)
$110 = 500.000    (x max rate, mm/min)
$111 = 500.000    (y max rate, mm/min)
$112 = 500.000    (z max rate, mm/min)
$120 = 10.000    (x accel, mm/sec^2)
$121 = 10.000    (y accel, mm/sec^2)
$122 = 10.000    (z accel, mm/sec^2)
$130 = 200.000    (x max travel, mm)
$131 = 200.000    (y max travel, mm)
$132 = 200.000    (z max travel, mm)
ok
Pls assist me

jahnj0584, 2024-06-22:
Sounds like a physical wiring issue with the DIR pinout

scottrcarlson, 2024-06-22:
Check your wiring.  Sounds like your direction pin never changes state
(probably float around 0v).

Bladezero99, 2024-06-22:
yes sir i tried running another Arduino code to drive the stepper motor via CNC Shield but the direction did not change.  i tried using another CNC shield and checked for wiring. as i placed the cnc shield on top of the ardiuno and only motor and power supply wires was given, the wiring was minimal and i checked it twice. Sir how should i proceed?

scottrcarlson, 2024-06-22:
What voltage level for your step/direction pin. I assume 5v?

Try manually putting the correct voltage on the direction pin. 0v and 5v
(or whatever your driver requires).  You should definitely see a difference.

Before you do anything, put your volt meter on the direction pin and
ground, what does it read? Jog in both directions and see if the voltage
changes.

1. Does the controller output step output signal high and low based on
direction.
2a. If yes, then double check terminal connections to the driver.
2b. If no, then your controller does not command direction change.

If you don't see a change in direction voltage, remove the step wire from
the stepper driver and repeat the test.

The point here is to find where the issue is and reduce the problem space.
First thing we need to know is if your controller is the issue, the wiring
or the amplifier.

Let us know what you find. Also, include the step driver make/model in your
next post.

-Scott

scottrcarlson, 2024-06-22:
Correction #1. Direction Pin (not step pin) type-o

Bladezero99, 2024-06-27:
Sir the driver i am using is A4988. thank you soo much for the advice to use multimeter on the direction pins. On checking i found out that the ardiuno doesn't change state and remains at 0 volts for the pinouts connecting the direction signal to the CNC shield. i replaced it with a new ardiuno and the commands work now. Thank you sooo much for guiding me.

scottrcarlson, 2024-06-27:
Fantastic.  Glad we could help!