grbl/grbl #657: New GRBL Front-End SerialComCNC

UlrichAM, 2015-04-15:
Hi there,
I like to introduce my freeware GRBL front-end for Windows "SerialComCNC".
If you are interested you will find more information here:
http://www.serialcominstruments.com/cnc.php

A short video about new comming features (EasyJob):
https://www.youtube.com/watch?v=PAm-B0i21cY 

For the newest downloads (version0.9c) look at the bottom of following thread:
http://www.mikrocontroller.net/topic/345380#new

[image]

The software itself provides english and german language panel, but the description and help file are still in german only. This will be changed in future.

Greetings
Ulrich Albert M.

cri-s, 2015-04-15:
> ![bild09
> website](https://cloud.githubusercontent.com/assets/11962409/7160046/ec2aa174-e386-11e4-8870-ea97b55f5e7d.jpg)

It is possible that you release linux version too, or as alternative
optionally allows disabling video camera, because this is presumibly
the reason it don't work under wine.

Scott216, 2015-04-15:
That looks really nice, I can't wait to try it.

UlrichAM, 2015-04-15:
@cri-s 
The software is built with Delphi/Pascal. Linux version is not possible. The video interface is always disabled before explicit starting it. I doubt that the software will run under Wine, some people tried it but were not successfull establishing ComConnection.

csdexter, 2015-04-15:
@UlrichAM said:
  > The software is built with Delphi/Pascal. Linux version is not possible. (...)

On the contrary :-) Lazarus (powered by FPC) can compile Delphi (Object Pascal) code just fine and if you really try, you can dust off a copy of Kylix (Borland's former "Delphi for Linux") and re-compile.

Nothing is impossible, unless you make up your mind that way.

UlrichAM, 2015-04-15:
@csdexter 
I am sorry to say that I am using some quite expensive Components not available for Lazarus / Kylix. So no way for a Linux version of SerialComCNC.

UlrichAM, 2015-04-17:
A new version 0.9d is out today. Software download at the bottom of following thread:
http://www.mikrocontroller.net/topic/345380#new

UlrichAM, 2015-10-05:
Newest versionSerialComCNC 0.9f is available.
Enhanced video functions:
Estimate drill points with the cross of the video (WebCam). All drill points will be converted to a g-code list. The list is then transfered to the main window for direct use.
The software is in english and German language. You will find a download link here:
http://www.serialcominstruments.com/cnc.php
[image]
[image]
[image]

closed 2015-10-05 by UlrichAM