grbl/grbl #658: about datalog's problem

ben123456789, 2015-04-16:
Now I used the  firmware Grbl 0.9g,I have some question .Please help me.
1. 
If I transferring the coordinates to Arduino, the Arduino will feedback "OK" , but the motor still moving and not yet arrived the coordinate .How can I solve the problem ? I just want "If I transferring the Coordinate and after motor arrival the coordinate send OK  " 
2. 
Which command can detect the motion status, the program UGS's field show three status is  "active status : idle、alarm 、Run" ,so how can I detect it .

thanks a lot

cri-s, 2015-04-16:
Use G4 . G4 wait until motion done and time expired.
G4P0 should do the job.

timryder, 2015-04-16:
@ben123456789 The UGS field will show Run while grbl is still creating the motion.  When the motion has stopped then the UGS will display "Idle".   Using @cri-s method of adding a G4P0 command to the end of your gcode will ensure that Grbl will emtpy it's planner buffer first before responding to the G4P0 command. AKA motion has stopped.

ben123456789, 2015-04-17:
1.
Thanks cri-s and timryder,I try  send the command "G91 G0  Z50 G4 P0",the motion still moving ,but the Arduino feedback message "OK" show in the UGS interface.

I know the command G4P"second", the mean is pause few second .If I send the command "G4P10",the machine will pause 10 second ,and then  continue to start. 

2. 
so no have any command can detect the arduino motion status ?

thanks a lot.

ben123456789, 2015-04-17:
I just want the motion already done and then feedback the message "OK". Just only that.

thanks

UlrichAM, 2015-04-17:
Sending "G91 G0 Z50 G4 P0" does not what you want.
But
Sending in seperate Lines will do it:
G91 G0 Z50
G4 P0

Greetings Ulrich

ben123456789, 2015-04-20:
If I send the command "G90X-8.3Y8Z-50.0G4P10",the motion will pause 10 seconds and then move to the Coordinate "X-8.3Y8Z-50.0",but I just want the motion move to the coordinate done and then feedback the some words let me know the moving is done. 

Now the command G4P0.is mean pause few second and then motion move to the coordinate,but the arduino got the command "G90X-8.3Y8Z-50.0G4P10" will feedback the "OK" let me know it got the command,but the motion still moving.

I just want some command can feedback the motion moving status ,and let me know the while I can send the other command to arduino.
thanks .

UlrichAM, 2015-04-20:
@ben123456789 

For single commands use EXACTLY what I described above.
For sending G-Code files to  the Arduino read the following:

https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl

and write a program to achieve all what is said in that link.
Otherwise use any ready to use GRBL Frontend.

closed 2015-05-17 by chamnit