grbl/grbl #660: modulate spindle speed with feed speed for laser driving

idanielrobert, 2015-04-18:
Hello, 
I use a laser in place of a drilling machine.
The power of laser is modulated as the speed of a driling machine with PWM.
I use the great accelerating feature of GRBL (ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING).
If I chose a good laser power for the fast moving head, the power is a few to hight when the head is  slowest ( starting and stopping phases)
I try to find where to make an little modif in code, but it's difficult because it's difficult to trace and debug in Arduino.

I find some interesting think in spindle_set_state()  in spindle_control.c module.
I try to use  prep.current_speed and current_pwm to check. But it look that prep.current_speed is not updated in the time when spindle_set_state() is executing.

Is it some one who can help me to find the right way in the labyrinth ?

The sources are well documented and clear, but i need a litle help.

Thank-you in advance

Daniel ROBERT

alpharesearch, 2015-04-18:
Multiple people use lasers already, for some history you may want to have a look here: https://github.com/grbl/grbl/issues/492
But at this time the needed changes are not in the official edge branch (yet), just have a look at my edge branch for example to see the needed changes so the xyz axis don't stop if you change the S value.
https://github.com/grbl/grbl/compare/master...alpharesearch:edge

idanielrobert, 2015-04-18:
Thank-you very much, i will look at your modifs.
I continue to work and i find a solution for down the power when the speed is slow and proportionaly.
I can not give the lines numbers because i already add comments and some changes for test.
The too burned zones are very clean now.
 
In the module stepper.c :

in the st_prep_buffer()
in the   case RAMP_ACCEL: 
    ...
    ...

    setLaser() ;
    break ;
and  default: // case RAMP_DECEL:
    ...
    ...
    setLaser() ; 
    break ;

hare is the setLaser() function :

void setLaser() {
	//if ( gc_state.tool > 0 ) { // not yet working ??????
		if ( gc_block.modal.spindle != SPINDLE_DISABLE ) {
		
			#define SPINDLE_RPM_RANGE (SPINDLE_MAX_RPM-SPINDLE_MIN_RPM)
			float newSpindleSpeed ;
			float speedCoef, laserPart ;
			uint8_t newPwm ;

			speedCoef = gc_state.feed_rate / prep.current_speed ;
			//speedCoef = speedCoef * 1.5 ;
			laserPart = gc_state.spindle_speed / 2 ; // apporter la correction sur 1/4 de la puissance
			newSpindleSpeed = laserPart / speedCoef ;
			newSpindleSpeed = ( laserPart * 1 ) + newSpindleSpeed ;
			
			//newSpindleSpeed += LASER_MIN_POWER ;
			newPwm = floor( newSpindleSpeed * ( PWM_MAX_VALUE / SPINDLE_RPM_RANGE ) + 0.5 ) ;  

  			OCR_REGISTER = newPwm ; // Set PWM pin output
						
		}
	//}
}

[image]

Dikabro, 2015-04-19:
@idanielrobert could you share your grbl zip file? 
I want to test your modification on my laser cnc! :)

idanielrobert, 2015-04-19:
Yes of course. It is, my first try. I suppose that it's possible to make better.
Let me know if you can make better.
How can i add a zip file in this forum ?

idanielrobert, 2015-04-19:
Here is the zip https://www.dropbox.com/s/r7led3httuxwaqx/grbl.zip?dl=0

It's my second incursion in Arduino C world.

truehybridx, 2015-04-19:
Hi idanielrobert, what CAM do you use in your work flow with mod you posted?

idanielrobert, 2015-04-19:
Hello Truehybridx,

I dont Know what is a "CAM" in this context. I just poste a question!  By waiting an answer, i continued to look at a solution. When i find a solution, ( Probably not the best one), i simple post it as a simple suggestion. I stay open for an other better solution. I dont know if i have to make more ? 
It is my first intervention in GitHub. I'm sorry for my bad English.

truehybridx, 2015-04-19:
I meant to ask what you used to make the file that made the laser do the MyOpale design in your pictures. How did you make the g-code that did it

idanielrobert, 2015-04-20:
how, yes, I used Cut2d from Vectric. first I vectorized a bitmap file. then simply create a milling path. the G-code was created with a modified version of the Grbl 0.8 driver. I make my Grbl 0.9 by adding the S parameters for the "plunging" phase. I can give you the g-code file if you want.

idanielrobert, 2015-04-28:
I unfortunately close this issue by error. It's reopened now.

felixa4, 2016-03-30:
Hi, i'm italian,. i try your grbl mod and it hangs when finish .txt at (99%), and I have to press the disconnect button several times to unlock the system. i look at the source code and i see the mod in spindle.c and report.c; are there other ones? it works well with your cnc? thanks, bye

Norwegian-Gadgetman, 2016-03-30:
Are there empty lines at the end of the .txt file?  
Sometimes that can cause strange behaviour.

felixa4, 2016-03-30:
no, i have not empty lines. i use files txt. that other grbl uses  and finish correctly.
this grbl mod works fine for you?

chamnit, 2016-03-30:
@felixa4 : It's likely it's the M2 or M30 program end commands on the last line of your g-code program. Upon receiving one of these commands, Grbl will complete all buffered motions and then pause the job to keep the machine from resuming after a job is complete. Older versions of Grbl required a '~' resume command to exit, but I think I removed that requirement recently.

closed 2016-03-30 by chamnit

felixa4, 2016-03-30:
yes, there is M30. i leave it and try.
thanks

felixa4, 2016-03-30:
where is '~'? there isn' t  in my IT keyboard .....

chamnit, 2016-03-30:
@felixa4 : You can alter the command characters in config.h, if you don't have a '~'. Re-compile and flash the updated build via the Arduino IDE uploading method.

felixa4, 2016-03-30:
OK thanks, but for this now i think that is better to modify footer in laser postprocessor.
bye

felixa4, 2016-03-31:
Well, i have tested this grbl, and i see that:
the compensation is enabled for $28=0 or 1, 10, 20.....;
compensation works fine for up ramp, not for down ramp;
compensation works only when it starts from laser off, not from stop-restart ;
bye

idanielrobert, 2016-03-31:
Hello, thank you for your interest for this subject. In the first version, it's right $28 is just for activate or not. this first version was far to be good, but it was a concept of proof.   I have make a new complete version, and that look near to be good. there is now 3 paramètres for set thresholds and power compensation.  There is lot of modifications in lot of files.  All my Modifications are marked by a comment like //DRO
I can maybe make a zip file with all grbl for send to you ?
How can I give you this version ?

idanielrobert, 2016-03-31:
here is my last version: https://www.dropbox.com/s/luzei8ld6lb9h2b/grblLaserDroxx.rar?dl=0

Thanks to try and help to make perfect :-)

felixa4, 2016-03-31:
hi Daniel, i am happy to work with you to realize a full working laser grbl.
if you want send me the zip file, i look and study your solutions;
i can try grbl when i have time, .... i must work too...........!!!
many thanks, you can send the zip/rar via p.m. or i send you mi email.
bye

felixa4, 2016-03-31:
hi daniel, i have tested your grbl with accel x =2,measuring laser/spindle
PWM output and it works fine;
i don't understand $32 what is doing, but i must set to 1 for good working;
i think that $30 will be always 100%*S,so you complicates the life of the
MCU;
i haven't '~' in my IT kb an i change it in config and report with "*":
after M30 the system hangs and my "*" don't unlock it;
how other  i change for?
many thanks, bye

2016-03-31 15:17 GMT+02:00 filippo giardina <filippogiardina61@gmail.com>:

idanielrobert, 2016-04-03:
Hello, i have too this behavior at the end of a process. I will try with a new fresh not modified Grbl. If necessary, I will then try very carefully make only good changes in it. 

That will take some days....
:-)

idanielrobert, 2016-04-03:
how, sorry, the $32 is the minimum power when the heat is not moving. I added these parameters because the power of some laser are not completely linear. some laser driver are not perfect. If you think that to " uC time's hungry", it's possible to commentarize the codes lines. I can help you for that. first of all I have to re- dive in this code.

felixa4, 2016-07-25:
hi friend, have you fresh update for this beautyful project?
bye, filippo