grbl/grbl #661: Homing issue

Dikabro, 2015-04-19:
Hello. I noticed that homing function doesn't work for me with 0,9i grbl version. 
When I'm pressing any button on grbl sender software (any similar software) I always get "ok" message at the end, for example: 
$X>ok
M3>ok
G92X0Y0 >ok

But when I'm pressing homing button my machine freezes and I dont get any "ok" messages. 
For better understanding I uploaded a video of example. 
My settings:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=2 (dir port invert mask:00000010)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=6 (homing dir invert mask:00000110)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

Here is the video: https://www.youtube.com/watch?v=tJH_q8cXlMQ
Why it is happening? And why my homing function doesnt work?

Dikabro, 2015-04-19:
Maybe it's because of Z axis? I don't have Z axis, maybe I need undefine it?

chamnit, 2015-04-20:
@Dikabro : Yes. It's because of the Z-axis. Grbl assumes you have a 3-axis machine. You can tell Grbl to home only two axes in config.h. You'll have to alter it, recompile, and ref lash.

closed 2015-04-20 by chamnit