grbl/grbl #663: My visual basic controller and a grbl vs LinuxCNC head to head

109JB, 2015-04-21:
I've been working on my GUI in my spare time and thought I would give an update on it. It is to the point where it has just about all of the major components I wanted implemented. It is nothing special, but I needed one that would run in Windows 98 on an ancient laptop I want to use in the shop. I am not a programmer, but can do an OK hack job. Right now the GUI is all there and has the following features I wanted.

[image]

1. All major funcionality on a single page without tabs, other windows, etc.
2. Small window since the computer I plan to use is an 800x600 display. I made the gui 640x480 but it can be maximized to expose more of the code panel
3. Pseudo-jog buttons for each axis with pull-down increments
4. Easy access to G54-G59 work coordinate systems
5. Easy zeroing of each axis on each work coordinate system. On mine you just click the DRO for the axis to zero saving the room a button would take up
6. Persistent settings for many things so they don't have to be set up each time
7. Status line for things like coord system, Grbl run state, units, spndle state, coolant state, and tool #
8. A Grbl settings interface
9. A Com Port settings interface (Not on main page to save room)
10. single step mode
11. Start run from mid-program
12. common buttons for reset, feed hold, resume, homing, etc.
13. a Clock (might as well, and will also implement a cycle timer later)
14. a built-in serial monitor that can run at the same time as the gui in a separate window (Doesn't have to be on for the GUI to operate.
15. MDI Input on the main page
16. support for ALL grbl responses

All of the above is done and here are some of the future enhancements
1. cycle timer
2. link to open current file in a G-code editor (Notepad++)
3. Machining calculator (separate window)
4. Canned cycle support (pretty close on this. Hopefully within a week)
5. Manual tool change and tool table support (Similar to how LinuxCNC implements it)
6. Some other stuff I either forgot or haven't thought of yet.

As for how well it works. It does very well. I have a video of it up on Youtube at the following link. Not the prettiest, but does well. I have been testing it with everything from 10 line programs to 10,000 line programs. The video shows my mill running on grbl with my gui, doing the roadrunner program at 120 inches per minute. There is also an identical run by linuxCNC running the same program at the same programmed feedrate. The results are very interesting.

Program: Roadrunner CNC G-code program
Program lines: 4,471 lines long
program feed rate 120 IPM
Rapid feed rate 150IPM on GRBL, 180 IPM on LinuxCNC
Cycle time with my GUI/GRBL - 3 minutes, 20 seconds
Cycle time with LinuxCNC - 5 minutes 35 seconds

During the runs I noticed that LinuxCNC has a much more pronounced deceleration between programmed moves compared to GRBL. My version of LinuxCNC is a few years old, and I know that there is a new trajectory planner for LinuxCNC, so the differenc gap would likely close up but it is still impressive what GRBL was able to do. I have checked both extensively with regard to lost steps, and neither loses steps with the setup as i have it now. 

Incidentally, I also compared the run of my GUI against GRBL Panel and Universal G-code sender. In both cases, I am about 5% slower than either of them, but I haven't built my project yet and am still running in debug mode. Hopefully it will be a bit faster after building it.

I am interfacing with the machine using an Arduino NANO through my standard parallel port breakout board. here is that interface

[image]
[image]

Here is the link to the video of the GUI along with the GRBL and LinuxCNC head to head runs.
http://youtu.be/c1mxS7WI-tg

109JB, 2015-04-22:
Did a bit more work and now have the first canned cycle. The first one is a G81 drilling cycle, and at this point, it only works on G90, but it is a start.

Here is a video of grbl running my mill using the G81 drilling cycle to air-drill 5 holes. the code is as follows:

G90G20G54
G0X3Y3Z1
F10
G81X2Y2Z-1R0.1
G81X3Y2Z-1R0.1
G81X4Y2Z-1R0.1
G81X5Y2Z-1R0.1
G81X6Y2Z-1R0.1
G0X3Y3Z1

https://youtu.be/7gsQeShHsBk

chamnit, 2015-05-07:
@109JB : This is fantastic. Have you updated your LinuxCNC to see if their new trajectory planner performs better than in this test?

109JB, 2015-05-10:
@chamnit

I haven't upgraded my LinuxCNC install to the new one yet. It hasn't been "officially" released, and my cpu is in my barn where I don't have internet yet which makes it a bit problematic to upgrade right now.  Having said that, a linuxcCNC user that does have the new planner installed ran the same G-code file with the same stepper settings and got  the same 5:35 min on 2.6 and on 2.7 with the new planner he got 3:10 min. So, the new planner in LinuxCNC does make a big difference.

With that said, I have been working on my GUI when I can and I decided to change my approach for a few reasons.  First, for a full implementation of canned cycles, the GUI has to track a bunch of stuff in order to support things like G98/G99 and G90/G91. Tool changes are next and will require a bit more work too.  I also have the idea for future implementation of G40-G42, cutter compensation, which will require even more tracking of modal values, coordinates, etc. On major thing is having to look-ahead at least one move for CDC, so I have written my GUI to support these feature in the future.  

All of this takes computing time, so it has extended cycle times.  My GUI now does look-ahead, and fully parses each line, and supports canned cycles G81, G82, G83, G85, G86, and G89 with full support for G98/G99 and G90/G91, but the cycle times are up. On the roadrunner program, which doesn't even used canned cycles (only G0/G1 moves), the cycle times are about 3:50 compared to about 3:00 without look-ahead or canned cycle support. One reason is it doesn't start processing a line until the previous one has finished. 

It is still a bunch better than the cycle times from LinuxCNC 2.6, but in order to try to optimize things, I am re-writing my code to try to take advantage of the multi-threading features available. Along that path I have a basic implementation that runs the roadrunner code in a cycle time of 3:04 without the canned cycle or look-ahead enabled. This is an actual cycle time from start to GRBL completing everything in its buffer. The cycle timer waits for an "Idle" status at the end of the run and uses that point in time to calculate cycle time.

I also change the GUI a bit making it a little larger to make it easier for touch panel implementation.  I have been playing around with a remote desktop with an old android tablet and it works OK, but the small buttons on my previous version made it hard.  This new interface is 1024 x 600 to support use on a 10" netbook. I am a big fam of the smaller platform laptops/netbooks, and some of the other GUIs are just too big to use on them. I had a 10" netbook that I had added a touchscreen to and it would have been perfect, but I sold it about a year ago. 

Some people have sent me messages to the effect of "Why bother with canned cycles when CAM can take care of it."  This is true if you always use CAM, but I tend to still use my milling machine as a pseudo manual mill using MDI a lot to run it. Being able to type a canned cycle means having one line of code compared to many. Especially if you start looking at peck drilling multiple holes.

Here is my new interface and it has the following updated features

1. progress bars to show program progress, and buffer states
2. Cycle timer
3. User defined macro's

[image]

gerritv, 2015-05-10:
Impressive. 
I'm not sure if threads will give you what you need, I use events a lot where the gui is not blocked waiting for anything. Everything underneath is driven purely by the serial IO events. The various buttons control gates allowing things to continue or not, etc.

It will be a few months yet before I get my shop set up, just moving into the house at end of June. In meantime suffering from severe machining withdrawal since Nov 2014!

More than happy to help with coding advice in the meantime, my (masked) email is listed under my user name.

Gerrit

109JB, 2015-05-11:
@gerritv

Thanks.  I'd love the help, but I'm one of those amateur programmers that doesn't know enough to be able to know what to ask.  I know that you would likely laugh at my code as I know there are things I am doing contrary to best practices. I am learning a lot though and once I learn something I try to go back and correct those things throughout the program. 

 I do agree that the serial IO events are what is driving much of the cycle time stuff. I was playing around with my first version program earlier today and noticed that the GRBL planner buffer is near  empty most of the time, and the read buffer is essentially empty all the time when running with the canned cycles and look-ahead active. So, I modified a few lines to take them out of action and then the planner buffer remained more full during the run and the cycle time reduced quite a bit. This makes me think that the increase in cycle time is at least partly due to GRBL not having enough data in the planner buffer to use its trajectory planner to full effect. So, I'm thinking that one part of the  solution to this may be to go from the simple call/response streaming protocol to the more advanced character counting protocol.

My line parser does a few things that can aid this as well. Since it keeps track of the active modal values and coordinate data, it doesn't re-send that data with each line even if it is listed on the originally read line of code. It deletes whitespace and comments and it also chops off unnecessary precision in coordinates too. This one is user defined in the GUI user settings, but I default it to a maximum of 4 decimal places. So, take the roadrunner code for example. Every line has a G0 or G1 modal value and coordinates to 6 decimal places. A typical block of code is as follows:

G0 X0.000000 Y0.000000 Z0.200000 
G0 X1.179950 Y4.004260 Z0.200000 
G1 X1.179950 Y4.004260 Z-0.100000 
G1 X1.179950 Y4.004260 Z-0.100000 
G1 X1.179950 Y3.980210 Z-0.100000 
G1 X1.175140 Y3.980210 Z-0.100000 
G1 X1.175140 Y4.004260 Z-0.100000 

My parser reduces this block to 

G0X0Y0Z0.2
X1.18Y4.0043
G1Z-0.1
                   <---Blank line sent here
Y3.9802
X1.1751
Y4.0043

That is a difference of 248 characters vs 64 characters including the EOL characters. I'm not sure that makes any difference when using the send/response protocol, but in the character counting protocol, the original code would fill the receive buffer twice whereas the parsed and "cleaned" code only 1/2 fills the receive buffer, potentially giving GRBL 4 times the data for its look ahead.

Perhaps @chamnit  could comment on whether he thinks all of the above could make a significant difference in cycle times.

Thanks again,

John B.

109JB, 2015-05-13:
I went ahead and implemented the character counting streaming protocol. I did it so that I can select either the character counting or the send/response protocol during runtime. I ran several tests and in most cases, there is no discernible difference. The biggest difference was seen in a test program I wrote that simply increments one axis by 0.010 inches several thousand times. The cycle times for the two were 25.0 seconds for the character counting vs. 56.2 seconds for the send response. Obviously this isn't a realistic test, but it highlights where the character counting benefits. From this standpoint alone, it doesn't benefit too much to go to the character counting protocol.  

However, I found that using the character counting does benefit in other areas.  One thing I have found is that simply enabling scrolling of the datagrid that contains the G-code file can increase the cycle time quite a bit. With scrolling disabled and running the roadrunner file, the character counting and send/response are about the same at 2:58 each.  Just enabling scrolling increases the cycle time by 37 seconds when using the send/response protocol, but there is no change when using character counting. Prior to testing the character counting I was going to look at ways to reduce the cycle time impact of scrolling, but may not have to now. This may also take care of the increased cycle time when adding all of the stuff the program has to do for tracking for canned cycles and future CDC>

109JB, 2015-05-19:
I thought I would provide an update on my GUI and the implementation of canned cycles. Full implementation proves to be a pretty complicated task, at least for my feeble mind. For full implementation, the GUI has to know several things that require tracking of every line fed to GRBL. 

For example, a G81 drilling cycle needs to know the retract mode (G98/G99), and it needs to know the Z height at the start of the G81 cycle. Since GRBL buffers the commands, it isn't possible to simply query GRBL for what the programmed Z height from the previous command is. If in G90, absolute mode, one would think that it is possible to look at the previous line to see what the programmed Z height was, but it isn't that simple. Lets say that the program just ran a bunch of X, Y only moves. It may mean that the previous Z command is 1, 2, or 200 steps previous to the G81. It is also possible that the previous command was also a canned cycle in which case the Z command is the bottom of the hole, and not the ending point of the canned cycle. So, this requires tracking of the machine's Z position for every line of the program. 

Things are even more complex if lets say the whole program, including the canned cycle are in G91. In this case, you have to track from when the last G90 Z command was issued, or when the machine was idle at the beginning of the run. 

Complicate things even more if the program switches work coordinate systems (G54-G59).

I had several starts and re-writes, but I finally have it all working during a program run. I do this by having default settings of all of the modal values, and checking the machine position and offsets right when the user hits the run button. Since it was this complicated, I also decided to do a one line look-ahead so that in the future I can try cutter diameter compensation (CDC) implementation. Right now I am tracking 41 different values, and am tracking them for 1-previous move, 2-current move, and 3-next move. The previous move is because canned cycles needs to know some of these values such as G98/G99, the previous Z ending position, etc. The next move is purely for CDC. I am tracking basically everything that LinuxCNC would have even though not supported in GRBL or my GUI so that in the future I don't find myself boxed in for future implementation of features.

The problem I have right now is that it works perfectly during a program run with full implementation of G81, G82, G83, G85, G86, G88, G89, but I have to figure out how to make it work during MDI input. Canned cycles during MDI is what I was originally wanting it for.  I think my approach is going to be to not allow an MDI unless GRBL is idle and I could snatch the machine position from the GRBL status response. The other caveat is that I would default to G98 retract mode unless G99 was specifically on the MDI input line. Essentially making G98/G99 non-modal during MDI. My other option is to track those 41 parameters right from the first connection with grbl, or any time it sees a response containing "Grbl" which would indicate a hard or soft reset.

Still working on it, but don't get enough as much time to work on it as I would like. My wife seems to think there are other things I should be doing in my free time!!!!

chamnit, 2015-05-19:
Having to track states is one of the more unfortunate things of the gcode canon. To start a program mid way, you have to run it and tracks everything that has happened before this point to ensure all the modes are set correctly. It's awful. If it helps, I've been pondering taking some time to write a fully compliant gcode parser that GUIs can pick up and use. Te problem is finding the time and motivation to get it done. (macro B would be included so GUIs can automate custom machine tasks)

ashelly, 2015-05-19:
Would some variation on https://github.com/grbl/grbl-sim/pull/3 help?  It basically lets you compile the grbl gcode parser on your PC, and gives you the same output.  I could imagine turning it into a c library.   From there it's fairly easy to provide language bindings for most modern languages.
(I should have time to rebase that PR this week.)

109JB, 2015-05-20:
The parser for this actually wasn't too bad using regex and select-case. not sure what the best method is, but I did it this way

1- remove white space
2- convert all to uppercase 
3- use regex to remove comments
4- use regex to split apart command words (G1, F100, X10, etc)
5- use select-case and a few if thens to put it word into my tracking array in the appropriate spot.

I was able to do it with only 2 regex commands. Not sure how many code lines there are, but it is not long at all. 

The biggest problem I had was trying to track only what I needed, only to forget something. That of course required some rewrite. Then some more and more. I then eventually decided to just track everything whether I need it or not. Plus never having been a very good programmer and having been literally years since I did ANY programming and it slowed me down.

I have a few more things I want to do and once done I want to get this into the hands of some users. I guess you would probably consider them alpha testers, but I plan to have it pretty much ready to go by then.

Probably going to be another week or two though since I have some other stuff I have to take care of.

Wish me luck

109JB, 2015-08-13:
I finally had some time to do some more work on my GUI and had a situation that is puzzling. While testing some stuff I found that G2 and G3 commands were not working correctly. Rather than move in a arc, they were simply moving straight line from the start position to the final position.  This is a GRBL upload that has been working fine for about a month or so with no changes made during that time. What I found was that this was not due to my GUI, but something happened to GRBL. I verified it using a simple serial monitor to send the G2/G3 commands, but the machine still moved straight line rather than in an arc. The GRBL upload was set to echo commands and the echo response from GRBL was confirmed correct. I tried resetting and unplugging to no avail. I finally reloaded GRBL using the Arduino IDE and reset the EEPROM settings which fixed it.  I should have tried just an EEPROM reset first, but didn't. However, I don't see anything in the EEPROM settings that would account for this behavior. Has anyone run across anything like this before?

chamnit, 2015-08-13:
@109JB : Do you recall what version date you had when you ran into the G2/3 problem? Were able to see anything wrong with the EEPROM settings or stored values during this time? And was it a certain type of arc calculation that would cause this?

I can't recall any particular issue like this, except, maybe a year ago, I fixed an arc calculation for rare arc situations.

109JB, 2015-08-13:
Yeah, I should have checked the version date, but did not, but when I reloaded GRBL I didn't download the most recent, but used the one that I had previously loaded. So I should be able to get you the version date later when I am home. I can tell you that it was downloaded from github some time around the end of June. 

When the problem happened, I did the normal $$, $#, $G, and $N requests and didn't see anything out of the ordinary.  My $N blocks were as follows:

$N1= G90G20G54G17G0G94
$N2= empty

When it happened initially I though I may have issued a command that hosed things up, so I sent the following commands in this order:

G90G20G54G17G0G94
F20
G2X0Y0I1J0 (the start position was 0,0,0)

This yielded no movement at all, so I sent the following:

G2X2Y0I1J0 (start position still 0,0,0)

This resulted in a linear move to x=2 with no y or z movement

I then moved back to 0,0,0 and issued the following

G2X1Y1I1J0

This resulted in a linear move on a 45 degree angle to X=1, Y=1

I also tried similar commands using G3  with similar linear only results.

At this point I restarted my gui, tried again with same results. Then I used a basic serial monitor to send commands with same results. Then did a reset of GRBL with same results afterward. At this point is where I decided to reload GRBL and reset the EEPROM.  As I said, I should have just done the EEPROM reset first to see if EEPROM was the problem but in my haste I forgot to do that troubleshooting step.

109JB, 2015-08-13:
@chamnit The version date was 20150620

kfoltman, 2015-08-24:
@gerritv About threads vs serial events - I'm getting plenty of unexpected pauses in GrblPanel, probably because the sender part is waiting for the GUI to draw, and in my setup GrblPanel is running on a laptop with an Intel graphics chip. I think threads would give much better results - of course, they'd also make things enormously more complicated from programmer's point of view.

@109JB Are you planning adding buttons for zeroing a single axis? I've been in love with that feature since I first tried GrblPanel.

I've been working on my own Grbl controller (Python, Qt, Linux, open source - it's on Github), but abandoned it because apart from timing glitches, GrblPanel seems to do everything I need..

AlexHolden, 2015-08-24:
@kfoltman I'm currently working on (yet another) new sender which uses multiple threads and an asynchronous event loop for exactly that reason. And yes, it has proven pretty complicated! It's written in C/GTK+ for efficiency on low-end systems like the RPi, and I'll be releasing it as open source when it's ready for use.

109JB, 2015-08-24:
@kfoltman my gui does have the ability to zero an axis. You just click on the text of the DRO and it brings up a box to select what you want to set it for. 0.000 is the default but you can enter anything. For example, if using an edge finder you can enter 0.100.

gerritv, 2015-08-24:
@kfoltman While I don't explicitly use threads, underneath the system uses threads by its very nature. The GUI is on the main thread, the serial responses are on their own thread and I think sending is on its own thread once the send call is made. The only place where I block is reading in the file itself the first time, this only impacts when reading in huge files, as for laser engraving.
It has run happily on an Intel Atom with Intel video.

Without profiling the runtime it is not simple to determine where blockages are. This is seldom intuitive (from experience on lots of Z80 and 68000 code).
Are your movements frequent and small as in laser etching? If so then that is likely the issue as I haven't implemented the packing protocol. I send a line, wait for the OK and send the next line. E.g. you can see how well GrblPanel is feeding Grbl by watching the Buffer and Q bars. If they are usually empty then it is most likely due to small moves in the G code.

kfoltman, 2015-08-24:
@gerritv Yes, this is happening with detailed outlines consisting of small segments, so the lack of the data queuing protocol might be the reason. Unfortunately, some CAM programs can't emit the arc commands for rounded corners, which makes this kind of behaviour more painful.

Another interesting thing is that the coordinates displayed often lag behind the actual movement of the machine, even by a few seconds. Maybe the reason is that it doesn't read the whole buffer before pausing? I've tried changing the idle time, but couldn't find a good setting for this.

109JB, 2015-09-30:
Just an update of where my GUI is at right now.  I have not had a whole lot of time to work on it due to work and family stuff, but have carved out some time recently.  I am seeing light at the end of the tunnel for the initial phase of the program. A lot of the stuff for the initial phase is to set it up for the next phases.

Here is what the GUI looks like right now and I think I have finalized on the look.
[image]

Description of main window features:

[image]

The top 2 menu lines are pretty self explanatory with:
File - open or save G-code file
Connect - to GRBL
Settings - for GRBL settings, User setting, port settings
Edit tool table - brings up a tool table for tool changes (explained later)
Char Count - this is a temporary button for character counting streaming that will go away

The second line has common GRBL command buttons, along with 5 user definable buttons (Macros)

[image]

The DRO panel has work and machine coordinates. To zero an axis, you click on the axis numbers and a window comes up where you can set the desired coordinate for the present location. For example, if using an edge finder, you may want to enter -0.100" rather than entering 0. The default is 0.000.

[image]

The jog button panel has buttons for each axis along with a button to select incremental or continuous jogging. Right now only incremental works. Once GRBL V1.0 is released with continuous jog this will be fixed. The pull down box has graduated values for jogging, but you can type in that box  for whatever jog increment you want.

[image]

The lower left controls are for spindle and coolant. Pretty self explanatory.

[image]

On the right side of the main window is a tabbed box. One tab is for MDI input and Serial port monitoring. The upper MDI Input panel works very much like the MDI in LinuxCNC. The single line box is the current MDI entry box where you type and the upper box shows the MDI history. You can use up/dn arrows or click in the history box to get previous lines into the current MDI entry box.

The lower serial monitor shows what GRBL is sending back. There is a checkbox so that you can turn off the continuous stream of status reports.

[image]

The G-Code File tab is where your G-code file to run is placed. There are 2 boxes. The upper is the G-code file and the bottom shows the run progress once "run" is selected. The G-code can be edited right inside the text box.

All of the boxes are resizeable

Finally, the status strip has the current GRBL state, progress bars for the sending of the file, planner buffer state,  receive buffer state, a cycle timer, and a clock.

What is working now:
G81 - drilling cycle
G82 - drilling cycle with dwell
G83 - peck drilling cycle
G85 - Boring cycle (feed in - feed out)
G89 - Boring cycle (feed in - dwell - feed out)

The above canned cycles are fully implemented in both G90 and G91, and remember modal values, so for example, the following 2 lines of code will result in 2 drilled holes, one at x=2, y=4, and to a depth of z=-1", and the other at x=5, y=7.

G90G98G81X2Y4Z-1R0.1
X5Y7

Tool change is working using a tool table.

[image]

You enter the descriptions, and save the tool table. During program run, a M6T# command will pull the tool length from the table. It differs from NIST/LinuxCNC practice a bit. For example, all that this program needs is an M6T# command to both change the tool and set the offsets, whereas NIST/LinuxCNC requires a M6T# to change the tool, followed by a G43H# command to set the offsets. I have also made it so that you can store tool tables for multiple jobs (10 right now)

M1 optional stop support is working.

It has been a long road to get all this working. Canned cycles requires tracking of modal values, tracking of programmed position, etc. since the line the GUI is working on may be many lines behind what GRBL is working on, and the fact that the machine may be in-motion during calculations meant that I had to track all of that within the GUI rather than being able to query GRBL.  Also the GRBL equivalent movements for a canned cycle proved somewhat tedious to get programmed correctly. Think of a G91G83 peck drilling cycle. each hole can have hundreds of linear moves and if there is an L repeat value you can easily be into the thousands of commands sent to GRBL for one canned cycle line. But it's working!!!!

Future plans, if I ever get around to it are:
Cutter diameter compensation (pull out the vector math!!!)
Wizards for pocketing and drilling

I have a several minor bugs to work out and then I want to get this into the hands of a few people to do some beta testing. Don't have a timetable for that as there are a few family/work things that need to be done before I can attack those remaining minor bugs.

Cheers,

John B.

gerritv, 2015-11-21:
@kfoltman Can you send me an example file to 3d58b8e6@opayq.com?

muh67, 2016-10-12:
excellent work, please can you send VB code to    muhsabri1967@yahoo.com
thanks alot

rasciodc, 2017-02-01:
Wow, awesome work!

Any chance you´re going to share the code? If so, I´d really love a copy! rasciodc@gmail.com

Thanks!!

109JB, 2017-02-01:
Actually, since grbl doesn't support 4 axes yet, neither does my GUI. The A axis label is simply there as a place holder for when grbl does support 4 axes.  

Also, this particular rendition of my GUI is quite old and outdated. I have been working on a new one but it isn't finished yet.

rasciodc, 2017-02-01:
> Actually, since grbl doesn't support 4 axes yet, neither does my GUI. The A axis label is simply there as a place holder for when grbl does support 4 axes.
> 
I see. Actually there are a few efforts to make 4 axis versions of GRBL that are pretty much functional. If you look for Letartare and Skwee here on github you´ll find their work. Particularly, Skwee´s version works in my machine (I´ve adapted it to use on a RAMPS board and already cut some wings with it already, using the spindle commands and HeatBed output to control the nichrome wire).

> Also, this particular rendition of my GUI is quite old and outdated. I have been working on a new one but it isn't finished yet.
> 
Actually I came across your GUI when looking for ways to read and send GCode to arduino. I am working on my own 4 axis cutter software right now using VB, when I decided that no off-the-shelf software would do what I need. I started digging into that just 2 days ago...

The reason I asked for your coding is to verify ways to efficiently read a txt file, load the lines in a table (that is displayed in the GUI), store the code lines somehow (using an array or buffer) and send it over one by one to the arduino, making sure no command is missed. I´ve been reading a few forums and saw that you have increased the program´s speed by eliminating status bar and etc..so I thought it could be a good starting point for me...

It would just be a starting point though, as I am using a much much more simplified controller right now from Dan Royer ("Marginally Clever", also here in GitHub) which appears to work really well, considering that a hotwire is a really simple machine and does not need all tweaks a milling CNC does.

rcsaz, 2017-04-25:
good job
but I have a question 
I want to make a simple code reader that returns the values is wrote in text to the variables for example 
x10 y10 z40 
and after input this text the x value become equal to 10 y and z will behave the same
could you help me?about how I write the code?

lalo-uy, 2017-04-25:
You need to generate G-Code in order to control this ore any CNC.
There is more involved in the movement, like how fast to move or in which
direction to move first.
Take a look in LinuxCNC Doc about G-Code.

109JB, 2017-04-26:
> good job
> but I have a question
> I want to make a simple code reader that returns the values is wrote in text to the variables for example
> x10 y10 z40
> and after input this text the x value become equal to 10 y and z will behave the same
> could you help me?about how I write the code?

I'm not really sure what you are even asking about, but I don't have the time to help with any coding anyway.

If you just want to type coordinates and have the machine move, then as stated before you need to read about g-code. Very simply, Grbl can take commands from any serial terminal program and will move to coordinates simply by typing them, but you have to tell it how you want it done with preparatory commands

G0 means move as fast as you can
G1 Fxxx  means move at the xxx rate
G90 means move in absolute coordinates
G91 means move from the current point by the defined amounts.

so,

G90 G1 F10 will set Grbl to move in absolute coordinates at 10 units/minute
If you send this first, and then send x10y20z40, the machine will move to x10y20z40.

jinfu77, 2019-01-29:
Wow, great work!

Any chance you´re going to share the VB code? If so, I´d really love a copy! jinfu77@gmail.com

Thanks!!

randyhadi, 2019-12-12:
good job. i finished watching your video. but i see your vb just only  generate gcode. dont to send your gcode to port or machine. i really want to know how vb can send gcode to port. not use grbl to operate the machine.

MohmmedSwan, 2021-12-22:
Can you send me an example file to 
mohmmedswan2020@gmail.com

viskataboman, 2023-03-18:
Hello JB, 

can you share the VB code please? I´d really love a copy of your work. can you send me to    viskataboman@gmail.com

Thanks