grbl/grbl #664: Limit switch configuration truth table

samitray, 2015-04-21:
Trying to setup homing, with a NC(normally closed) switch.
the disable pull up and $5 settings have me all confused.
have attached a truth table for all possible combination of switch, supply combination
and corresponding $5 values and DISABLE_PULLUP compile time option

is the truth table correct, can something like this be included in the config file to make life simple.
[image]

chamnit, 2015-04-21:
@samitray : I think the $5 setting for the pull-up disabled is backward. It should be ` 1 | 0 | 0 | 1`.

samitray, 2015-04-22:
From your description above, it appears that we need to set $5 to 1 ONLY when we require normally low operation, $5 controls which state(H/L) is considered "active" and it is NOT affected by DISABLE_PULLUP compile time option. However the documentation above DISABLE_PULLUP lead me to believe that as we apply DISABLE_PULLUP the "active" state also flips, hence i applied $5 again to reverse the "active" state

closed 2015-04-23 by samitray