grbl/grbl #665: Z axis moving in the wrong direction for homing

samitray, 2015-04-21:
using version 0.9i, for homing z is moving in the negative space. note i have all direction pin inverted to make my axis form as per standard orientation. below are my configurations.

$0=20 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=7 (dir port invert mask:00000111)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=25 (homing debounce, msec)
$27=5.000 (homing pull-off, mm)
$100=1600.000 (x, step/mm)
$101=1600.000 (y, step/mm)
$102=1600.000 (z, step/mm)
$110=4000.000 (x max rate, mm/min)
$111=4000.000 (y max rate, mm/min)
$112=4000.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=800.000 (x max travel, mm)
$131=500.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

chamnit, 2015-04-21:
@samitray : Grbl sets the machine volume in all negative space by default. You can alter this behavior with a config.h option. 

Also, your max rate and steps per mm exceeds Grbl's processor capabilities. Your step frequency, as you  have them, is 106.7kHz. Grbl supports up to 30kHz. A setting of 200 steps/mm is way more than enough in terms of accuracy per step (micron level). If you don't have resonance problems, I would increase your micro stepping to reduce it to somewhere around that.

samitray, 2015-04-22:
default negative volume is perfect. Limit switches are on both ends, but 0.9v homing moves Z axis towards negative(down), even when Z homing configuration is default  positive (up). Does direction invert mask have any unintended side effects?

The homing seek and feed rates are 13.3khz and 0.66khz respectively, I will changes the the max rates to 1125 mm/min and post updated finding tomorrow. Interestingly 0.8v manages to move the axis in the correct direction during homing with similar configuration.

The motors are brused dc motors, with encoders running off a closed loop dc motor driver. given the pitch and motor + driver combination the steps / mm has to be at 1600 to get correct displacement. No microstepping here, but lots of torque at any speed, driver can support up to100Khz step rates, i am happy at 30khz ~ 18 mm/sec

samitray, 2015-04-25:
0.8 homing works sweetly. limit switch were on 5 volts rails, moved them to 12 volts, with a level converter now to see if 0.9 works on this setup.

samitray, 2015-05-01:
closing this for now, not yet tested on 0.9. Will create new question if I face a problem

closed 2015-05-01 by samitray