grbl/grbl #666: running out of i/o pins and RS485

alpha56, 2015-04-21:
I noticed how grbl is running out of I/O pins
I was wondering how are you going to add SD when that requires four more pins and those pins are already being used?

I'm looking into using RS485 as cheap long distance serial interface. I noticed there is one free pin left on the uno A4 to switch between TX and RX modes and looking at what additions are needed to serial.c to make this possible....

Anyone else interested in not having that host PC sitting right on top of the CNC?

chamnit, 2015-04-21:
@alpha56 : Grbl isn't running out I/O pins. It's completely out of I/O pins. :) 

Grbl won't be supporting SD-cards on an Arduino Uno-based system. The general Grbl philosophy is to do one thing well and not to over load it things that doesn't need. While an SD card is nice, it's not necessary.

IMO, the better solution is using an RPi to connect to Grbl, either through USB or directly through the RPi's serial I/O pins (with a $2 logic-level 3.3/5V step converter). You can have Wifi, web servers, network attached storage, and all the SD cards you want.

timryder, 2015-04-21:
@Chamnit  what is RPi.?

chamnit, 2015-04-21:
It's abbreviation for a Raspberry Pi. I've experimented with Andrew Hodel's GrblWeb last weekend and it work really really well. I'm exploring other potential GUIs that will work on it later this week.

timryder, 2015-04-21:
Sounds terribly exciting! Seemingly limitless power and hardware resources. That sounds magnificent.

Protoneer, 2015-04-21:
@chamnit I have ported GRBLWeb to a Python back-end and I am slowly including more Advanced GUI features not included in the original GRBLWeb.

One of the ideas I am playing with is enhancing/extracting the G-Code sender. By doing this a plug-in system can be built to do things that might not be CNC critical. Things like warning lights,notifications and backlash compensation. 

My code is available from https://github.com/Protoneer/Raspberry-Pi-CNC

chamnit, 2015-04-22:
@Protoneer : Awesome! I really like the simple webpage interface of GrblWeb. There isn't too much that can go wrong with it (even though it has some minor quirks). If it has webcam streaming and some additional advanced tools, I think it'd be a great and viable alternative. I'll check out your version later.

Meanwhile, I just got bCNC running on my RPi. It was surprisingly easy. The only thing that tripped me up was needing to install pip and update pyserial through that, rather than apt-get. [link](https://github.com/vlachoudis/bCNC/issues/12)

closed 2015-05-17 by chamnit