grbl/grbl #667: Travel distance different on X vs Y

SAMSOMILLER, 2015-04-22:
Tricky Question,

My Shapeoko 2 Full kit was finishing a job when all of the sudden the Y axis started traveling way faster than the X, and Z axis.  
Example, Using manual control via Easel, and Universal G Sender,
I am set to move machine .1", When I jog machine on X Axis, it moves .1", for the Y axis, it moves 1.1". 

I tried swapping the X and Y stepper motors on the GShield to test if its the motors, or the GShield. After Swapping, now the X axis moves too far.

I swapped back to normal after learning it must be the GShield. 

Has anyone seen this issue, or know how to fix it? 

FYI, i haved flashed the hex file numerous times trying to fix this problem.
Thanks ahead of time, Sam.

Norwegian-Gadgetman, 2015-04-22:
Are you certain about the scale of difference?

11x sounds a bit weird...
(8x would make more sense)
To me, it sounds like the Y-axis microstepping is messed up somehow.
Possibly, the chip has cooked or something?
(The driver ICs are NOT 'user serviceable' parts unless you have a reflow oven, so if the driver is caput, you'll need a new board)

How long was the job, and have you ever done such a long job before?

chamnit, 2015-04-22:
@SAMSOMILLER : What position does Grbl report after you move the Y-axis 0.1"? Does it say 0.1" or 1.1"? If it says 0.1", that mean Grbl is sending enough pulses to the stepper driver IC for only 0.1" worth of steps and something is likely wrong with your gShield.

If it says 1.1", it might be Grbl. How do your settings look? Were they altered somehow?

chamnit, 2015-05-07:
This issue thread is seemingly dead. Closing it.

closed 2015-05-07 by chamnit