grbl/grbl #668: New hardware

Timvrakas, 2015-04-23:
If I am designing new hardware (hopefully somewhat future proof) what are your recommendations for choice of processor? Should I still be using a 328p? is the 32u4 compatible? Or something ARM? I am an EE student, and I am planing to work on this as a project, so I want to get preliminary idea. Where does the new Shapeoko Board stand? Does it use a more powerful CPU?

Also, I'm open to general recommendations for features. Should I use on-board or off-board stepper drivers? should I use an arduino (due?), or have it all one PCB. 3 or 4 motor channels? DC spindle controller built in? What are your opinions on these things?

FIY this going to be Open Source (Of course!).

chamnit, 2015-04-23:
@Timvrakas : Nothing is future proof. Choose whatever you think will be easiest for you to get started. Grbl only supports the 328p officially. There is some 2560 Mega support but this is user-supported only and may have bugs.

closed 2015-04-23 by chamnit

Timvrakas, 2015-04-23:
Ok thanks for the help!