grbl/grbl #669: Continuous jogging enabled from GUI

109JB, 2015-04-23:
Just a note to say that I think I have found a way to implement continuous jogging from GUI's and it seems to work just fine. I am using visual basic and here is how I did it.  As an example, for x+ jogging, I have a VB timer that is dedicated for continuous jog. When the jog button is pressed and the increment set to "continuous", the program senses the KeyDown or MouseDown event and sets a flag to true and enables the timer. Within the timer the true flag causes the program to continually send "G91X0.001" to grbl until a KeyUP or MouseUP event is sensed. The flag is then set to false and the timer turned off..

I also added a scrollbar and when the user changes the location of the scrollbar it changes the jog distance sent to to grbl.(Ie: G91X0.001 is minimum, and G91X0.1 is maximum). This thereby varies the jog speed.

seems to work good in testing with an Arduino with GRBL, but not hooked to a machine. I will test it on the machine tonight. This may be a way to avoid having to spend the time to incorporate something into grbl that isn't really needed since this workaround looks like it will do the job.  

Right now I have it in a testing program that I set up that is very spartan. i plan to incorporate it into my full GUI tonight and will post a link to a video showing it in action on the gui and the machine..

winder, 2015-04-23:
Another way to do this is to send a large move on KeyDOWN and issue a GRBL feed hold (!) on KeyUP. Now just do a soft reset to clear the rest of that large move from GRBL's buffer.

109JB, 2015-04-23:
Nice.  That sounds like it may be a better way. I'll play around with it. So why is one of the tasks to push getting jogging integral to GRBL when it can be easily accomplished in the GUI?

109JB, 2015-04-23:
I thought of a reason that the method you suggested won't work  in all cases.  Soft limits. If a user has soft limits enabled, then simply sending a large move won't work. However, it can still be done by having the GUI calculate the distance from the present position to the soft limit boundary and issue a move of that distance. I'm going to work on it.

gerritv, 2015-04-23:
This is how I do it in GrblPanel as well. The repeat rate is configurable.

 

One advantage is that Grbl’s planner kicks in and smooths the moves out once a few are queued up. The Queue length indicator is useful as a visible reminder to see how far ahead you are from the present position.

 

Gerrit

gerritv, 2015-04-23:
Exactly, a bit of thought in the GUI deals with jogging adequately.

chamnit, 2015-04-23:
It's also possible to limit the amount of motions in the planner, if you are monitoring it with status reports. Limit to something like 5 motions and when a user releases the key, just let the Grbl finish out the 5 motions. Works really well and no resetting required.

bensuffolk, 2015-06-12:
Interestingly enough I had used @winder method of a large movement with eh feed hold, but before reading this I found a bug with the status which I have logged as issue 732 - https://github.com/grbl/grbl/issues/732 thats makes this somewhat more inconvenient.