grbl/grbl #670: Arduino reporting alarm when Y-axis moves

Keboose, 2015-04-24:
Just today I updated grbl on my Arduino Uno (running an X-Y table) from 0.9h to 0.9i, and for some reason the Y-axis stopped moving. The console in UniversalGCode Sender shows that Y moves about a step, and then the emergency stop is triggered (basically Y doesn't move at all, X moves just fine.) I transferred my settings between versions, so it should be running fine. I double-checked all my wiring to make sure nothing got crossed, so there must either be a bug or a setting I forgot about.

chamnit, 2015-04-24:
@Keboose : This is strange behavior that I've never heard of. Grbl should never have a scenario were one axis completely stops working while another does. Can you describe what's going on in detail a bit more? Is the emergency stop something you wired in?

Keboose, 2015-04-24:
I'm sorry, I can't really describe it better. All I saw was that after
updating, every time I try to manually move Y, the console gives the error:

"ALARM: Abort during cycle. "

I have an emergency stop button wired in, but it's not currently pressed.

Keboose, 2015-04-24:
Also, I have an issue with the version nuber continually appearing in the
console. Maybe the software keeps resetting? I hope the image attaches to
this email.

Keboose, 2015-04-24:
https://www.dropbox.com/sc/oyfsnp6dcip0rt5/AADC1O8glzB7ZuA9XiTdcSXQa

chamnit, 2015-04-24:
@Keboose : It sounds like Grbl is getting a soft-reset command intermittently, likely due to some electrical noise from your e-stop or elsewhere. Remove any input wiring from your Arduino like limit switches, e-stop, etc. Keep in mind there are thousands of Grbl users and there hasn't been a reported issue like this. 95% of the time, it's something electrical in your system that is causing the weird behavior.

Keboose, 2015-04-24:
How convenient, I reached that conclusion as I checked my email for your
message: I swapped the step and direction wires on the arduino (I don't
know why I didn't do this yesterday,) and the x axis moved just fine, which
means either a bad motor driver or, as you said, a bad emergency stop
connection. I'll keep looking into it

Keboose, 2015-04-24:
I've confirmed it's the estop button, the Y works fine with it detached.
Now I have to find out why. Just to check, the button should just short pin
A0 to ground, correct?

Keboose, 2015-04-24:
I found the problem: the wires for the Y axis ran with the estop wires for
a few inches. What I can only assume is a bit of inductive load was causing
the alarm to trigger. What's confusing me is why this wasn't a problem
until I updated grbl (the wires had been like that for several days
beforehand.)

closed 2015-05-03 by chamnit