grbl/grbl #673: Homing 'pullout' actually 'pulls-in'

buserror, 2015-04-27:
On my machine, homeing detects the limit switch, but then pulls IN (into the switch!) instead of pulling away from it.
I'm using 0.9i, fairly recent build. I only have one single inverted bit for the 'Z' homing as I've put a limit switch there, the other two axis are not inverted.

Also, it'd be nice if the firmware detected it was starting into the limit switches, as if you reset the board when it has crashed itself into the limit switch, you have to reset, then manually move the arms away from the switch, then re-enable movements.

chamnit, 2015-04-27:
@buserror : This is a common problem caused by a limit switch being set incorrectly, i.e. a NO switch behaving or mis-wired as an NC switch . v0.9i is the most current build. Invert 'Z' homing only inverts the direction for Grbl to search the homing limit switch for that axes. It does not invert the meaning of the type of switch. 

I've already started to make revisions on making it clear to the user that they have their switches misconfigured, as this problem comes up at least once a month.

closed 2015-04-27 by chamnit

buserror, 2015-04-27:
My switches are indeed wired as NC, from my research and my experience on the reprap, it seems that it's the least likely to cause noise problem. I have 0.47uF on them, only use the internal pullup, and they are reasonably free of noise...
Also, they /work/ as hard limit switches if I manually click them, so what's with the 'mis-wiring' here? It doesn't appear to be an issue with wiring, the problem is that the X axis for examples moves toward the switches (X- direction)... HIT the switch, does a couple of pull out+pull in, and when it's supposed to park itself 5mm away, it moves again in X- instead of X+ ... making a mess.

chamnit, 2015-04-27:
Make sure your Grbl invert limit pins setting is enabled (=1) for the NC switch. Grbl defaults to a NO switch connected to GND. It's also common misunderstanding that clicking the hard limit means that everything is wired and configured correctly. Grbl monitors pin changes only for hard limits, not absolute state due to bouncing. There are numerous closed threads on this and why it was decided to behave this way and why Grbl doesn't automatically enter a hard limit state when powered up or reset. Like I said before, I'm currently revising certain aspects in the code to make this problem you are having, along with many others, be obvious, while not changing too much to prevent breaking GUIs.