grbl/grbl #675: Soft limit check called without checking if homing is enabled

paulkaplan, 2015-04-27:
Should the [soft limit check](https://github.com/grbl/grbl/blob/4c20a2173fc99b6fffd4f73d93bc32dc2ed49753/grbl/motion_control.c#L41) happen only if homing is enabled and the homing sequence has been run?

In the [documentation](https://github.com/grbl/grbl/blob/46c943f13ddf42dde5a9584c21865e963cdd2f90/grbl/limits.c#L310-L311) for `limits_soft_check` it says
```
// Performs a soft limit check. Called from mc_line() only. Assumes the machine has been homed,
// the workspace volume is in all negative space, and the system is in normal operation.
```
But it looks like the soft limit check is performed in `motion_control.c:41` regardless of whether homing is enabled or run.

chamnit, 2015-04-27:
@paulkaplan : It's done at a settings level. Grbl won't allow you to enable soft-limits unless you have homing enabled.

paulkaplan, 2015-04-27:
Oh ok! I suspected it might be something like that. Thanks!

closed 2015-04-27 by paulkaplan