grbl/grbl #677: Question...

Majestad278, 2015-04-29:
i would to connect an external device to control Step an dir X, Y, and Z, but  I don't know the secure voltage.

any help?

Norwegian-Gadgetman, 2015-04-29:
Exactly WHAT are you trying to do?

Please describe what you have, how it's connected, and what you want to do with it.

Also, note that GRBL is a Software product, not HW, it just happens to be compiled to run on an Arduino UNO with a shield with some TI DRV8818 or similar driver chips.

Majestad278, 2015-04-29:
i have an arduino uno and CNC shield V3. i want to do some similar at the joy in USG, controlled with a game pad logitech wingman rumblepad, its simple the joystik control the movment in X Y Z, and the buttons control start, pause, hold, spindle start and stop, E-stop. The UNO and the pad i will connect to a Raspberry Pi 2 b+

chamnit, 2015-05-07:
@Majestad278 : Grbl doesn't have any joystick or external functionality. It only interprets g-code and executes them.

closed 2015-05-07 by chamnit