grbl/grbl #678: Homing location question

1bigpig, 2015-05-01:
Chamnit,

Why does grbl report negative coordinates when the home switches are in the bottom left (looking down on the machine work area)?  I can understand that if your machine is 300mm x 300mm and the homing switches are on the upper right, you would want your "Home position" to be 300,300.  What I cannot understand is why when the home switches are on the lower left, you would want -300,-300 to be reported at your machine coordinates.  Wouldn't that be 0,0?  You would never mill/cut/machine in a negative machine coordinates, only in local offsets/workspace coordinates.  

What was the rational for this decision?  I can understand the upper right, but not the lower left home switch location report.  While I can change it for myself, I think it is causing problems in programs like bCNC and anybody else who asks grbl what kind of machining area the machine has.

Thanks,
Bruce

Scott216, 2015-05-02:
I just asked this question on the shapeoko forum today.  It seems this is common practice because it's tradition.  
See my forum post here: http://bit.ly/1JfC153

1bigpig, 2015-05-02:
After reading the shapeoko forum response, I guess this is going to be a closed issue.  While I disagree with the reason, I guess that is why there is an option in the config.h file.  Search for HOMING_FORCE_SET_ORIGIN in config.h and uncomment the line and recompile grbl.

I am gonna close this issue.

closed 2015-05-02 by 1bigpig

chamnit, 2015-05-02:
@1bigpig : I agree. It doesn't make a lot of sense. I'd like to change it so I no longer have to answer this question, but, at this point, it'll probably screw up a lot of GUIs by having Grbl default to the force origin option. It'll have to be a slow change if it happens.