grbl/grbl #679: Homing ($H) moves only z and y Axis to Limit Switch

thomasgieser, 2015-05-03:
Homing ($H) moves only z and y Axis to Limit Switch(seek speed ). After y Limit Switch was found both (y and x) are doing Homing feed and the pull-off

chamnit, 2015-05-03:
@thomasgieser : It's a bit hard to understand what the issue is. Could you be a bit more elaborate? Thanks.

thomasgieser, 2015-05-04:
Hi, sorry... I try to explain:
The issue is that I send the homing command.
Then z moves as expected. Then only y seeks for the end switch. After it is found both, x and y are moving to the pull off position. (1mm in my setup)
This has the effect that x is traveling 1mm away every time I send $H.
Hope this explains the situation

twforeman, 2015-05-04:
Sounds like your X switch is not working as expected. Is it wired correctly? It sounds like the X switch is wired to the normally closed terminals.

thomasgieser, 2015-05-04:
I've already checked the funktion ( electrical )--> works
Btw
Sometimes it works without problems

twforeman, 2015-05-04:
If it works sometimes and doesn't work other times then you probably have a loose connection somewhere. Double check all your connections and switches for function.

thomasgieser, 2015-05-04:
Ok,
The switch is normally open. If it would be a loose connection it would travel "over" the switch but it is not starting to move

chamnit, 2015-05-05:
@thomasgieser : Grbl will only behave like this if your Y-axis switch is low(closed) upon the beginning of the homing cycle. Please check your wiring and switch.

thomasgieser, 2015-05-05:
No, it's not...
Could it be a problem with the cable length?

chamnit, 2015-05-05:
@thomasgieser : Possibly. The most common problems like yours is caused by electrical interference, like an unshielded cable or improperly buffered signal (low-pass filter). Please do a process of elimination test on your setup. Start by removing as much as you can from your Arduino, especially the limit switches. Operate them by a simple jumper wire connected to ground. You should see everything working properly.

closed 2015-05-17 by chamnit