grbl/grbl #681: Grbl nerver responds to M02 or M30

embededhell, 2015-05-04:
Hi all,
Just started playing with a clone of 'master'
Having an issue with M02 and M30 commands. When either of these are sent Grbl never responds. When using Universal G Code sender, the ok response is never seen.
When using a simple serial terminal, after entering M30 or M02, Grbl will respond to a status query '$?' but nothing else. The status response suggest Grbl is Idle. A rest has to be performed before any other commands will be accepted.

Is this the expected behavior or am I just missing something stupid.

chamnit, 2015-05-05:
@embededhell : Yes, this is a bug in Grbl v0.9i. Grbl enters a suspend state, much like a feed hold, upon a M0 pause and M2/30 program end, but reports that it's IDLE. To exit, simply send Grbl a '~' resume command, and it'll continue. 

I've been trying to think of a way to make this behavior more clear to a GUI and a user what is happening without introducing something new for all GUIs to have to support. 

- For a M0 program pause, I can have Grbl enter a "HOLD" state. That's easy. It should be compatible GUIs.

- For M2/30 program end, this is a little more tricky. When an M2/30 is issued, the machine controller has to reset by its definition (Grbl soft-resetting), but Grbl has to prevent any streaming data from a GUI. You cannot allow any g-code to be executed after an M2/30. So, it's really up to the GUI to stop streaming data and reset Grbl. I can have Grbl report a "HOLD" state, just like M0, but this doesn't tell the GUI why it's in the hold state. Perhaps, all I need to do is send back a feedback message like `[Pgm End]` to let a GUI know what's going on and that it needs to send a '~' or a `Crtl-X` to manually reset Grbl. 

I'm still trying to figure out what the best way to do this is, and I'd welcome any ideas.

embededhell, 2015-05-05:
@chamnit : Many thanks for the reply.

I would be far from qualified to make any serious suggestions, my understanding of the whole Gcode thing is limited to playing around with one machine. I can only pass on my observations.

Sending a '~' via a serial terminal works Ok. 
Sending the same via Universal Gcode Sender is not working. I would assume it is waiting after the M2/M30 for Grbl to reset and buffering the commands. 
Adding '~' to the end of the program, after the M30 yields the same result. As far as the Gcode Sender is aware, the program never completes.

chamnit, 2015-05-05:
@embededhell : I recall having that problem too. I think UGS may auto-filter out `~` characters from manual commands and the g-code program.

chamnit, 2015-05-05:
@embededhell : Try pausing the job and unpausing it. It might have to correct key bindings to let it move on.

embededhell, 2015-05-05:
@chamnit : I reverted back to 0.9g, that solved the issue for me for now. Once time permits I can go back and try out pausing, but not right now.

LETARTARE, 2015-05-05:
hello,
and mode 'Check' what happens there ?

chamnit, 2015-05-05:
@LETARTARE : It enters that same suspend mode. Good catch. I'll need to fix that too.

UlrichAM, 2015-05-05:
To prevent the M30 issue I do ignore the M30 command in my frontend software SerialComCNC. For the next release M2 will be ignored too. 
The benefit for this is to leave the zero points of axis vaild. This is important if you like to go ahead with your work on the same workpiece.

twforeman, 2015-05-05:
Or you could just change the post-processor for your CAM software to not put an M30 at the end of the program. That's what I did when I figured out what M30 was doing.

LETARTARE, 2015-05-05:
in GUI it's easy to have an option  to eliminate M2 or M30 and leave the choice to the user until the removal of the bug.

109JB, 2015-05-10:
In my GUI, I have the GUI intercept all of these and handle them internally. Easy to do in the GUI if you know about it in advance.

soundstorm, 2015-05-11:
Wouldn't it be possible if writing a bit into EEPROM when reading any of those stop commands and after executing->soft reset reading the EEPROM value and sending 'ok' then? Or maybe a full byte representing the number of line feeds and carriage returns to respond to with the right amount of 'ok'. So if you use e.g. UGS and set it to single command mode it will have the behavior as expected.

chamnit, 2015-05-11:
@soundstorm : Grbl should have one expected behavior for this, not a configurable one. That would make it harder on GUIs to know how to handle it. I've been thinking about this and here's the proposal:
- M0 Program Pause: Simply place Grbl into a HOLD state. Cycle start/resume to continue.
- M2 Program End: Place Grbl into a HOLD state. When resumed, it resets the controller.
- M30 Program End and Reset: Just reset Grbl. Don't bother with blocking out commands.

soundstorm, 2015-05-11:
@chamnit :
As the GUI waits for an (or multiple with \r\n) ok, the ok(s) must be send after the reset by M2/M30. Therefor my idea with the EEPROM.
For M0 it just works fine, pressing the button on A2.

chamnit, 2015-05-11:
The GUI doesn't need the 'ok'. For an M0 or M2, it's told that it's in a HOLD state and that should be enough to tell the user what is happening. For M30, it doesn't matter since it'll automatically reset.

soundstorm, 2015-05-11:
Okay, than @winder has to add this behavior to UGS. Don't know how other GUIs handle that, so you may be right.

ashelly, 2015-05-11:
@chamnit, Is it correct to assume that all of these will wait until the protocol buffer is empty to avoid holding/resetting while there are queued motions?

gerritv, 2015-05-11:
GrblPanel stops sending gcode when Hold is detected/returned from Grbl. There may or may not be gcode in the planner buffer on Grbl, so a Resume will continue on (in case of M0)

chamnit, 2015-05-11:
@ashelly : Yes, the planner buffer is emptied before an M0/2/30 is executed. I suppose it'd be hard for a GUI to tell if a HOLD is caused by an M0/2 or a user feed hold with a confirmation of some sort. Perhaps I can have Grbl respond with a feedback message like `[PGM PAUSE]` or `[PGM END]` when it holds. This would be before an 'ok' is sent back, by the general design of how the g-code parser executes things. Or, have it send an 'ok' for that M0/2/30 line before it goes into a hold.

gerritv, 2015-05-11:
I look for Hold in the status response, doesn't matter to GrblPanel if there was an ok or not. As soon as the Hold is detected, I go into Hold state. The source of the Hold shouldn't matter?

chamnit, 2015-05-11:
@gerritv : It shouldn't matter, but that doesn't mean other GUI implementations handle it in the same way GrblPanel does.

soundstorm, 2015-05-11:
@chamnit @gerritv : as said, Universal G-Code Sender waits to infinity if no ok comes back. So a feedback would be nice. As I'm currently working on a LCD bridge with SD-Card, it would be easier to handle the states that way, instead of parsing each command sent to Grbl and check if it's one of those mentioned.

109JB, 2015-05-11:
Here is my spin on this.

M0 and M1 (implemented in my GUI) are pause commands that need a user action to continue the run. Is anyone really using a command line to run GRBL. I doubt it and that gives 2 options:
       1. The GUI intercepts the M0 and handles it within the GUI. Easy to do and is how the program I am writing is set up
      2. GRBL pauses by going into a Hold state. This is fine, but still requires a user interaction which will still need to be handled by the GUI. 
Option 2 is obviously the best option here and would work well. It has the advantage of not relying on the GUI to do this for ultra simple programs that are essentially just streaming interfaces.

In regard to M2/M30, I don't really see where GRBL needs to soft reset. I know the logic of the M2/M30 is to reload the default modal values and a reset theoretically does that, but in reality, given a scenario where the user starts up his machine, jogs around to find part zero, maybe does some other MDI stuff, etc, some of those modal default values are changed anyway. Good G-code programming practice is to put a line or two at the beginning of the file to set the modal values to what they should be for the program. Also, having a soft reset at the BEGINNING of a program run would be more logical than at the end. In this way, the modal values are set to their defaults just before the program begins and you can be certain that everything is at default value.

As for preventing further streaming of data to GRBL after a M2/M30, that will have to be done at the GUI level anyway. If the GUI is waiting for an OK and GRBL resets when it encounters a M2/M30, the GUI will get its OK response and restart data streaming.  Even if GRBL reports something like [pgm end], which is a good idea, the GUI will have to be written to know what this response means and act accordingly. A full featured GUI should know this already from seeing it in the program.

Regarding intercepting a Hold response in the GRBL status response message so that the GUI knows there is a hold seems pointless, My program doesn't do it and what follows is why. As I see it, a hold is going to come from only a few things:
     1. First is the user (or GUI) sending a hold command to GRBL. This could be an MDI input, or a push of the Hold button on the GUI In either case the user or GUI knows that a hold is coming even before GRBL does, so why wait for a response to verify what is already known?
     2. Second is a command in the G-code program to hold/pause (M0/M1).  Again, this is a case where the GUI that is streaming the data knows before it is sent that a hold is coming. Waiting for a response from GRBL seems pointless.

Certainly, GRBL needs to act on a hold command, and it does, but how a GUI knows that a hold is issued should be from within. In my opinion anyway.

A GUI can be written to take care of all of this without changing anything in GRBL as it presently is. My GUI handles all of this internally and works fine that way. The only question is should GRBL be changed to support GUI's that aren't smart enough to somehow handle these things internally. I would say that the GUI's should be changed, but that is just my opinion.

Having said all that, if changes to GRBL are eventually made, I would vote for the following:

1. M0 pause - Just have GRBL enter a hold state "!" and wait for a resume command "~"
2. M2/M30 program end - Have GRBL send a response like "pgm end" or similar and let the GUI figure it out from there. 

Just my 2 cents.

John B.

gerritv, 2015-05-11:
Re:
Having said all that, if changes to GRBL are eventually made, I would vote for the following:

M0 pause - Just have GRBL enter a hold state "!" and wait for a resume command "~"
M2/M30 program end - Have GRBL send a response like "pgm end" or similar and let the GUI figure it out from there.

When Grbl resets a status gets sent back tot he GUI. The GUI should act on that status, e.g. Alarm and internally close up shop. UCS seems not to do some of these things (from memory a while ago) and goes catatonic when it doesn't get an 'ok' at times. One of the things that prompted me to write my own (I don't do Java)

chamnit, 2015-05-11:
The linuxcnc gcode description of M2/30 only states that the machine controller goes in the MDI mode and resets most of the gcode states. So, the assumption that Grbl needs to soft-reset isn't correct. When it needs to do is reset the parser state only. 

So here's the modified proposal:
- M0: Simply set HOLD state and resume with `~` command.
- M2/30: Send `[PGM END]` feedback message, restore machine control defaults (maybe run $N startup lines?), but don't soft-reset. It's up to the GUI to stop sending any remaining g-code in the program after an M2/30. In this state, Grbl technically IDLE and can still accept MDI commands. G92 offsets are still active.

soundstorm, 2015-05-11:
Okay, rollback: UGS act's as it should, my fault.

109JB, 2015-05-12:
Having the GUI intercept the M2/M30 allows it to not send any more data to GRBL. If the GUI waits for GRBL to tell it that there is a M2/M30 event, that is where there can be data left in the planner buffer.  

My GUI recognizes any of these commands and acts internally. No command lines are sent to GRBL once a M2/M30 are found, meaning there is no data in the buffer after an M2/M30 to worry about.

It has been commented that you can just leave out the M30/M2, which is true but is not industry practice. It is not all that uncommon to insert a M2/M30 into the middle of an existing program to do partial runs.  For example, a program that has roughing and finishing cycles. An M2 could be inserted to allow just the roughing operation. 

GRBL should have something minimal to catch an M0/M2/M30, but in my opinion, GRBL shouldn't really even see them if being used with a GUI as the GUI should see them first and act on them there.

So, I think the most recent proposal by @chamnit is perfect from a GRBL standpoint. It is minimal impact and gets the job done.

chamnit, 2015-05-12:
@109JB : Could you do me a favor since you have a LinuxCNC system running? Can you check the behavior of M2/30 on LinuxCNC with regard to parameters and parser defaults? I'm particularly curious if G92 is reset upon M2/30. Thanks!

109JB, 2015-05-12:
I'll see what I can do. I have it in a virtual machine on my laptop as well so shouldn't be a problem.

109JB, 2015-05-12:
It looks like G92 doesn't reset on a M2 or M30.  From the linuxCNC user manual, 

"LinuxCNC stores the G92 offsets and reuses them on the next run of a program. To prevent this, one can program a G92.1 (to erase them), or program a G92.2 (to remove them - they are still stored)."

So, G92 stays unless specifically erased or removed using a G92.1 or G92.2. I'm not sure the difference between 92.1 and 92.2, as functionally they appear to do the same thing.

chamnit, 2015-05-12:
@109JB : Thanks! That clears things up a lot. I saw the LinuxCNC description too and wondered why it was contrary to how I understood it. After some digging, it turned out that the NIST g-code standard document states that G92 should be canceled upon M2/30. Grbl was originally written to follow NIST, rather than LinuxCNC's standard. But at this point, Grbl is now more LinuxCNC compliant, than NIST, in a lot of respects. 

For G92.2 and G92.3, LinuxCNC stores the offsets into memory. It's separate to the running offsets. You can explicitly clear the stored offsets or fetch them. Grbl doesn't support that, and I don't expect that it will anytime soon unless there is a good reason to.

Long story short, the current proposal stays the same. I'll need to check if the spindle speed S, feed rate F, tool number T, and G43.1 tool length offset are handled upon a M2/30. I may try to get a LinuxCNC virtual machine running later, as you have, when I get the time.

109JB, 2015-05-12:
I can check those for you later and report back. I have some stuff that I have to do right now, but can check those later this afternoon.

109JB, 2015-05-13:
Here is what it looks like. I just used the LinuxCNC modal list to check. Obviously some won't apply to GRBL, but I did them all.

Motion (Group 1) G0, G1, G2, G3, G33, G38.x, G73, G76, G80, G81
G82, G83, G84, G85, G86, G87, G88, G89
RESETS TO G1

Plane selection (Group 2) G17, G18, G19, G17.1, G18.1, G19.1
RESETS TO G17

Distance Mode (Group 3) G90, G91
RESETS TO G90

Arc IJK Distance Mode (Group 4) G90.1, G91.1
DOES NOT RESET

Feed Rate Mode (Group 5) G93, G94, G95
RESETS TO G94

Units (Group 6) G20, G21
DOES NOT RESET

Cutter Diameter Compensation (Group 7) G40, G41, G42, G41.1, G42.1
RESETS TO G40

Tool Length Offset (Group 8) G43, G43.1, G49
DOES NOT RESET

Canned Cycles Return Mode (Group 10) G98, G99
DOES NOT RESET

Coordinate System (Group 12) G54, G55, G56, G57, G58, G59, G59.1, G59.2, G59.3
RESETS TO G54

Control Mode (Group 13) G61, G61.1, G64
DOES NOT RESET

Spindle Speed Mode (Group 14) G96, G97
DOES NOT RESET

Lathe Diameter Mode (Group 15) G7, G8
DOES NOT RESET

Stopping (Group 4) M0, M1, M2, M30, M60
The subject of this check

I/O on/off (Group 5) M6 Tn

Tool Change (Group 6) M6 Tn

Spindle (Group 7) M3, M4, M5
RESETS TO M5

Coolant (Group 8) (M7 M8 can both be on), M9
RESETS TO M9

Override Switches (Group 9) M48, M49
RESETS TO M48

S Spindle Speed
DOES NOT RESET

T Tool number
DOES NOT RESET

Tool length offset
DOES NOT RESET

F Feed rate
DOES NOT RESET

Hope this helps

John B.

chamnit, 2015-05-13:
@109JB : Wow! That's very thorough! Thanks for doing that. It's interesting to see what gets defaulted and what doesn't. It definitely much different to a soft-reset.

chamnit, 2015-05-16:
@109JB : I've been working on updating the program stop control and implementing what you found. The one unclear thing is 'Tool length offset DOES NOT RESET'. This should be covered by Modal g-code group 8. Do you mean the 'H' word for tool length number when applying an offset via a G43?

109JB, 2015-05-17:
Correct. I didn't really have a way to view the H parameter directly to see what it actually does, so I did the following:
G90
G49
G0X0Y0Z0   (DRO reads Z=0.0000)
G43.1Z0.500       (DRO reads Z=-0.5000)
M2                 (No change to DRO)
M30               (No change to DRO)
G49               (DRO changed back to Z=0.0000)

I also checked it by entering a tool into the tool table with length = 1 and using G43

Based on this, I reasoned that LinuxCNC retains the current tool length offset value even after an M2 or M30 command.  I can't say whether it retains the H value, or if it just remembers the length offset. It at least does the latter since there is no change in the DRO.

chamnit, 2015-05-17:
All: I just pushed a fix for M0/2/30 into the edge branch. I need some volunteers to test the changes before I pull it into the master branch. 

I've also changed how homing works to help users diagnose limit switch wiring problems by having the locate part of the homing cycle not move as much. It now seeks toward the limit, pulls off the limit switch by the pull-off setting distance, and then slowly moves into the limit switch to determine the axis home location. It's a lot cleaner in its behavoir. 

If anyone has any issues or problems with M0/2/30 or the new homing routine, please let me know! I'd like to get any problems now, rather than later! Thanks!

ashelly, 2015-05-18:
If I'm reading the new homing code right, it looks it will shift the home location away from the machine center by the amount of sensor hysteresis. This isn't a bad thing - for some sensors types the 'on' trigger point can be much more repeatable than the 'off' point. But you should probably warn users that previously calibrated locations may shift.

chamnit, 2015-05-18:
@ashelly : Thanks for the review (and for the limit state check suggestion). Both valid points. You're right though. The homing cycle now locates home based on trigger location, rather than release. By doing it this way, I was able to get rid of the pull-off motion code and save a couple hundred bytes of flash. It also made the homing behavior cleaner and visually better understood what Grbl is doing.  

I expect that for most users that this change won't be a problem. For OEMs, it might break their stored offsets. I'll wait to hear from people before merging on this issue.

aldenhart, 2015-05-18:
It’s probably more accurate and repeatable to set the homing position based on the switch close instead of the open, so it’s a good idea. Many switches have hysteresis so the onset and the offset are not the same. The onset seems to be more repeatable based on tests done by some of our users.

chamnit, 2015-05-24:
All : Ok. Implemented a limit state check to ensure the switch is cleared upon pulling off. If it doesn't, the homing will report a failure. I've tested it as much as I can and everything seems to be operating correctly. Anyone willing to put the edge branch through some testing? I'd like to get this pushed to master asap. Thanks!

chamnit, 2015-05-26:
This fix has been pushed to master. Please report any problems immediately. Thanks!

closed 2015-05-26 by chamnit