grbl/grbl #682: CO2 laser

Bill1959, 2015-05-04:
Hi 
I have a 40w K40 co2 laser engraver/cutter that uses Moshidraw and it is horrible has anybody on here done a conversion that is well documented anywhere please ??

BackEMF, 2015-05-05:
https://sites.google.com/site/3dprinterlist/lasercutters/arduino-co2-laser-project
He seems to just have started, but maybe it's an idea to co-operate?
Note that there are many different CO2 cutters that use Moshi, they all look the same on the outside, but the hardware can be very different. So if you find a well documented conversion it will most likely not be the same hardware as the machine that you have. 
https://sites.google.com/site/3dprinterlist/lasercutters/laos is an MBED based CO2 conversion
https://sites.google.com/site/3dprinterlist/lasercutters/lasersaur a DIY CO2

No full documented conversion but i hope it is of any help.

Bill1959, 2015-05-05:
Thank you BackEMF 
I will have a look at those the first one I have read a long time ago and I think he is building it from old printer parts as he worked for the printer company years ago and can modify the driver board from a printer. I am more interested in the stepper steps and so on. Thanks again.

BackEMF, 2015-05-05:
The main difference between CO2 an diode machines is the high voltage unit to start the CO2 beam. Well, and then that the mirrors are moved instead of the laser itself.  Controlling the steppers should be straightforward as on any other GRBL machine.  I think most of those Chinese CO2 lasers have 1.8 degree steppers. Just like most 3D printers. Unless it is very old. 

On the LAOS site i found this http://redmine.laoslaser.org/attachments/download/249/DSCN0285.JPG
I wonder if there isn't already someone who back engineered this with GRBL as bain. 
A PCB with the same formfactor/mount holes and same connectors as a one on one replacement.
Just like LAOS but then with an Arduino Nano instead of the MBED MCU. (Anyone?)

Bill1959, 2015-05-05:
Hi Thanks again I am also a bit bothered about the power % to the laser it is controlled by a pot and a 30mA analogue meter on the front control panel but that needs to be controlled by software so you would be able to engrave and cut on the same project without having to use a pot. I know of the one board that has been made as a direct replacement but he never replies to emails I send him. As for age of my laser it is only nearly 3 months old and has the MS10105 v4.7 board it has been said on forums that everything is controlled by the large chip that is not soldered onto the board  (but all markings ground off) so you would think a board could be made to remove that chip and plug into its socket and a mod to the laser power supply feed for power %.

truehybridx, 2015-05-07:
I am partially in the middle of wiring my K40 up to use GRBL. I am mostly waiting for the official branch to have laser and hopefully raster support.

Right now I have an arduino uno and a CNC shield that will replace that junky moshiboard. So far it will only require me to change the connectors from the stepper motors since they are too big to plug into the shield.

I have yet to determine if the uno itself can trigger the laser to fire or if a beefier transistor needs to be inline with it. Eventually I want to have the potentiometer on the laser console set the max power and have the uno control the laser power within that range.

Although I am moving at an incredibly slow speed :/

Protoneer, 2015-05-07:
What about using the new PWM spindle control the laser intensity?

abearden, 2015-05-29:
I just finished adding a K40 to my ShapeOko 2. Protoneer is right, you just need to use the PWM variable spindle pin. If you want analog control it requires adding a resistor and capacitor to your variable spindle output pin (spindle PWM->resistor->laser control->capacitor->ground). Tie the grounds together, attach your RC circuit, and off you go. I used 500 ohm and 4.7 uF and it takes about 0.01 seconds to change voltage.

I completely stripped the Moshi board out and only use the GND and LO pins from the connector between the power supply and the Moshi. But, like mentioned above, there are several versions. The problem is most K40 power supplies are active low which means your M3, M5, and S commands are all backward with stock firmware if you get one of these (and the laser starts immediately, scary!). You can get fancy with a hardware inverter, but I modified my firmware instead and added INVERT_VARIABLE_SPINDLE to config.h. Then, modified spindle_control.c to set PWM output to PWM_MAX_VALUE in spindle_stop() and "rpm = SPINDLE_MAX_RPM - rpm" in spindle_set_state() when INVERT_VARIABLE_SPINDLE is defined. Reverses all of the outputs for the appropriate spindle commands with minimal, and reversible, changes to code.

With the exception of pin fluctuation on reset it works fine. I just avoid turning the laser on until the board is booted.

polygontwist, 2015-07-27:
hello, i have convert a c02 laser to grbl you cav view my projekt on: http://www.a-d-k.de/20150611_213938-Lasercutter%2Breloaded.htm

lehuuloc-0104, 2019-12-25:
> I just finished adding a K40 to my ShapeOko 2. Protoneer is right, you just need to use the PWM variable spindle pin. If you want analog control it requires adding a resistor and capacitor to your [...]

Hi @abearden,
I am having some problems, can you help me?
I want to refer to your wiring diagram (including capacitors and resistors), can you share it with me?