grbl/grbl #683: Exec each line of G-Code

vangalvin, 2015-05-06:
My custom M-Code is finally working with the distance trigger/counter on the Y working as i was hoping and the single button Hold/Start operating perfectly HOWEVER! The M-Code is being executed as soon as it is sent to GRBL and not waiting for the previous command to finish. 

To resolve this would i be correct in thinking i need to check the STATE_IDLE and is this set after each line of the G-Code is executed and finished?

the ISR ended up like this. 
ISR(CONTROL_INT_vect) { if (!(WDTCSR & (1<<WDIE))) { WDTCSR |= (1<<WDIE); } }
ISR(WDT_vect) // Watchdog timer ISR	
{
  WDTCSR &= ~(1<<WDIE); // Disable watchdog timer.
  uint8_t pin = (CONTROL_PIN & CONTROL_MASK);
  #ifndef INVERT_CONTROL_PIN
    pin ^= CONTROL_MASK;
  #endif
  // Enter only if any CONTROL pin is detected as active.
  if (pin) {
// Set the timer overflow from when the first click was : vangalvin	  
    if (bit_istrue(pin,bit(RESET_BIT))) {
      mc_reset();
// One click should start the cycle Needs some form of Debounce
    } 
	if (bit_istrue(pin,bit(CYCLE_START_BIT)) && SUSPEND_FLAG == 1 ) {
      bit_true(sys.rt_exec_state, EXEC_FEED_HOLD);
	  SUSPEND_FLAG = 0;
	  }
//another click should pause the cycle Needs some form of Debounce
	else if (bit_istrue(pin,bit(CYCLE_START_BIT)) && SUSPEND_FLAG == 0) {
	  bit_true(sys.rt_exec_state, EXEC_CYCLE_START);
	  SUSPEND_FLAG = 1;
	  }
	if (bit_istrue(pin,bit(Y_COUNT_BIT)) && (gc_block.modal.dst_count == Y_COUNT_ENABLED)) {
	  	Y_COUNT++;
		  if(Y_COUNT == gc_state.distance_counter){
		      printPgmString(PSTR(" Y-DST"));
		      printFloat_RateValue(gc_state.distance_counter);
			  Y_COUNT=0;
			  gc_state.distance_counter = 0;
			  gc_block.modal.dst_count = Y_COUNT_DISABLE;
			  bit_true(sys.rt_exec_state, EXEC_CYCLE_START);
		  }
 	}
  }
}

I also added this for the custom M66 command to gcode.c
          case 66:
          word_bit = MODAL_GROUP_M66;
          switch(int_value) {
			  // Get an M66, Pause the machine, zero the y-count and start counting until value is reached
	          case 66: gc_block.modal.dst_count = Y_COUNT_ENABLED; 
					   gc_state.distance_counter = gc_block.values.s;
					   // need to let the last command finish before stop
					   bit_true(sys.rt_exec_state, EXEC_FEED_HOLD); 
					   break;
          }

and as i am using the S var to pass the required clicks to wait for before resuming I also added this
    if (gc_state.distance_counter != gc_block.values.s) {
	    gc_state.distance_counter = gc_block.values.s;
    }

Protoneer, 2015-05-06:
Mmmm... is it possible to set the gcode buffer to 1? (not sure if its in
the settings)

That way each command needs to finish before the next one can run...

vangalvin, 2015-05-07:
I will take a look, however it looks like I may be working in the wrong place, the additions I have made are for real-time execution and these are monitored during the normal execution of the G-Code. Isuspect i need to move it all to the non realtime enviroment. will however try what you have suggested. The Mach3 addition i wrote has a function to wait until the machine stops moving before continuing so that was easy but cant seem to see anything in this GRBL to use the same way.

ashelly, 2015-05-07:
> The M-Code is being executed as soon as it is sent to GRBL and not waiting for the previous command to finish.

In gcode.c most commands are buffered, and passed off to the planner for execution.  If you want to run  a gcode command only after the previous command has completed, you need to wait for the buffer to empty. The preferred pattern for this seems to be calling `protocol_buffer_synchronize()` before doing your action.  See for reference `mc_probe_cycle` or `coolant_run`.

> To resolve this would i be correct in thinking i need to check the STATE_IDLE and is this set after each line of the G-Code is executed and finished?

No, `STATE_IDLE` is set when the motion segment buffer is empty. If you stream multiple motion commands, it will not go to IDLE between each one.

vangalvin, 2015-05-07:
Thanks ashelly, I managed to nail it yesterday and works a treat now. :) another issue i ran in to was after the custom m-code had run it would execute the next command then stop. adding plan_cycle_reinitialize(); to the ISR routine were the counter terminates after hitting the correct count for the Y position sorted this out. now to customize the post and build the machine :) oh and i am kind of stunned by the speed of GRBL given its sitting on a nano! 2100 mm/sec for the feed rate was not too bad

closed 2015-05-07 by chamnit