grbl/grbl #684: bit_false_atomic and bit_false implementation should use cast

Andy1978, 2015-05-07:
In nuts_bolts.h line 57, 60
#define bit_false_atomic(x,mask) {uint8_t sreg = SREG; cli(); (x) &= ~(mask); SREG = sreg; }
#define bit_false(x,mask) (x) &= ~(mask)

I think we should cast to uint8_t first like "(x) &= (uint8_t)~(mask);
See avr-libc FAQ "Why does the compiler compile an 8-bit operation that uses bitwise operators into a 16-bit operation in assembly?" http://www.nongnu.org/avr-libc/user-manual/FAQ.html#faq_intpromote

chamnit, 2015-05-07:
@Andy1978 : Interesting. This is behavior confirmed in the assembly code?

Andy1978, 2015-05-07:
@chamnit : Hm, I had a look at the generated asm code. I've chosen gcode.c because there are 6 real occurrences (others are commented) and compiled with
avr-gcc -fverbose-asm -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -MMD -MP -S grbl/gcode.c

But after digging through the diff it seems that the avr-libc FAQ entry isn't true anymore and avr-gcc does a really good job to optimize this. I have to admit that the last time I tested this behaviour mentioned on the FAQ is 6years ago and since then I automatically cast to uint8 before ~.

Sorry for the noise.

chamnit, 2015-05-07:
@Andy1978 : Not a problem. Thanks for checking into this.

closed 2015-05-07 by chamnit