grbl/grbl #686: Modify G20 & G21 (units in inches) Command?

typ1, 2015-05-07:
Hi All--
I'm working on a project for which I would like to use the same gcode files but with two different scales.  I will need to change this scaling factor every couple minutes and I'm planning on having multiple instances of this machine running for years.  I came up with a bit of a hack by modifying the $100-102 settings (steps/mm) when I change scaling factors.  However, these settings are written to EEPROM, which has a limited life in regards to the number of times it can be changed.  I'm interested in modifying the G20 and G21 commands in the grbl source code to be able to toggle back and forth between two custom units instead of inches and mm.  Any ideas on how to do this?

xwalter, 2015-05-07:
hi All
I'm looking at this project but I would like to understand the theory or matematical ideas behind the function of to create the pwm for the drives .
Pratically I see that there is an interrupt handler function and then ....
How to change frequency and so the speed of stepper motor ...I would like to see some document and to study how the project is build 
Thanks
Walter

chamnit, 2015-05-07:
@typ1 : Is there an issue where you can't generate another set of g-code that has the scaled values? This will likely lead to less problems down the line. 

Changing the scale in EEPROM is ok, but then your rapid speeds and accelerations are all scaled and may stall your motors, depending on how you scale things. BTW, the EEPROM can go about 1,000,000 cycles before failure. That will take several, several years. Even if it fails early, you can always buy another DIP AVR328 for a few bucks and drop it in your arduino.

I suppose you could also change the G20/21 scaling. I think it just entails the millimeter to inch conversion scalar in nut_bolts.h. But I can't guarantee that you won't screw up something else in the code by doing this.

closed 2015-05-07 by chamnit

xwalter, 2015-05-07:
I mean I have to start from main.c and then follow the code flow ....only one interpput handler I saw in stepper , pratically where gbrl code generates the direction and the pulses for the stepper motor drives...

chamnit, 2015-05-07:
@xwalter : This thread is about altering G20/21 scaling in the code not step generation. You've posted in the wrong area, but to answer your question anyway, there is no official documentation on how everything works. It would be an enormous waste of time, because the source isn't that big and it's very well documented. Stepper.c is where all of the step and direction execution occurs in that one ISR.

xwalter, 2015-05-07:
ok thank you sir :)

csdexter, 2015-05-07:
And, for technical completeness, I will note that there is a command in G-Code (apparently non-standard, but most controller manufacturers have it) that does scaling. Grbl doesn't implement it due to (a) memory and (b) lack of a FPU constraints.

`G50` cancels scaling
`G51` enables scaling around given `X`, `Y`, `Z` or around current position if none given; by `I`, `J`, `K` on each respective axis or by `P` on all axes

Some controllers have a modified version in which scaling is always uniform (by `P` in all directions) and `I`, `J`, `K` are the scaling origin and must always be given and must always be absolute.