grbl/grbl #690: Work Position resets every new job

handpanda, 2015-05-13:
Hi!

I am using a gShield v5b running grbl 0.8c. I am experiencing strange behavior. This is what I have observed:

-The work position of my machine resets the work position Z axis to 37.084mm every time a job starts or finishes.
-The machine position Z axis resets to 0 on job completion.

__The effect is that the machine ends each job 37mm closer to the bed than when it started.__

I can mitigate this by calling G90 G28 X0 Y0 (return to zero) at the end of each program, which moves the Z into the desired position, and G92 Z0 at the beginning, which resets the work Z position. 

My questions to you are:

__1) Is this expected behavior? Are my toolpaths degenerate in some way?__

__2) Is this a grbl problem? If so, what can I do about it?__

Thanks for looking
Graham

Norwegian-Gadgetman, 2015-05-13:
Which CAM tools and which sender do you use?

Can you post the first few lines of the g-code, and maybe the last few, also?

I haven't seen this behaviour before...

109JB, 2015-05-13:
Sounds like you are using G92 to zero your axes rather than zeroing the g54-g59 work coordinate systems. my guess is that there is a reset going on somewhere in there. When you reset you lose the G92 offsets which are supposed to be "temporary" offsets. If you zero by using the G 54-G59 coordinate systems you won't lose those during a reset.

handpanda, 2015-05-13:
109JB, you are correct. Coordinate system 1 (G54) was set to (0, 0, 37.084), and grbl was setting the work coordinates to this after every job. I reset it by calling 

G10 P1 L20 X0 Y0 Z0

and the problem went away __(the coordinates do not reset to (0, 0, 0) as expected from previous behavior. They do not change at all)__ Regardless, I can move on.

closed 2015-05-13 by handpanda