grbl/grbl #692: 4th Axis questions

109JB, 2015-05-15:
I am looking forward to a time where I may want to go to 4 axes. I have been reading the closed issues regarding 4th axis implementation and it appears that it is doable un an UNO/NANO board by using the spindle speed control pins. However, I have no idea how one would go about this. The 4th axis I am thinking of would not have to do anything all that special. About the only thing different from a standard linear axis would be wrapping; (ie: For a rotary axis, to get from 350 deg to 10 degrees would be a +20 degree move rather than a -340 degree move.)  I could even live with it just acting like a linear axis and not being able to go past 360 deg. 

Also, if done, would there be a way to generate the position of the axis in status reports.  If not, no biggie as I can put that in my GUI to track.

Thanks,

John B.

chamnit, 2015-05-15:
@109JB : 4th axis isn't too hard to implement, but it'll pretty much use up all of the remaining flash and RAM in the Arduino Uno. For now, I'm reserving it for v1.0 features. FWIW, I'm working on a Grbl Pro that will include 4+ axes. It'll have a Grbl compatibility mode for GUIs written for Grbl.

109JB, 2015-05-15:
I know you are busy and have other more important items to address, but do you have any kind of estimate as to when the GRBL Pro might be available?

Thanks Again.

chamnit, 2015-05-15:
Nope. It'll be ready when it's ready.

closed 2015-05-15 by chamnit

langwadt, 2015-05-15:
do you need all 4 axis at the same time? I've done stuff on a rotary axis by locking down one axis and using that driver to run the rotary axis. Just have to write the gcode as if was "unrolled"

Norwegian-Gadgetman, 2015-05-16:
Having a 4th axis allows you to do stuff you can't easily do by locking one axis.

Imagine a set of 4 round table legs... 
Sure, you can cut those by locking one axis and running the rotary axis instead, in effect turning the mill into a lathe...

Now, you need a couple of slots in those legs...
If those slots are wider than the endmill, or not placed so that they go through the exact center of rotation, you'll need to be able to move the tool to compensate.
Same goes if the slot is widet than the endmill and you need the hole to have parallell walls.

langwadt, 2015-05-16:
sure it gets complicated but you could still do it by locking an axis, you'd just have to alternate between locking the linear and rotary axis. There is lots of stuff you can do while never moving both at the same time

chamnit, 2015-05-16:
For the most part, hobby-grade CAM programs only do automated 4th axis indexing between 3-axis operations. So it's completely feasible to have a separate controller that moves them between index position with something like [this](http://www.ebay.com/itm/CNC-Engraving-Router-Rotational-Rotary-Axis-50mm-chuck-3-jaw-4th-Axis-A-Axis-/261886530614?pt=LH_DefaultDomain_0&hash=item3cf9a75436). With a RPi-based GUI, you could even have the RPi's GPIO send out the step pulses to move the 4th axis, since it's not a synchronized motion.

Regardless, the vast majority of CNC users that use Grbl don't have access to CAM programs that generate 4th axis g-code programs. Nor do they have the proper machines to do it with. You need a mill that is very stiff to be able to take the cantilever loads and with a decent amount of z-axis travel so you can machine something of respectable size. Most hobby machines are gantry-style with about 3" of z-travel. That translates to parts only about 2" diameter.

Norwegian-Gadgetman, 2015-05-16:
If we assume that any pocket or other feature doesn't need to 'go past Z=0', there's nothing to stop a hobbyist from making a trench in his wasteboard and suspending the material in that so that the centre is at the height of the top of the wasteboard.

I currently don't have the need for a 4th axis, but if I did, I don't think it would be all that difficult to build a machine with 4 or 6" of Z-axis travel. If one assumes long Y-axis(no problems on that point), one can double up on the X-axis(2 x makerslide+20x40 assemblies) and suspend the Z-axis between those, using another pair of Makerslides for stability there.
Or take a look at Cvoinescu's MegaRail and Z-rails.

Adding a 120degree 5th axis shouldn't be too much trouble, either, I believe.
I think it's the lack of affordable 4-axis CAM tools that's the biggest problem, really.

chamnit, 2015-05-16:
The cost of additional tooling/fixturing and the cost of making the necessary modifications to a machine that isn't designed for it is just as prohibitive.

109JB, 2015-05-17:
Some things I have in mind would not allow locking an axes and running the 4th axis with the driver from the locked axis. However most of the stuff I envision would not require any type of syncronized motion of the Z axis. I won't be using a Raspberry Pi, but a PC. I suppose I could make sure that the lines containing 4th axis moves are on lines by themselves and I could write some code into my GUI to intercept them and send them to a second GRBL controller on another com port.  I'll have to think about that.