grbl/grbl #694: Spindle speed inverted?

Jibeji, 2015-05-17:
Hello,
I am using Grbl 0.9i with a CNC 3020 (cheap CNC from China) and I have wired the PWM output on D11.
It works fine, but...
At startup, the spindle works at its max speed (10000 in config.h). When I enter S10000M3 it turns a few hundred RPM, and when I enter S1000 it turns almost at max speed.
It looks inverted.

chamnit, 2015-05-17:
Grbl's variable spindle output is 0V when disabled and 5v at max speed. If your spindle controller requires it inverted then all of the spindle control methods are in spindle_control.c. You should be able to modify it to be compatible with your controller.

closed 2015-05-17 by chamnit

viewsat, 2016-03-20:
Hello,
I how to modify spindle_control.c to inverted spindle output voltage.
thank you!