grbl/grbl #696: Losing Direction on all Axes

nsiatras, 2015-05-19:
Hello,

I am using GRBL 0.9 with an Arduino Uno and Protoneer Shield.
I bought a few new A4988 and DRV8825 drivers and I noticed some strange movements on all Axes.

Description of the problem:
Lets say that I move the X Axis to one direction, LEFT for example.
Sometimes before the CNC stars moving LEFT the X Axis for a very short period it moves RIGHT and then changes direction to the correct one (LEFT).

Any ideas on why is this happening?
Thank you

Protoneer, 2015-05-19:
My initial thought would be that the direction pins are not making proper contact,but the chances of it happening on all axes are very low. (Have you checked for obvious things like badly soldered joins)

What size steppers are you running?

Norwegian-Gadgetman, 2015-05-19:
Are those drivers from China maybe?

Some of those have timing issues and you may need to mess with the timing parameters in GRBL to stop it from happening if that's the case.

nsiatras, 2015-05-19:
All pins are secured i double checked them.

Yes! They are from china.
You mean to change the step pulse usec ?

nsiatras, 2015-05-19:
I am running Nema17 steppers

nsiatras, 2015-05-20:
Can this problem happen cause of the current the motors need to work ?

Protoneer, 2015-05-20:
Pololu drivers are rated up to 2A. The Chinese versions tend to be lower.
If the steppers are rated well over the the above ratings they will not run
at there full rated torque.

Could it be that your acceleration is to fast? Are you using micro stepping?

nsiatras, 2015-05-20:
The funny thing is that I have 3 older A4988 stepper drivers that works correctly and without any problems.

I am using 40mm/sec^2 acceleration and NO microstepping.

josemariaaraujo, 2015-05-21:
Maybe it's the driver phase sequencer that has been reset when that happens, and when the phases are first energized it skips to the first sequence step, that can be behind the current position. But in this case it would only move backward a little (2 steps at most).

ashelly, 2015-05-21:
Perhaps compile with `STEP_PULSE_DELAY` enabled. Maybe there's a race condition with the direction and step pins.

nsiatras, 2015-05-21:
Can it be a heat problem ? My motors need 2Amps

MichielD99, 2015-05-26:
Same problem here!
I am also using drivers from china, the X-axis is fine, but the Y-axis doesnt use the right scale and at a random moment, it switches from left to right.

nsiatras, 2015-05-26:
Well after a small research I found the following....

The most China-made A4988 and DRV8825 drivers are made from single layer copper boards.
These drivers are fine if you are going to use them with 0.12Amps~ 0.2Amps stepper motors (even with additional cooling) but with something bigger it is just a waist of time and money.

closed 2015-07-02 by chamnit