grbl/grbl #697: ALARM: Probe fail

beckmx, 2015-05-20:
I have been trying to make the probe command to work now for several days, but at least 9 out of 10 times the probes command fails to complete and stops anywhere. I was using chilipeppr and with verbose it says:

G0Z1.5
G0X25Y0
G0Z0.5
G38.2Z-2F5
<Run,MPos:-264.000,-284.225,-30.866,WPos:25.000,4.775,1.500>
<Run,MPos:-264.000,-289.000,-30.934,WPos:25.000,0.000,1.431>
ALARM: Probe fail
<Alarm,MPos:-264.000,-289.000,-31.866,WPos:25.000,0.000,0.500>
The work coordinates are ok but the machine coordinates look diferent so I am guessing it can be an error related to that.

Greetings!

chamnit, 2015-05-20:
@beckmx : Grbl moves based on work coordinates, so if it's failing when it reaches z=-2, then it's operating correctly. Grbl also supports G38.3, G38.4, and G38.5. See LinuxCNC's gcode descriptions to understand what they do.

closed 2015-05-20 by chamnit

beckmx, 2015-05-20:
Hello, it doesnt stop there, it stops before making the probe, just when it is about to start it

chamnit, 2015-05-20:
Grbl checks if the probe has been triggered before it runs the probing cycle. It sounds like either it's triggered or electrical noise is causing it be detected as triggered.

beckmx, 2015-05-20:
Yes, you were right it was triggering just a few milis it started probing,
great!

cesarrmota, 2019-02-09:
Hi, i am experiencing an Alarm: Probe fail. I have a milling machine and I use to do some PCBs... So from last week until today my probe is failing and I don't know why. When I hit the button: Probe, it comes the probe fail, the motors don't have even a chance to move and seek the PCB... BTW, I don't know the versions of GRBL and bCNC. I need some help ASAP.

Thanks

109JB, 2019-02-10:
It isn't likely to be Grbl or bCNC. The probing routines have been a part of Grbl for some time now and any recent probe problems are likely setup, wiring, the probe itself, etc.  Some things you can try.

If it doesn't even start to move when you invoke a probe routine then Grbl is likely thinking the probe is triggered before you even start.

1. See if $6 somehow got changed.  $6=0 is for a probe that is normally open and closes on contact. $6=1 is for the opposite.

2. Check your probe and wiring for shorts, or dirty contacts, etc.

3. Noise could be an issue. Route the probe wires away from power wires, stepper wires, etc.

Look at a status report BEFORE probing to see if there is a Pn:P in the status line.

Other than that I can't think of anything we wouldn't need more information for.

cesarrmota, 2019-02-10:
Thanks for replying, but it was working before with all wirings software configs. I don't know if happened a short at arduino or shield, so i'll try to replace and then a come back here.