grbl/grbl #705: Y axis has more steps/mm than the X axis

raymonkroeze, 2015-05-24:
hello,
i have a problem with my scratch build CNC machine. I've calibrated my cnc machine very precise. i have more steps/mm on the Y axis then the X axis. I use 16 mm trapezium thread with anti backlash nuts. also the angles of my machine are 90 degrees. I really don't understand why this is..
I have a problem with circles. the circles are not perfectly round. I've tried various settings in grbl but stil can't make perfect circles. The only thing i can think of is the steps/mm.

have anyone any suggestions?

Norwegian-Gadgetman, 2015-05-24:
You don't describe your machine, but I will assume that it's a ShapeOko or similar enough...

You have two rails along the edges, lets call them the Y-axis rails.
The X-axis is a gantry suspended between those Y-axis rails, and the Z-axis is mounted to a wagon of some sort that's moving along the X-axis.
And there's where you'll most likely find your problem. The entire Z-axis will tend to pivot around the X-axis whenever it's traversing in the +/- Y direction. 

To test, install a pen or something, measure it's prefectly vertical, lower it until it touches the board, then movit in the Y+ direction. Measure verticality...

raymonkroeze, 2015-05-24:
my machine is 1 meter by 1 meter. i use igus dryline as guiding rails. it is a very strong and tight system. but is does not glide as smooth as regular bearings. this is my second cnc machine. with my first machine i had the same problem with the same axis. i alway blamed the backlash. but with my new machine i don't have backlash. if tested a lot of things. i installed different drivers. different microstepping. different shaft couplers. it should work just perfect but is doesn't. if i design a square then the result is a square! just perfect. the only thing it can't do is making perfect circles. and the only thing i can think of now is the steps/mm...

Norwegian-Gadgetman, 2015-05-24:
If the axis is pivoting, it can still do perfect squares. 

The fact that you have different number of steps/mm for X and Y tells me that this is probably the issue.

raymonkroeze, 2015-05-24:
i understand what you mean. but i does not pivot at all! the system is very very tight! the only thing that is not perfectly straight is the leadscrew. it has a little wobble in the middle. 
can it be that the y carrier is much harder to move than the X axis? the motors do not stall, but if you move by hand you can feel a bigger resistance.

chamnit, 2015-05-24:
Please describe how your machine isn't making perfect circles, and how you are squaring and verifying your machine. Otherwise it's very hard to help you identify the problem.

raymonkroeze, 2015-05-25:
i will make a video today of the problem so i can explain better.

raymonkroeze, 2015-06-06:
i've re checked my cnc today. the Y axes does indeed pivot a bit. thank you for your advice. i want to make a lot off changes to the machine so i decided to build a new one from scratch.

chamnit, 2015-06-07:
@raymonkroeze. Great to hear you isolated the problem. Thanks for reporting.

closed 2015-06-07 by chamnit