grbl/grbl #709: Change from "Queue" to "Hold"

Tom-Nelson, 2015-05-27:
I had implemented a pseudo-jog control by using a joystick  to issue a G0 when toggled, and then issue a "!" when released. Then on receipt of a "Queue" status, immediately issued a ctrl-X to reset.
This worked fine on version g, and on upgrading to version i, I looked for a "Hold" instead of "Queue". 
The problem I now find is that if I don't delay at least about 500mS between reading a "Hold" and issuing a ctrl-X, it goes into "Alarm". This just slows down the whole jogging functionality and makes it less usable. Any thoughts on how I could get back to the sort of functionality that I had before (apart from going back to ver g)

chamnit, 2015-05-27:
@Tom-Nelson : This was a known issue when deliberating about the removal of the QUEUE state. The benefits outweighed the negatives, mainly due to greatly simplifying the state machine. It made it infinitely easier to manage and to add new functionality. This is the way Grbl will work going forward, and you'll need to change the way you jog. There is a little bit of a write-up on this in the interfacing with grbl wiki page.

The 500ms delay is from an alarm delay to clear out the serial write buffers. To prevent it from going into an alarm, you need to make sure the hold is complete before resetting.

closed 2015-05-27 by chamnit

Tom-Nelson, 2015-05-27:
Thanks for the feedback. I'll certainly look at the way I jog and see if
some of the alternative techniques can provide an overall better jogging
experience. The technique of issuing a continuous stream of very short
moves is something I'll definitely try.