grbl/grbl #713: How does the "Safety Door" features work?

Protoneer, 2015-05-27:
I am looking at adding this to my laser cutter and was just wondering how to hook it up and what config changes I need to make.

From what I can see its connected to the "Feed Hold" pin,right? 

Do I just use a Normally Closed switch with the door breaking the circuit once its closed?

chamnit, 2015-05-27:
@Protoneer : Yep. You need to enable it in config.h and it'll use the feed hold pin. It assumes normally-open switches connected to ground, but you can use normally-closed connected to ground with the INVERT_CONTROL_PIN compile option.

The safety door will disable the spindle and coolant outputs and will only allow you to resume via '~' once the switch is closed. It'll repower the spindle and coolant after a configurable dwell before resuming the program.

Protoneer, 2015-05-27:
@chamnit Thanks... I will give it a go.

closed 2015-05-27 by Protoneer

StrictlyTech, 2018-10-09:
Where can I find the INVERT_CONTROL_PIN compile option?  I need to have a normally-closed connected to ground for my safety door switch.   I have uncommented the #define ENABLE_SAFETY_DOOR_INPUT_PIN from the config.h file.  I have also changed (1<<safety_door_bit)  to (0<<safety_door_bit) in the cpu_map_atmega328p.h file.   What else do I need to change to make the door pin normally-closed?  And how?

I have flapping doors instead of a sliding door, which is why my door switch will be closed when the doors are closed.

whitetd, 2018-10-09:
does the door close the switch? to grd the circuit? that would be a
normally open. closing the door activates it to close ( not the normal
stand by state)

jahnj0584, 2018-10-09:
> I would like to use jog command when the door is open. i.e. when using a
> feeler gauge to set the z zero before starting a program. With the door
> open now and the Safety Door feature enabled I can't jog without closing
> the door and this prevents me from usinf a feeler gauge to set the z zero.

It SHOULD only be active when a job is sent. Not sure how GRBL a handling
it now

StrictlyTech, 2018-10-09:
I am using GRBL version 0.9j and UGS.

yatyas72, 2018-10-09:
That's what we thought also but not what is happening with GRBL. Is there a quick fix to enable the Safety Door feature but allow jogging for set-up?  I understand that when the program is running it shouldn't allow jogging but this would be for set-up.
Also a good safety feature would be for the user to have to send the resume command before the Spindle and Axis started working. Having it start automatically could be dangerous in my opinion.