grbl/grbl #715: Mega 2560 question

109JB, 2015-05-28:
I'm thinking about buying an Arduino Mega 2560. Is it simply a matter of changing the following line in the config.h file

#define CPU_MAP_ATMEGA328P // Arduino Uno CPU

to this

#define CPU_MAP_ATMEGA2560

to be able to compile and run GRBL on the Arduino Mega 2560. I tried searching in the closed issues but couldn't find a definitive answer.

Thanks,

John B.

chamnit, 2015-05-28:
@109JB : That should be all you need to do. Just be aware, I don't check for bugs with the Mega build, so there might be a few lingering. Also, it's not any faster than the Uno. It only gives you more pins and memory.

closed 2015-05-28 by chamnit

109JB, 2015-05-28:
Understood.  I am looking toward the future when I try to tackle some other things I would like to do with GRBL. Thanks again.

maaadbob, 2015-05-28:
Hi John,

GRBL does run on the 2560.
You'll need to pay attention to CPU-map.h to configure your hardware pins correctly.

I'm using grbl as a motion controller for my open hardware smt pick and place machine. (OpenPnP)

Br,
Bob.

Connected by MOTOBLUR™

https://github.com/grbl/grbl/issues/715

ashelly, 2015-05-28:
To compile for Mega, you also need to change the Makefile:   
`DEVICE     ?= atmega328p` becomes ` DEVICE     ?= atmega2560`

109JB, 2015-05-28:
@ashelly 
I use the Arduino IDE to upload GRBL to the Arduino. Wouldn't what you are taking about just be a matter of selecting the right board from the pulldown in the IDE?

chamnit, 2015-05-28:
@109JB : If you upload through the Arduino IDE, you don't have to change the Makefile.

ashelly, 2015-05-28:
Oh, I haven't tried the IDE.  Listen to @chamnit, he's the expert :)