grbl/grbl #716: Just a thought/suggestion

109JB, 2015-05-28:
This isn't  a problem, but rather a suggestion for a possible change.  When a user wishes to change units from mm to inches, or the other way around, they have to accomplish 2 things. First a switch in G20/G21, and then changing the value of $13 to reflect.  This seems redundant to me and I personally can't think of a time when I would have mm for one and inches for the other. My proposal would be to have the status report reflect the G20/G21 state and do away with the $13 value altogether. There may be a reason that this is set up the way it is. That I don't know, and this is just a suggestion. 

Thanks for all the great work on GRBL!!!

John B.

chamnit, 2015-05-28:
@109JB : It does seem a bit redundant doesn't it. The thought was that it lessens what a GUI is responsible for, so that it doesn't have to track what units mode Grbl is in.

ashelly, 2015-05-28:
I'm not a machinist, but I'd think that $13 would be fixed for the machine - you normally always want to see your positions in the units you are accustomed to. (And interface software may want units always reported the same way.)  But G20/G21 is for the file. Even if I like to see inches, I should be able to cut a pattern specified in millimeters.  But that's just a guess.

109JB, 2015-05-28:
I worked for several years as a machinist on vertical machining centers, and to tell the truth, I have no idea how they were set up.  In the time that I worked on them I only ever ran parts in G20 and the machine displayed inches.

My perspective is that if I am running a G20 part and I want to verify something, I can simply look at the DRO display and the code that is running and verify that things are going as they should. If the units on the display are different from the program, now I can't compare the two without doing a conversion. 

Maybe an option would be to have $13 have 3 values 0=mm, 1=inches, 2=match G20/G21.

Probably not worth it though. Even though it has to write to EEPROM, it would literally take decades to wear it out with just that.

cbaugher, 2015-06-03:
I'll add my current experience as a mechanical designer/machinist.  I work on mainly two machines, a Haas mill and an old 2-axis Bridgeport.  The Haas gets used with inch units exclusively because I program it with CAM software, always in inch units.  Even for metric parts I use inch units because the tool paths are programmed from the part geometry, meaning that the original units are mostly irrelevant.  I could just as easily use mm for everything, but my tools are mostly in inches so I use inches.

The Bridgeport I use mostly manually, with the occasional simple program to do drilling or facing.  When programmed with CAM I again always use inches.  But for manual use, I do sometimes switch the machine to mm if I'm working from metric drawings or a metric model.  Of course in that case G20 doesn't apply.

The interesting thing is that on the Haas G20/G21 is really just a check to make sure the machine is set to use the same units as the program.  If you try to use G21 with a machine set to inches it will alarm.  I'm not sure what the Bridgeport will do.  I should check and see.

I guess the bottom line is that it doesn't make any sense to allow the program units to be different from the machine units.

C|

winder, 2015-06-03:
I don't think it would make sense to have GRBL switch the units used in status updates based on the current state. The main reason is because the current implementation allows switching between units on the fly over the course of a program, and if someone were to do that during their program there would be a brief period after sending the G20/G21 where it would be impossible for a GUI to know what units a status is using.

109JB, 2015-06-03:
@winder In the case of GRBL status reports this is not true. It is very easy to tell what units are used in a status report because inches are 4 digits after the decimal point and mm are 3 digits.

chamnit, 2015-06-03:
@winder : That's partially true. A GUI would know what kind of unit it is, based on the G20/21 in the g-code state report. There are two problems though:
- G93 inverse time uses the F word, which is required for every motion executed when in this mode. It has units of 1/sec. It's possible to convert the move to mm/min internally, but it requires extra math to do so. This lends to keeping F unit less.
- When a G20/21 changes from inches to mm or vice versa, what happens to the F value? Does an active F200(mm/min) still mean mm/min or inch/min when it goes from G21 to G20? This lends to keeping F with units.

Does anyone know the behavior of the latter?

csdexter, 2015-06-03:
Once again, if my memory serves me right, The Book explicitly warns against playing with G20/G21 specifically because the machine is dump and will just use the raw value without "thinking" about it.

In plain English, if the last F was "20" and we're in G21, that will be interpreted as "20 mm/min". If you then issue a G20 and continue sending movement commands, the machine will simply take that to mean "20 inches/min". The same goes for all registers (i.e. including the last value of the axis words), for the WCSes, for the tool diameter/length data in the tool table and so on.
This is the reason why changing units mid-program is apparently taboo in the CNC world.

Having said all that, The Book is concerned with dinosaurs by Fanuc & the gang; it is entirely possible that a modern system like Grbl could gracefully handle mid-program G20/G21 transitions but the million dollar question is: do you really want to encourage people to mess with the units mid-program? :)

chamnit, 2015-06-03:
@csdexter : Now that you mention it, I do recall asking my machinist friend about this behavior once. I think you're right. The F word is unit less and the machine will take that value to whatever units mode the machine is in. G20 or G21. It doesn't matter. There are a lot of things that don't make sense in the g-code canon, and this is one of them. 

Either way, if Grbl handles the values gracefully or as a "dumb" machine, you'll get questions on why it doesn't do it one way vs the other. Lose, lose situation. The only tie breaker is the inverse time handling.

winder, 2015-06-03:
@109JB I didn't realize that. If this is a stable behavior it would simplify the GUI side even more. Previously I always figured you needed to query the $13 flag to be sure what units you were getting.

@chamnit My concern was more that it means the application would need to do more than just periodically sending a '?' to get the current coordinates. In this case it would now also need to inject a $G to make sure the state has changed before it can resume status requests. If the coordinate precision can be relied on like @109JB suggested then none of this really matters since the units are implicitly reported.

chamnit, 2015-06-03:
@winder : It's true that the number of digits after the '.' can be used to track the type of units, based on if the data is a position, rate, or settings value. But, it's configurable in the config.h. So someone could easily alter this themselves and break a GUI that relies on this. However, I don't know of any reason why anyone would need to change it, since you can get a machine to be much more accurate than 0.0001" or 0.001mm.

chamnit, 2015-06-03:
@winder : I was also thinking about having Grbl automatically send g-code state updates back to a GUI, when one has changed. It might be the whole string or just the one that has changed. I'm still thinking about it at this point. It'll be in Grbl v1.0. As usual, I'll setup a development branch for GUI writers like you to review and make suggestions or request changes.

109JB, 2015-06-03:
I can comment on how LinuxCNC has it set up.  As an example, take the following code snippet that I just ran through LinuxCNC on my machine:

G90 G20 
G0 X0 Y0 Z0
F10
G1 X10      <<<This 10 in. move took 1 minute as expected and the modal readout was F10
G0 X0 
G21            <<<Here the modal readout changed to F254
G1 X254   <<<this 254 mm move also took 1 minute

Therefore, it is apparent that LinuxCNC maintains the same velocity, and converts it when G20/G21 is changed.

chamnit, 2015-06-03:
@109JB : What does it say during a G93 inverse time move? (This is important for A-axis support)

109JB, 2015-06-03:
I'm not sure how useful automatic state updates would be. The reason I say this is because of buffering. By the time a state change is sent back, the GUI will be many lines ahead of where GRBL is at. This is just like the problem I am having with my GUI needing to track everything since it is operating ahead of GRBL.

109JB, 2015-06-03:
@chamnit 

In linuxCNC, when operating in G93, every line that is not a rapid move requires a F word on the line.

chamnit, 2015-06-03:
@109JB : Oh forgot to say.. thanks for taking the time to check the G20/21 behavior. It seems that I'll need to change the $G printout to be based on G20/21 state, rather than reporting state. I'll keep the $13 reporting units setting as is for the real-time status reports. This should only affect the real-time machine and work positions and real-time rate, if it's enabled.

With the automatic g-code state feedback, an updated state would only be sent back immediately after parsing a line that changed it. Any gcode sent to Grbl from that point would be subject to those states, as it would normally. This should be independent of what Grbl is doing and what it has buffered. I can't think of a case where you would need to have the parser state and the machine execution synced. Do you have an example of when it would be important?

As for G93, yes, an F word is required for every non-rapid move in inverse time mode. That's supported in Grbl. I'll check what the readout says during the move when I get home. It's either say the inverse time value or something else. I'm not sure which and what it'll say after, since it should be expunged.

cbaugher, 2015-06-03:
From a practical standpoint I can't see any good reason for allowing unit changes in mid program.  That's not to say it shouldn't be allowed, but it feels a bit like walking through a mine field without any clear benefit for doing so.

C|

109JB, 2015-06-03:
Oh.  The modal window in linuxCNC reports exactly the F value sent when in G93. For example a F4 is reported as F4.

On the state feedback, I guess maybe I'm confused. My thinking is that the automatic state feedback would be so that the GUI can do something useful with it. In my GUI I am working on for full canned cycle implementation, there are many state changes that need to be known. For example, a change from G54 to G55 needs to apply the work offset changes immediately before even proceeding to the next line. If the GUI were to wait for GRBL to report it, then the character counting streaming won't work at all, and waiting for the response will slow down the GUI. For this reason, a state change from GRBL is of limited value. It may prove useful for things like MDI where things happen at a slow human pace, but while running a program, it doesn't seem like there would be much benefit.

109JB, 2015-06-03:
@cbaugher 

I think you are missing the point. While I agree that changing G20/G21 within a program isn't likely or even recommended, GRBL doesn't run the program, so it doesn't know where one program ends and another begins. It is simply running a stream of G-code lines that could be in a program, from MDI, or any combination.  So, the user expectation of what a command is to do is important, and I think what we have found is that there is a difference between GRBL and LinuxCNC. Not to say one is absolutely right, but GRBL is now more following how LinuxCNC operates, it being a more mature program.

cbaugher, 2015-06-03:
@109JB 

Yeah, I was vaguely thinking of something along those lines after I posted.  The part I was missing is that grbl doesn't track whether it's inside of a program or not (I'm still coming up to speed here), in which case it does make sense for grbl to support G20/21 unit switching.  It might be bad practice at a higher level but that's not for grbl to enforce.

C|

chamnit, 2015-06-04:
@109JB : I've recently had thought about what commands that a GUI has to wait for Grbl to send something back in real-time. I've been working with the bCNC developer to get "macros" installed. He's done an absolutely amazing job and has already gotten a working test build in a dev branch on his repo. The idea was to install macros so people can insert canned cycles, tool changes, or custom (auto leveling)probing cycles easily with a high-level pseudo code. It's not quite Fanuc Macro B, but it's an easier-to-implement Python-based variant that takes queues from it. So, during our discussions, yhis exact question came up.

The only thing I could think of was the probing data. Everything else can be queried early on and managed with the streaming g-code. All work coordinate offsets can be done in this manner on the GUI side, if needed. Everything else should be able to be streamed to Grbl without having to wait.

Going back to the original topic, have we decided to keep $13 as is? Only the $G gcode state string will be updated to show F values based on G20/21 and G93. 

Is there an easier way for GUIs to know which units the real-time status reports are showing without needing to look at and parse the $$ settings? The number of digits is a decent solution, although not robust. Do I need to have Grbl send something in the welcome string(s) like a handshake to setup GUIs?

gerritv, 2015-06-04:
Personally I think the GUI’s should find it easy to watch F and G settings going out. And display them.

At start up it is reasonable to expect a GUI to do $$ to get the current settings from Grbl, how else can you know what it is configured to.

 

Gerrit/GrblPanel

109JB, 2015-06-04:
Personally I think that the current status reports with 3 digits for mm and 4 digits for inch, along with the ability to do a $g is fine for the GUI to know what is happening. Like @gerritv mentioned it is quite easy to pull all of the GRBL reports at start up to let the GUI know what is going on. That is how I have my GUI set up.  IN mine where it becomes very complicated is the tracking of all of the things required for a full implementation of "things".  For setting up canned cycles without "full" implementation is rather easy after having gone though some of this. For example a G81 is easy IF you require all of the G81 information on any G81 line, essentially making G81 non-modal. Where it gets complicated is having to track coordinates and modal information. For example, a snippet like this one:

N001 G90G0X0Y0Z2
N002 G91
N003 G98G81R0.1Z-1
N004 X1
N005 X2

In line N003, since a G98 is used, it requires knowing what the previous Z position is from line N001, in this case Z=2, and both G98 and G81 aren't GRBL supported so the GUI needs to track those too. Also the N004 and N005 lines are also G81 since a G80, or G1-G3 hasn't been issued. That means all of the G81 info on line N003 needs to be tracked until the canned cycle is cancelled. 

Now on top of that, the Z position in line N001 is OK since it is a G90 move, but if it were a G91, we really need to know the machine position and not just the last Z move. Throw on top of this if there is a change from say G54 to G55 and it gets really involved.  There are other things too that need tracking. This is what I have been racking my brain over on my GUI.  In some cases, it would work to just wait for GRBL before proceeding on a canned cycle. For instance, you could wait for GRBL to become idle, read the status report for the Z position and proceed. This could work for the position information, but you still have to track the modal info for the canned cycle unless, as I said before, you just do a partial implementation and require the user to re-enter all of the info making the canned cycles non-modal. I have thought about doing just that, but am still trying to track the position and all the other information, which I am able to do at least mostly right now.  The way I have it set up is to track the machine position and apply the current work offset as needed while also tracking the modal info. It is working that way now with a few things that need to be added. The problems I have to work out right now are what happens when a user does a G20/G21 switch, a user reset, or re-zeroing of work positions. These need to be captured either by the GUI doing conversions to the arrays used for tracking or by re-querying grbl. 

This is something that maybe someone out there with Visual studio (Visual Basic) experience can help with.  I have a routine to detect a reset command sent to grbl. If detected, all of the reports from grbl are requested to re-initialize the GUI. The problem I have is that the GUI detects the reset, and sends the $ requests before GRBL has a chance to reset itself, so the requests fail.  I have tried several methods to get the GUI to wait, but no success. If I just wait and have a button to reset it works fine. if anyone has an idea, let me know.

As for the $G reporting of the F value.  The reason I brought it up is because in my GUI, if G20 is current it expects a in/min value for F. I have written into my code a workaround for this in that the GUI looks at the current G20/G21 state and sets $13 to match. This way the $G report of F value always matches what I expect.  It was just something that I thought was contrary to the norm so I thought I would bring it up.

Thanks,

John B.