grbl/grbl #717: Headless interface

vince87, 2015-05-29:
Hi, I created a small sketch that allows grbl to function without PC , you need :
2 arduino ( 1 for GRBL, 1for control )
1x LCD shield
1x SD shield

https://codebender.cc/sketch:119423

then I include photos and video

vince87, 2015-06-01:
[image]

Here a modified sketch that work with a reprap smart controller:
https://codebender.cc/sketch:119503

freegroup, 2015-07-22:
Hi

this is greate!

Why? Because my USB connection get lost after 1/2 hour or so. The reason
for this are heat or the controller/driver, milling motor, electric interferences, ....
and a lot of other stuff.

And I didn't want migrate to a PC based solution with LinuxCNC or something similar.
I use your code and hopefully I get it up and running with my limited know how :-)

...did you have an improved version of your code?

Best Regards

Andreas

vince87, 2015-07-24:
Hi, thanks for your feedback
I'm sorry but I have no other solutions, because of risk of saturating the RAM of the Arduino uno.
when I have some time I'll try something better
Thanks
Vincenzo

dragoblaztr, 2017-05-15:
@vince87 

What are the functions of pins 2 and 3 in the first sketch? I can't figure it out...

I made a few modifications to the code to work with a joystick and a i2c 20x4 lcd, and it works.

vince87, 2017-05-15:
Hi, You're right, pin 2 and 3 are not used ... can you share your modified
sketch?
Thanks
Vincenzo

dragoblaztr, 2017-05-15:
Here is it

```
// GRBL without pc use!
// Shield used: datalogger (for sd), lcd shield
// free pins: A3, 2, 3

#include <Wire.h>
#include <SD.h>
//#include <LiquidCrystal.h>
#include <LiquidCrystal_I2C.h>
//#include <elapsedMillis.h>

String gcd;
char* myfile;

const int chipSelect = 10;
#define _LCDML_CONTROL_analog_pinx A0
#define _LCDML_CONTROL_analog_piny A1
#define _LCDML_CONTROL_digitalread 8 
#define _LCDML_DISP_cfg_button_press_time          200    // button press time in ms
unsigned long g_LCDML_DISP_press_time = 0UL;

String RxString;

int lcd_key     = 0;
int adc_key_in  = 0;

#define btnRIGHT  0
#define btnUP     1
#define btnDOWN   2
#define btnLEFT   3
#define btnSELECT 4
#define btnNONE   5

//String words[6] = {"Attesa GRBL...", "Errore SD", "SD OK!!!", "Errore!!!", "Stop!!!", "Fine invio!!!"};  //Italiano
String words[6] = {"Waiting GRBL ...", "SD Error", "SD OK !!!", "Error !!!", "Stop !!!", "End Send !!!"};

//LiquidCrystal lcd(8, 9, 4, 5, 6, 7);
LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address

void setup()
{
	pinMode(2, INPUT);
	pinMode(3, OUTPUT);
	digitalWrite(2, LOW);
	digitalWrite(3, LOW);
  pinMode (_LCDML_CONTROL_digitalread, INPUT_PULLUP);
	Serial.begin(115200);
	while (!Serial)
	{
		;
	}

	lcd.begin(20, 4);
	lcd.clear();
	lcd.setCursor(0, 0);
	lcd.print(words[0]);
	lcd.setCursor(0, 1);

	//pinMode(10, OUTPUT);
	if (!SD.begin(chipSelect))
	{
		lcd.print(words[1]);
		return;
	}
	
	lcd.print(words[2]);

	//Awake GRBL:
[... 245 lines cut ...]

		leggisd();
	}
	
}

/*
  void wait(unsigned long alfa){
  elapsedMillis waop;
  while(waop < alfa){
    }
  }

*/

```

Actually I'm doing a more complex interface using this library

https://github.com/Jomelo/LCDMenuLib
](url)

PD I'm working on show the cordinates on 3 axis at the same time to execute the gcode, marlin style

Regards

vMeph, 2017-05-15:
does this work for grbl 1.1?

dragoblaztr, 2017-05-15:
@vMeph 
Yes it does, but special characters are no showing properly on the screen on startup, I tried on grbl 1.1 with a +17000 lines g file and no crashes.

vMeph, 2017-05-16:
If that could be implement with this
https://github.com/grbl/grbl/issues/1125 
by using second arduino
Would rox specialy with those lcds with encoders like the ones used in marlin 3d printer software

vince87, 2017-05-16:
@dragoblaztr 
It's fantastic, when is done you can tell us?, the idea of ​​the menu is great, but in my opinion a more complex hardware is needed .. (32u4? M0?), So you also have the option of a PC debugging and a more complex gcode to stream (10/20MB now it freeze with 5/6MB ).
Regards

dragoblaztr, 2017-05-16:
@vince87, @vMeph

That's the idea, actually I'm using a Mega to run the tests and I'm planning to run with it, not with uno. Besides in a future version using a tft LCD instead of I2c , my idea is make it like a cnc panel for jogging, read and edit cnc code, and streaming without PC (using encoders, joysticks, keypads and so on) any effort is well done received.

tklus, 2017-05-16:
This will be awesome when it is all worked out! Thanks guys for all the hard work!

109JB, 2017-05-16:
It needs a way to zero work coordinates.

I made one several years ago using a SD/ touch screen LCD on a Mega 2560. It worked but was way too slow for my taste.  I had implemented jogging, zeroing, running, etc., but was too much work for too little payback.  I went back to using my netbook.  Here are a few pictures of what I had done.  It had the DRO for 3 axes and was written for Grbl 0.9 The pictures don't show the DRO because this was taken with just the host arduino, not hooked to a Grbl enabled Arduino.
[image]
[image]
[image]
[image]
[image]
[image]

The .ino file is in this text document if anyone wants to play with it.  I don't think touch screen is the way to go, and the TFT may be too much as well.  I think a monochrome LCD like you are using would be best to maintain speed.

[GRBL_Head.txt](https://github.com/grbl/grbl/files/1005198/GRBL_Head.txt)

winder, 2017-05-16:
Here's another similar project from a while back, I think this one went between the GUI and Grbl: https://github.com/xpix/XLCD/wiki

dragoblaztr, 2017-05-16:
@109JB 
Thanks for sharing your code, I'll give a look.

As I said before the first version comes with I2C display, and depending on results would be for the tft or not.

@winder 
I already see that on my research, I'll keep it on mind.

@ everyone

Ideas and request are welcome to first version.

at this point as I said the main functions are these:

- Supporting for SD read and stream GCode without PC
- Display coordinates (work or machine) at runtime of code or jogging.
- Jogging support
- Beep sound on interaction with buttons
- Interfacing via encoder, joystick, keypad (another devices pending).

If I forgot something for main idea let me know.

Cheers

vMeph, 2017-05-16:
with a big screen lcd like they use in 3 d printers like this would rox
[image]

it has enconders, buzzer,sd card

dragoblaztr, 2017-05-17:
@vMeph 
Would be, but there's a problem, I have not that display to make test...

vince87, 2017-06-23:
[image]
[image]

It's arrived today, finally after almost 2 months ... I let you know as soon as I try, they should work with both the raprap smart display  or fullgraphic smart controller !!!

vMeph, 2017-06-23:
Cool

dragoblaztr, 2017-06-23:
@vince87 
The thing is that donesn't have function for streaming code via SD card... besides I tried that a time ago on arduino uno doing the respective changes on pinout.

109JB, 2017-06-23:
It appears that the boards pictured have 10 pin connectors for hooking up to a 3d Printer smart controller. The SD would then be in the smart controller meaning that board doesn't need it.

vMeph, 2017-06-23:
what you mean @109JB  the smart controler with sd card have 2 cables of 10 pin each cable,

109JB, 2017-06-23:
Yes, and the boards pictured have spots for 2 10-pin cable connectors. Since the smart controller has the SD on it, the boards in the previous post don't need an SD card slot because the one on the smart controller can be used.

vince87, 2017-06-23:
[image]
[image]

@109JB  You're right! I have made a screw terminal for power supply and 2 10-pin connector for the reprap smart controller or the big full graphic smart controller, and a 2 simple pin to connect on the serial pins of grbl!

vince87, 2017-07-28:
[image]
Hi guys, here's the new improved version of my PCB.
Software still needs to improve but at least it works !!!

Here the code...

// GRBL senza uso del pc!

#include <SoftwareSerial.h>
#include <SD.h>
#include <LiquidCrystal.h>
#include <Rotary.h>

String gcd;
char* myfile;
String RxString;

long time_now, seconds;
int flag = 0;
int refresh = 0;
int enc = 0;

int file_number = 0;
int how_many_file = 0;

const int chipSelect = 10;

int menu_entry = 0;

String words[6] = {"Attesa GRBL...", "Errore SD", "SD OK!!!", "Errore!!!", "Stop!!!", "Fine invio!!!"};  //Italiano
String menu[2] = {"Scelta File", "PC"};  //Italiano

char* file[20] = {};

LiquidCrystal lcd( 5, 4, 3, 6, 2, 7 );

SoftwareSerial mySerial(8, 9); // RX, TX

int pinEncSW = 11;
int pinEncA = 13;
int pinEncB = 12;

Rotary r = Rotary(pinEncA, pinEncB);

void setup()
{
  pinMode(pinEncSW,INPUT_PULLUP);
  pinMode(A0,OUTPUT);
  Serial.begin(115200);
  mySerial.begin(115200);
  //while (!Serial) { ; }

  lcd.begin(20, 4);
  lcd.clear();
  lcd.setCursor(0, 0);
  lcd.print(words[0]);
  lcd.setCursor(0, 1);

  pinMode(10, OUTPUT);
  if (!SD.begin(chipSelect))
  {
    lcd.print(words[1]);
    return;
  }
  
  lcd.print(words[2]);

  // sveglio GRBL :
  delay(1000);
  mySerial.write("\030"); // invio cr+lf
[... 246 lines cut ...]
  if ( enc == 1 && menu_entry < 1) { lcd.clear(); menu_entry++; refresh = 0; }
  if ( enc == -1 && menu_entry > 0) { lcd.clear(); menu_entry--; refresh = 0; }
  if ( enc == 100) { 
    if ( menu_entry == 0) { lcd.clear(); refresh = 0; chose(); }
    if ( menu_entry == 1) { lcd.clear(); refresh = 0; pc(); }
    }
  if ( enc == 0) {
    while(refresh == 0) {
      while(i != 3) {
        lcd.setCursor(0, menu_entry);
        lcd.print(">");
        lcd.setCursor(1, 0);
        lcd.print(menu[0]);
        lcd.setCursor(1, 1);
        lcd.print(menu[1]);
        //lcd.setCursor(1, 2); //if needed
        //lcd.print(menu[2]); //if needed
        //lcd.setCursor(1, 3); //if needed
        //lcd.print(menu[3]); //if needed
        i++;
      }
     refresh = 1;
    }
  }
}

vince87, 2017-07-28:
The serial passtru does not work very well ...
Streaming and reading gcode instead looks ok!

dragoblaztr, 2017-09-01:
Look at that its more powerfull and does't require grbl

https://translate.google.com/translate?sl=auto&tl=en&js=y&prev=_t&hl=es&ie=UTF-8&u=http%3A%2F%2Fwww.cnczone.ru%2Fforums%2Findex.php%3Fshowtopic%3D3334&edit-text=

LeoSanta, 2017-09-19:
Ciao, dove posso trovare lo schema per il wiring dei due Arduino con la SD e lo schermo LCD? 
Grazie mille.

LeoSanta, 2017-09-19:
Scusami, ma non trovo nulla. Me lo puoi linkare? 

Grazie

vince87, 2017-09-19:
https://easyeda.com/editor#id=8fff9e5f282e4371bb80ce178759623a
Eccolo!
Lo schema è disordinatissimo!!!

maaadbob, 2017-09-19:
> <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 	Virus-free. www.avg.com

Drive the relays from their own regulator.
Add reverse biased diode across the relay coils, else when you switch them off you might destroy the transistors.

vince87, 2017-09-19:
@maaadbob yes, you are right, I added under the pcb because I forgot them!

francescobic, 2017-09-27:
Ciao Vincenzo
hai per caso lo schema per le connessioni tra lcd e arduino1? 

https://codebender.cc/sketch:119423
questa è l'ultima release che hai fatto?

grazie mille

vince87, 2017-09-28:
Ciao @francescobic ho abbandonato arduino uno come controller.
Questa è l'ultima versione!:
https://create.arduino.cc/editor/vincenzobini/a9a47d2f-e858-4930-80ff-421df3c3a1e8
uso arduino micro che monta un atmega 32u4 molto più performante!
lo schema di collegamento lo trovi 4 post più sopra!
Ciao

francescobic, 2017-09-28:
grazie! ho ancora una domanda, sembrerebbe funzionare tutto...dice stream e fine invio...ma poi la macchina non parte...inoltre ho visto che c'è un file sketch.json, va inserito da qualche parte?
grazie

vince87, 2017-09-28:
è strano che la macchina non parte, c'è abilitato solo la verifica in grbl? pin tx/rx corretti? non sono quelli standard, ma D8 e D9! scollegato usb da grbl?
dove hai trovato il file sketch.json? non l'ho creato io!!!
Fammi sapere come va!

francescobic, 2017-09-28:
il .json era nella cartella scaricata... usb scollegata...pin 8 su tx e pin 9 su rx...
cosa intendi per  "c'è abilitato solo la verifica in grbl?"

il mio grbl è questo : https://github.com/misan/grbl-servo

grazie ancora

francescobic, 2017-09-28:
ah!!!  mi dice fine invio ! ! ! subito dopo stream
poi rimane così

vince87, 2017-09-28:
Prova a ricaricare ora lo sketch, ho modificato un paio di cose.
ho anche abbassato la velocità della seriale a 57600 baud così il 32u4 è meno stressato.
Dovresti ricompilare grbl e impostarlo a 57600 baud

vince87, 2017-09-28:
P.s. cerca qui su github "cncjs"
configuri un raspberry come controller e visualizzi tutto via wifi da un altro pc è molto piu' comodo!

francescobic, 2017-09-28:
gentilissimo grazie!!
ora mi guardo anche cncjs grazie

francescobic, 2017-09-29:
Ciao @vince87 
alla fine dopo un pò di smanettamenti ce l'ho fatta!!!!!
con baudrate a 57600 non funzionava così dopo alcune provo sono riuscito con baudrate 9600..
alè
Grazie ancora

cri-s, 2017-09-29:
28.8k o multipli non funzionano, 38,4k va e meglio ancora 250k .

luisluna2, 2018-09-28:
hello @vince87  Amazing Work!, congratulations, this Job need more recognition.

Excellent, hope for more videos and pictures about you work!

vince87, 2018-09-28:
@luisluna2 the project is dead, now I use cncjs connected to a raspberry ... much better!

luisluna2, 2018-09-28:
maybe, cans explain more friend?

vince87, 2018-09-28:
have a look:
https://github.com/cncjs/cncjs/wiki
is a sotware (that runs very well on raspberry) to control the grbl, has many functions, including remote control, the ability to connect a display to see the process etc ...

algo1980, 2018-10-01:
hi, I would like to join the dev team. are there any crude procedures?

vince87, 2018-10-02:
For cncjs: https://github.com/cncjs/cncjs

my board:
https://easyeda.com/editor#id=|8fff9e5f282e4371bb80ce178759623a|2bdadbb8457b4f33bf99c5b31b052e09
the arduino micro sketch:
https://create.arduino.cc/editor/vincenzobini/57ff6378-9ff6-463a-8479-8aac81978e04/preview

luisluna2, 2018-10-02:
> For cncjs: https://github.com/cncjs/cncjs
>
> my board: [...]

have PinOut _Friend?

AsadollahA, 2019-10-18:
Hi
I want to make a new CNC that I want to it work without PC (Headless CNC)
Can you help me to explain step by step it?

AsadollahA, 2019-10-18:
I want to it work by GRBL framework

vince87, 2019-10-18:
I abandoned this project.
i use cncjs, and it's really good!

AsadollahA, 2019-10-19:
Dear vince87
Thank you for your reply
I have to make headless CNC
What is cncjs?
Is this software or controller?
Did you success to finish the electronics of headless CNC and used it?

vince87, 2019-10-19:
see it: https://github.com/cncjs/cncjs
it work with a raspberry connected to the grbl.
You can control it by a lcd screen on raspberry or via web interface from pc, tablet or smartphone.

AsadollahA, 2019-10-19:
Really!!
I only work by Ardoino and GRBL and don't work by raspberry pi
Do you have step-by-step instructions for it?

gore80, 2019-10-23:
Hi!
Could you please upload the whole sketch you use with the 2004lcd? I think the rotary.h causes problems for me.
Thanks René

vince87, 2019-10-23:
@AsadollahA step-by-step instructions are in README.md file inside the project folder
@gore80 the full sketch is linked in second post, and i don't use the library rotary.h

AsadollahA, 2019-10-24:
@gore80 
Dear gore80
Did you make a headless CNC?
I just want to use one simple G-code, and using the computer is not convenient for me

AsadollahA, 2019-10-24:
@vince87 
Dear vince87
Many site use of 2 Ardoino. one for GRBL and one for read SD card
But they do not have electrical plan or code for read SD card
Using raspberry pi is hard for me and I do not use it :(
Do you know a way for using 2 ardoinu to make step by step headless CNC? (code and elecrical plan)
Tanck you

[5 more replies]