grbl/grbl #718: Question: M05 on G00, M03 on G01, after end M05

ikalafat, 2015-05-30:
Hello,

I have rather specific question. I'm doing my master thesis at college, and I'm building 2D (no Z axis) laser powered PCB photoresist 'engraver'. I would like to achieve that laser is powered off (M05 with pwm control) when parser encounters G00 ... command, and enable laser (M03) on G01.

This is what i have done

gcode.c (lines 956 and forth..)

```
      switch (gc_state.modal.motion) {
        case MOTION_MODE_SEEK:
          **spindle_set_state(SPINDLE_DISABLE, gc_state.spindle_speed); **
        case MOTION_MODE_LINEAR:
          **spindle_set_state(SPINDLE_ENABLE_CW, gc_state.spindle_speed); **
        case MOTION_MODE_CW_ARC: 
         **spindle_set_state(SPINDLE_ENABLE_CW, gc_state.spindle_speed); **
        case MOTION_MODE_CCW_ARC:
        **spindle_set_state(SPINDLE_ENABLE_CW, gc_state.spindle_speed); **
```
(I've removed unnecessary parts of code here to simplify explanation)

Can anyone tell me, is there any better / simpler way to achieve laser control without heavily editing gcode files and adding M03/M05 commands?

Thanks!

langwadt, 2015-05-30:
why not use Z to control the laser?

abearden, 2015-05-30:
Several software solutions that will generate proper gcode for you with M3, M5, and S codes. This is your best option.

The reason your solution didn't work is you're executing the change in laser power in the command parser and not putting it in the buffer to be executed when the machine gets to that point. As soon as it reads the command the change is output to hardware when implemented that way. You'd want to do something like "gc_block.modal.spindle = SPINDLE_ENABLE_CW;" and "gc_block.modal.spindle = SPINDLE_DISABLE;" to put the spindle change into the buffer. You'd need a "values_words |= bit(MODAL_GROUP_M7);" in there too so the buffer knows an M command was used.

ikalafat, 2015-05-31:
@langwadt 
Machine physically doesn't have Z axis, as it is intended only for homemade PCB laser photoresist marking (and pcbs in most cases have same height)

@abearden 
I'll give it a try tomorrow. Thanks for guides and explanation.

ikalafat, 2015-06-01:
@abearden 

This is what I have made - so far it seems ok, but I'll test it more.
```
              case 0: gc_block.modal.motion = MOTION_MODE_SEEK; gc_block.modal.spindle = SPINDLE_DISABLE; word_bit |= bit(MODAL_GROUP_M7); break; // G0
              case 1: gc_block.modal.motion = MOTION_MODE_LINEAR; gc_block.modal.spindle = SPINDLE_ENABLE_CW; word_bit |= bit(MODAL_GROUP_M7); break; // G1
              case 2: gc_block.modal.motion = MOTION_MODE_CW_ARC;  gc_block.modal.spindle = SPINDLE_ENABLE_CW; word_bit |= bit(MODAL_GROUP_M7);  break; // G2
              case 3: gc_block.modal.motion = MOTION_MODE_CCW_ARC; gc_block.modal.spindle = SPINDLE_ENABLE_CW; word_bit |= bit(MODAL_GROUP_M7);  break; // G3
```

Thanks again!

langwadt, 2015-06-02:
@ikalafat What meant was use a "virtual" Z axis, look at the Z position variable and use that to turn the laser on/off below/above zero

ikalafat, 2015-06-02:
@langwadt 
Sorry, I misunderstood you. I must admit that your idea hasn't come to my mind while looking for solution.

As modifications suggested by @abearden work fine, I'll close this 'issue'.

Thanks!

closed 2015-06-02 by ikalafat