grbl/grbl #723: Request for info about F word

onekk, 2015-06-02:
@chamnit what is the correct mode to set the F feedrate.
I'm writing a postprocessor for a CAM program (and trying to write also a 3D CAM program). I'm seeing in some parser that claim to work with Linuxcnc a structure like this:

    G21 (Metric)
    G40 (No Offsets)
    G90 (Absolute-Mode)
    G61
    F100
    G0 (something)
    (and then some Gcode)

But when sending this code in GRBL i often obtain an error if the first G1 didn't have the F word in it.
after writing F100 (for example) on the first G1 line often even if the others G1 line there is no F word GRBL seems to accept the G-Code file flawlessy. (Sorry for not being too exact but I'm a novice in GCode) reading that GRBL aim to be compatible with LinuxCNC i wonder why the above code didn't work.

regards

Carlo D.

chamnit, 2015-06-02:
@onekk : LinuxCNC often tweaks their gcode definitions. It may be that they updated the way feedrate is set. Or it's a bug in Grbl's gcode parser. I'd would need to make sure an F word can be set by itself. I can't recall if that is true or not off hand.

csdexter, 2015-06-02:
The Book appears to suggest that (a) F can exist alone and behave like any other register (i.e. contain a dimensionless value that becomes the default when such a value is needed by the current command but not specified) and (b) G01 should flag an error and terminate execution if F was not given and there is no default (i.e. F was never seen in the program up to this point).

chamnit, 2015-06-02:
It does (b) but with (a), I vaguely recall a single F word was supposed to generate an error. Can't remember where or when. Not can I find that reference. It's possible it was on a pervious LinuxCNC gcode description version. I'll have to look again when I get time.

109JB, 2015-06-03:
Linuxcnc can take the F word by itself without any problem. I has been able to since at least version 2.4.

I checked Grbl 0.9I and it does the same. I cannot duplicate the problem @onekk mentioned even using his sample.

onekk, 2015-06-03:
OK i will try to generarate a sample file and test with  my GRBL 0.9i then if the issue it is not here, sorry for the inconvenience, if it is here, sorry for the bad report and resubmit a better error report with files and so on. My machine is in another place where i didn't have an internet connection.

Many thanks to all

Regards 
Carlo D.

chamnit, 2015-06-03:
@109JB : Yep, you're right. F can be sent by itself, and it does work in Grbl. I think it was a `G1 Fxxx` command without XYZ axis words that I was thinking about. At one point, I had mis-interpreted or read somewhere that it is an error if you send a `G1` command without any axis words. So, `G1 Fxxx` was illegal on Grbl for a short time. That has since been fixed.

onekk, 2015-06-04:
I can confirm that this code work on GRBL 0.9i sendit it with bCNC:

    G21 (Metric)
    G40 (No Offsets)
    G90 (Absolute-Mode)
    
    F100
     (--------------------------------------------------)
     (Object: #3)
     (Layer: 0)
     (Overcut: 0)
     (Depth: -3.000000)
     (Offset: Inside)
     (--------------------------------------------------)
    M6 T1
    M03 S10000
    G00 X0.000000 Y0.000000 Z4.000000
    G00 X9.551426 Y4.022796
    G01 Z-0.500000
    G02 X8.181803 Y7.890494 I5.525276 J4.132956 F200
    G02 X9.278293 Y11.844310 I6.896607 J0.216352
    G02 X12.384212 Y14.404632 I5.751052 J-3.812532
    G01 X17.724922 Y14.358025
    ....

Sorry for the inconvenience.

Carlo D.

chamnit, 2015-06-04:
@onekk : Thanks for reporting.

closed 2015-06-04 by chamnit